merbanan/rtl_433 #2654: Source control and integration plans for flex decoders is unclear

gdt, 2023-10-01:
The main path of rtl_433 is to have decoders written in C, that have a protocol number inciidentially, recognize packets, and emit data records.   There are many of them, and they are enabled by default if we believe they will not create false decodes annoyingly often.  Decoders with a high false rate (due to poor checksums) are disabled by default.

There is a flex decoder, where someone can configure decoding rules.  This is conceptually like a decoder in C, except it is declarative in  a Domain-Specific Language.  This is great because it greatly reduces the effort required and the amount of code to maintain.  Flex decode outputs lack a protocol number, which is not necessarily a good thing.

People ask for help decoding a transmitter and those issues often end up with a flex decoder recommendation.   That can be put in a config file.

It seems intuitively obvious :-) that we should have a place in the repo to capture these definitions, and adding that should be how device support issues (that end up with flex decoders) are resolved.   It is less clear how these should be treated by default.

As I see it, flex decoder shoudl  not be different because they are in the flex DSL instead of C.  If they do  not false, they should be enabled by deafult.  If they do, no.

This issue is to decide on what we are doing so we can document the plan.

gdt, 2023-10-04:
Also, I have only recently grasped that flex decoders do not support checksum/CRC, and are thus generally deficient compared to a "real decoder".

klohner, 2024-01-08:
I'd love to have flex decoder options to validate CRCs and to whitelist/blacklist signal decoding based on a bitmask on a "get=" somehow.  

It'd also be useful to be able to set a default named value for some `get` data, like `get=XYZ:@0:{8}:[0:A 1:B 2:C *:DEFAULT]` or like `get=XYZ:@0:{8}:[0:ZERO 1:ONE 2:TWO *:%02x]`.

I'd think that with a little more development in the flex options to cover some deficiencies,  it could probably duplicate many, if not most, of the dedicated decoders.  Maybe we could talk through what this syntax might look like?  Is something like this feasible?

zuckschwerdt, 2024-01-09:
There are ideas how to make getters easier to use and how to add checks like sums and crcs. But I'm reluctant to write a parser and such in C.

klohner, 2024-01-09:
That's reasonable.

Just brainstorming, if we get away from thinking flex decoder, does it open up options for using something like [Squirrel](http://www.squirrel-lang.org/) to handle custom decoder scripts?

Or, what if there were a supported framework for offloading a bit decoder to a separate process?  Maybe have something in rtll_433.conf like:

```
decoder {
    name=EV1527,
    modulation=OOK_PCM,
    short=200,
    long=200,
    reset=5000,
    parser=EV1527
}

parser {
    name=EV1527,
    executable="ev1527_parser"
}
```

And maybe provide a defined JSON structure as a parameter (with info like frequency, sample rate, etc., as well as the decoder options like modulation and bit string, etc.), and accept JSON back.  Reply could tell rtl_433 success/failure with parsing, parsed fields, etc., and maybe a new device name string for rtl_433 to report to the end user.

Is there already a vision to where rtl_433 is headed?  I'd imagine it's starting to get unwieldy to compile an ever-growing myriad of decoders directly into the base rtl_433 program.

zuckschwerdt, 2024-01-09:
Squirrel, Lua, JS (Duktape), all workable. But the core use would be manipulating fields of bits, and none of those langs make it easy. We could better offer some expressions native to bits from a bitbuffer and implement a small DSL.

It wouldn't be too challenging to code a DSL parser and executor in a lang other than C. Not sure if that is user-friendly to build though. (Compile the parser to lib in some way, then compile and link rtl_433 using C. And if the parser is optional users stuck with just a C compiler could opt out.)

All that would exclude the interesting use-case with the ESP port. Maybe we need to offer a pre-compiler to transform flex decoders to C code so it can be integrated in (embedded platform) builds?

The "external decoder" idea is okay and can be done right now, just get data from the HTTP API and process it. No way currently to return results though.

I would like to have a more complete HTTP API so an eco-system of external programs can interact with rtl_433 in more interesting ways. E.g. a more flexible hopping controller that controls the freq. And message processing, maybe decoders -- as you also mention.

gdt, 2024-01-09:
It would be straightforward to have custom datatypes and operators in guile, and use scheme :-)

As to compiling ever-growing decoders and unwieldy, I'd say the big issues are:
  - figuring out and writing each decoder which is 90% not about C
  - balancing decoders that over-decode and false with usefulness, because of bad protocols without strong checksums

and the 'build the code' part just works and is not the issue.

zuckschwerdt, 2024-01-10:
Maybe a good first step would be an example external decoder that attaches to the HTTP API. Then see where that leads.
We'd need to reimplement/port the slicers and bitbuffers as the API will only deliver OOK pulse text.