merbanan/rtl_433 #2660: rtl_433 looks in wrong places for config

gdt, 2023-10-06:
The correct place is arguably 1) current directory 2) XDG 3) $prefix/etc as configured by cmake.

It does 1 and 2, and then:
```
        paths[2] = "/usr/local/etc/rtl_433/rtl_433.conf";
        paths[3] = "/etc/rtl_433/rtl_433.conf";
```

The first should vary by prefix; /usr/local is simply the default.   The last should not be consulted, as the normal expectation is to look where the etcdir was configured by cmake.   (That will be /etc if prefix is /usr, so there's no real change for the GNU/Linux packaging case.)

I think it's only an artifcact of  nobody having a `/usr/local/etc/rtl_433/rtl_433.conf` file that we aren't seeing problems from this.

gdt, 2023-10-06:
We'll have to figure out the details.   This seems to be yet another case of autoconf doing something right and cmake reinventing it badly!  It is absolutely normal for packages to find the appropriate etc given their prefix.

zuckschwerdt, 2023-10-06:
Ok, it actually does work properly with
```cmake
include(GNUInstallDirs)
add_definitions(-DINSTALL_SYSCONFDIR=${CMAKE_INSTALL_FULL_SYSCONFDIR})
```

zuckschwerdt, 2023-10-06:
Do we want to change to that and drop the `/etc` default? @merbanan

commit a36f9a20 (2023-10-18): Change conf dir defaults to just SYSCONFDIR (closes #2660)
closed 2023-10-18 by zuckschwerdt