merbanan/rtl_433 #2669: Fails to work on second load

adminy, 2023-10-14:
Using nixos with the following device:
```
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
```
I get the following errors after closing it and running it a second time:
```
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
No supported tuner found
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
Enabled direct sampling mode, input 1
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
Exact sample rate is: 250000.000414 Hz
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
WARNING: Failed to set sample rate.
Tuner gain set to Auto.
rtlsdr_write_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
WARNING: Failed to set center freq.
Allocating 15 zero-copy buffers
Async read stalled, exiting!

LIBUSB_ERROR_NOT_FOUND: Entity not found!
Check your RTL-SDR dongle, USB cables, and power supply.

WARNING: async read failed (-5).
rtlsdr_write_reg failed with -7

```

gdt, 2023-10-14:
This is unlikely an rtl_433 bug.  You didn't explain what rtl_433 version you are running (if not git master please update and try again).    You didn't explain if you removed/reinserted and if that fixes it, if there's anythign in syslog, and what happens when you use other programs with the dongle.

adminy, 2023-10-15:
I'm running version 22.11 which is the latest release on nixos, I can try master to see if that fixes it. The dongle is in the attic, I don't want to plug it in and out everytime I restart the app. Reboot fixes it so its not a connectivity issue for sure. journalctl gives back this after the app restart:
```
kernel: usb 1-1.3: reset high-speed USB device number 3 using xhci_hcd
kernel: usb 1-1.3: device descriptor read/64, error -110
kernel: usb 1-1.3: device descriptor read/64, error -110
kernel: usb 1-1.3: reset high-speed USB device number 3 using xhci_hcd
kernel: usb 1-1.3: device descriptor read/64, error -110
kernel: usb 1-1.3: device descriptor read/64, error -110
kernel: usb 1-1.3: reset high-speed USB device number 3 using xhci_hcd
```

I recompiled master and tried it again: Got:
```
usb_open error -4
```
which is different than what I was getting before with 22.11. So I rebooted the os again.

back to:
```
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
No supported tuner found
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
Enabled direct sampling mode, input 1
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
Exact sample rate is: 250000.000414 Hz
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
WARNING: Failed to set sample rate.
Tuner gain set to Auto.
rtlsdr_write_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -7
rtlsdr_demod_read_reg failed with -7
WARNING: Failed to set center freq.
Allocating 15 zero-copy buffers
Async read stalled, exiting!

LIBUSB_ERROR_NOT_FOUND: Entity not found!
Check your RTL-SDR dongle, USB cables, and power supply.

WARNING: async read failed (-5).
rtlsdr_write_reg failed with -7
```

Then I restarted the app one more time and:
```
rtlsdr_write_reg failed with -7
Resetting device...
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

back to working ???

Can I always make it do the "resetting device ..." step?

gdt, 2023-10-15:
This seems like you have a flaky device, or there is buggy kernel/firmware code.   It doesn't seem like an rtl_433 bug.
I do not find "Resetting device" in either the rtl_433 or rtl-sdr source code, so I'm guessing it is in Linux.

This might be  https://github.com/raspberrypi/linux/issues/3060

zuckschwerdt, 2023-10-18:
We have seen problems like this before, but we never found a solution. It just seems some devices glitch and only a bus reset gets them back. There are various usb_reset tools if you want something short of a full power-cycle.

closed 2023-10-18 by zuckschwerdt

tux-, 2024-05-16:
I have had this same issue, and for it happens on Ubuntu 22.04, but not Ubuntu 23.10 and 24.04. Tested with multiple sticks and multiple machines. I now have all machines upgraded, and problem has not returned.