merbanan/rtl_433 #2671: New device LIVARNO HG09649 (LIDL) blind remote

schroeder-paul, 2023-10-14:
Hey,

Is there any how-to, to create a decoder for new devices? I did not find anything (although I was not super carefully searching)...

I bought some LIDL blind a while ago, and they feature a remote control:

[image]

I would like to read the button presses, here are some [captures cap.zip](https://github.com/merbanan/rtl_433/files/12906173/cap.zip)

gdt, 2023-10-14:
more docs/*

klohner, 2024-01-08:
Interesting signal.  I only see data in g003, g005, g007, and g008.

You can see the raw data with something like:

rtl_433 -R 0 -X "n=Livarno,m=OOK_PCM,s=333,l=333,r=5000," -r g008_433.92M_250k.cu8

Sync looks like a lot of `0xaa`,
There then seems to be 3 repetitions of the data, each maybe prefixed with `0xccccccccff00`.  Data is Manchester encoded, but if using 333μs per symbol, the MC data is double-wide, ie. `0x1100` or `0x0011` for logical 1 or 0.

You can pick out these rows with something like:

`rtl_433 -R 0 -X "n=Livarno,m=OOK_PCM,s=333,l=333,g=2000,r=5000,bits=216,symbol_one={4}c,symbol_zero={4}3" -r g008_433.92M_250k.cu8`

Or, pick out just the first repetition by itself with something like:

`rtl_433 -R 0 -X "n=Livarno,m=OOK_PCM,s=333,l=333,r=5000,preamble={16}ff00,symbol_one={4}c,symbol_zero={4}3" -r g008_433.92M_250k.cu8`

I can't seem to find a CRC.  Here's a [BitBench](https://triq.net/bitbench#c=%7B51%7D83d68d207b800%20%5Bg003_433.92M_250k%5D&c=%7B51%7Dc3d68d6047800%20%5Bg005_433.92M_250k%5D&c=%7B51%7Dc3d68d2007800%20%5Bg007_433.92M_250k%5D&c=%7B51%7Dc3d68d407b800%20%5Bg008_433.92M_250k%5D&c=%5B%5D&f=%3F%3F%3Abb%20ID%3F%3Ahhhhh%20DATA%3F%3Ahhhhh%20SYNC%3F%3Abb%20bbbbbbbbbbb&cw=4) of what I found.