merbanan/rtl_433 #2675: Version string for master / nightly builds hash an extra g in the commit hash

deviantintegral, 2023-10-18:
When building rtl_433, the version string will be something like:

```
rtl_433 version nightly-17-gfc727b3d branch master at 202308051113 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
```

However, `gfc727b3d` is incorrect. The actual commit hash the build was on is `fc727b3d`. We can see this in a recent build at https://github.com/pbkhrv/rtl_433-hass-addons/actions/runs/6558257508/job/17811466141#step:9:822.

I believe either the g should be removed, or perhaps if it means something ("git version?") then a dash needs to be added.

Originally reported / discovered at https://github.com/pbkhrv/rtl_433-hass-addons/issues/161.

zuckschwerdt, 2023-10-18:
The output is from `git describe --tags`. It's not a commit hash but an "abbreviated object name of the most recent commit." Yes, "the "g" prefix stands for "git" and is used to allow describing the version of a software depending on the SCM the software is managed with."
In short, it's a name not a git commit hash.

We don't currently have the capacity to challenge the basic tooling.  But the even worse problem is we really should skip the (ephemeral) "nightly" tag. And we can do something about that at least. (`git describe --tags --exclude=nightly` -- done with 138b389)

zuckschwerdt, 2023-10-18:
To illustrate why this change should already help: we now get `22.11-215-ge826a465` which is easier to read and simple to compare "the 215th patch on 22.11".

closed 2023-10-18 by zuckschwerdt

deviantintegral, 2023-10-19:
I wasn't familiar with the `git describe` command directly. Thanks for the details. Excluding nightly leaves much clearer string. Also from the help page:

```
The "g" prefix stands for "git" and is used to allow describing the version of a software depending on
       the SCM the software is managed with. This is useful in an environment where people may use different SCMs.
```

I'll add a note to our docs as well. Thanks!