merbanan/rtl_433 #2691: Fine Offset Electronics WS90 weather station

Oleg-Sob, 2023-10-24:
I have a WS90 433 MHz weather station and TTGO LoRa32 V2.1 1.6.1 with rtl_433, without Ecowitt GW2000 Gateway Wi-Fi. I don't see the station. Tell me WS90 with rtl_433 will work without GW2000 Gateway Wi-Fi.

zuckschwerdt, 2023-10-24:
Yes, but WS90 support is rather new, see #2448
You need to check if the rtl_433 esp32 port already includes protocol 244.

closed 2023-10-24 by zuckschwerdt

Oleg-Sob, 2023-10-24:
TTGO LoRa32 V2.1 1.6.1 was flashed today from here https://docs.openmqttgateway.com/upload/web-install.html firmware lilygo-rtl_433. Tell me how to check in the console whether there is protocol 244.

zuckschwerdt, 2023-10-24:
If that is the original port by @NorthernMan54 then it looks to be not included.
https://github.com/NorthernMan54/rtl_433_ESP

Oleg-Sob, 2023-10-24:
Thanks, I already saw that it is not supported. I'll wait until they turn it on.

NorthernMan54, 2023-10-25:
@Oleg-Sob About every 6 to 12 months I do refresh of the device decoders, and this will be included then.   You can close this here, and track it via this issue - https://github.com/NorthernMan54/rtl_433_ESP/issues/111