merbanan/rtl_433 #2693: New device Bresser Air Quality Sensor PM2.5/PM10, PN 7009970

matthias-bs, 2023-10-25:
Hi,

I found out that the Air Quality Sensor can be supported with small changes to the [bresser_7in1.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_7in1.c) decoder. The data field `sensor_type` (as in other Bresser decoders) allows to distinguish between weather and air quality sensor. Only the actual sensor data would be different:

**Weather Sensor (Sensor Type = 1)**
```
static char const *const output_fields[] = {
        "model",
        "id",
        "sensor_type",
        "temperature_C",
        "humidity",
        "wind_max_m_s",
        "wind_avg_m_s",
        "wind_dir_deg",
        "rain_mm",
        "light_klx", // TODO: remove this
        "light_lux",
        "uv",
        "battery_ok",
        "mic",
        NULL,
};
```

**Air Quality Sensor (Sensor Type = 8)**
```
static char const *const output_fields[] = {
        "model",
        "id",
        "sensor_type",
        "pm2_5_ug_m3",
        "pm10_0_ug_m3",
        "battery_ok",
        "mic",
        NULL,
};
```

What is the preferred way of adding this device?
- Creating a new decoder source file by copying from `bresser_7in1.c`?
- Adding a new decoder function to `bresser_7in1.c`?
- Modifying the existing decoder function in `bresser_7in1.c` and adding or multiplexing (how?) the output fields?

Thanks in advance!

Matthias

matthias-bs, 2023-10-25:
Example:
```
          Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25
        Raw Data: 7E E0 6E BB AA FA 81 AA FA AA FA AA CA AA 2A A9 AA A9 2A AE AA 92 FA AA AA 00 
De-whitened Data: D4 4A C4 11 00 50 2B 00 50 00 50 00 60 00 80 03 00 03 80 04 00 38 50 00 00 AA
                        ^^ ^^ ID
                                    ^ SENSOR_TYPE=8 (raw)
                                     ^  STARTUP | CH=1 (raw) 
                                                 ^ ^^ ^ PM2.5=6
                                                       ^ ^^ ^ PM10=8
                                                                ^ BATT
```

zuckschwerdt, 2023-10-25:
Is it the same CRC? Then I'd use the sensor_type (`b[6]`?) and split the output with an `if`

matthias-bs, 2023-10-25:
Yes, same CRC. Could you please give a little more detail on how to split the output?

matthias-bs, 2023-10-27:
Ah, I think I got it now...

MacH-21, 2023-10-27:
Hi Can we not use one CRC calc for most Bresser Stations /Sensors?

FYI way back in Issue 1214 feb 12 2021
(For Digest I use CCIT-16 :X16+X12+X5+1 CRCseed = All 0's) which is 
CRC-16/XMODEM check-0x31C3 	poly-0x1021 	init-0x0000 	false 	false 	0x0000

You can use https://crccalc.com to test
  
Raw Data:		 7E E0 6E BB AA FA 81 AA FA AA FA AA CA AA 2A A9 AA A9 2A AE AA 92 FA AA    AA 00 
CRC over 22 bytes	   6E BB AA FA 81 AA FA AA FA AA CA AA 2A A9 AA A9 2A AE AA 92 FA AA 
7ee0  > CRC = ok
	
De-whitened Data:D4 4A C4 11 00 50 2B 00 50 00 50 00 60 00 80 03 00 03 80 04 00 38 50 00    00 AA

FROM     rtl_433/src/devices/bresser_7in1.c

    631d 5c 09 e9 a1 8a ba ab aa aa aa aa a8 ad ac ba cf f9 ca fc aa aa aa      a000000000000000000
	631d  > CRC = INVALID should be f05c

    10b8 b4 a5 a3 ca 10 aa aa aa aa aa aa aa 8b ca cb aa aa 2a aa aa aa aa      aa0000000000000000 [0.08 klx]
	10b8  > CRC = ok
	
    543b b4 a5 a3 ca 10 aa aa aa aa aa aa aa 8b ca cb aa aa 28 aa aa aa aa      aa00000 [0.08 klx]
	543b  > CRC = ok
	
    2492 b4 a5 a3 ca 10 aa aa aa aa aa aa aa 8b da cb aa aa 2d aa aa aa aa      aa0000000000000000000 [0.08klx]
	2492  > CRC = ok
	
    9a59 b4 a5 a3 da 10 aa aa aa aa aa aa aa 8b da c8 af ea 28 a8 ca aa aa      aa000000000000000000 [54.0 klx UV=2.6]
	9a59  > CRC = ok
	
    fe15 b4 a5 a3 da 10 aa aa aa aa aa aa aa 8b da cb ba 38 2a ac da aa aa      aa00000000 [109.2klx   UV=6.7]
	fe15  > CRC = ok
	
    2544 b4 a5 a3 2a 10 aa aa aa aa aa aa aa 8b da c8 8a aa aa be aa aa aa      aa00000000000000 [200.000 klx UV=14
	2544  > CRC = ok
	

	
	7in1 outdoor = CRC-16/XMODEM over 22 bytes	
	Lightning = CRC-16/XMODEM over 7 bytes	
	Leakage = CRC-16/XMODEM over 5 bytes	
	PM2.5/PM10 = CRC-16/XMODEM over 22 bytes
	
	
	6in1 outdoor = CRC-16/XMODEM over 15 bytes
	F043 19 10 00 0F 18 FF FF FF 20 98 FF FF FF FF 01 FD
	f043 > CRC = ok

zuckschwerdt, 2023-10-27:
The Bresser-7in1 works with CRC-16 of 0x1021? We found it to be a LFSR-16 generator 0x8810 -- but of course that could be wrong.

matthias-bs, 2023-10-28:
It makes sense to unify the checksum calculation across decoders if possible. There doesn't seem to be much different in terms of computation effort, though.

gdt, 2023-10-30:
@matthias-bs If there is a PR that captures all the useful information in the issue, please close the issue.  Otherwise, please add a comment summarizing what's in the issue and not the PR, why, and what you think the way forward is.  Thanks!

matthias-bs, 2023-10-30:
See https://github.com/merbanan/rtl_433/pull/2698

closed 2023-10-30 by matthias-bs

ME-DataStudio, 2023-11-18:
I have an Bresser Air Quality sensor, but it shows as a 7in1. 
rtl_433 version 22.11 branch  at 202211191645
Output:
time      : 2023-11-18 10:37:29
model     : Bresser-7in1 id        : 11585
Temperature: -9.9 C      Humidity  : 40            Wind Gust : 1.0 m/s       Wind Speed: 0.2 m/s
Direction : 9            Rain      : 25.0 mm       Light     : 154.016 klx   Light     : 154016.000 lux
UV Index  : 0.3          Battery   : 1             Integrity : CRC

ME-DataStudio, 2023-11-20:
I have a Bresser air quality sensor and a rte-sdr v3 dongle. The PM10 value from the decoder seems to be wrong. The value on my lcd-display with the sensor gives other values. I tried to get some examples but the -vvv gives a lot of output so when I figured out how to catch only the air quality I will post an example

zuckschwerdt, 2023-11-20:
> the -vvv gives a lot of output so when I figured out how to catch only the air quality

Try `-R 173:vvv`

Mart124, 2024-02-06:
For those who wonder, Bresser PM2.5/PM10 device, PN 7009970, runs a [Sensirion SPS30](https://sensirion.com/products/catalog/SPS30) sensor 👍