merbanan/rtl_433 #2697: [rtl_433_mqtt_hass.py] Topic prefix with slashes

haraldhh, 2023-10-27:
I publish my mqtt data using a prefix in the form of /clan/house, so rtl_433 devices go into /clan/house/rtl_433/devices[/id] and events into /clan/house/rtl_433/events[/model][/id]. I'm running rtl_433_mqtt_hass.py to try to autoconfigure devices into HA, but I'm having problems with the configuration messages having missing paths in the topic. 

In the configuration message I get the following topic;

`/clan/devices/Fineoffset-WH2/135`

I would like it to be like;

`/clan/house/rtl_433/devices/Fineoffset-WH2/135`

Looking at the code, I see this on line 487;

`    topicprefix = "/".join(msg.topic.split("/", 2)[:2])`

I'm not that used to coding, especially Python but to my Perl-eyes this looks like it's omitting stuff from the topic-string. Am I right, and could I fix this somehow so I could get my measurements to show up in HA? 

Thanks

haraldhh, 2023-10-27:
Just by guessing I changed the topicprefix line to this;

`    topicprefix = "/".join(msg.topic.split("/", 4)[:4])`

And it looks much better. What did I break?

zuckschwerdt, 2023-10-27:
That seems the correct fix. The script expects the default prefix of `rtl_433/HOSTNAME/` and can't deal with other patterns currently.

haraldhh, 2023-10-27:
All of the Fineoffset-WH sensors (Telldus) now work. I have a Prologue-TH (Biltema) that doesn't. The topic for the Prologue becomes 

`/clan/house/rtl_433/devices/Prologue-TH/9/3/213/temperature_C`

The real topic should be 

`/clan/house/rtl_433/devices/Prologue-TH/213/temperature_C`

But where does that /9/3 come from? Probably from my missing knowledge of split() :)

haraldhh, 2023-10-27:
I'm having the same issue with a Norgo energymeter. 

`homeassistant/sensor/Norgo-NGE101-1-167/Norgo-NGE101-1-167-B/config : {"device_class": "battery", "name": "Norgo-NGE101-1-167-B", "unit_of_measurement": "%", "value_template": "{{ float(value|int) * 99 + 1 }}", "state_topic": "/clan/house/rtl_433/devices/Norgo-NGE101/1/167/battery_ok", "unique_id": "Norgo-NGE101-1-167-B", "device": {"identifiers": "Norgo-NGE101-1-167", "name": "Norgo-NGE101-1-167", "model": "Norgo-NGE101", "manufacturer": "rtl_433"}}`

This one is apparently named Norgo-NGE101-1-167, but the topic used to publish is /clan/house/rtl_433/events/Norgo-NGE101/167 note missing /1/

gdt, 2025-07-04:
still awaiting a PR!