merbanan/rtl_433 #2699: DSC PG 9945 Contact Sensor

shivamchoudhary, 2023-10-28:
I would like to write a decoder for [DSC Security Contact Sensor PG 9945,](https://www.dsc.com/manual/24725) it looks like it uses 915Mhz per 
[PG9945_PG8945_PG4945_Installation_Guide_29010199R003_EN_FR_ES_PT.pdf](https://github.com/merbanan/rtl_433/files/13194076/PG9945_PG8945_PG4945_Installation_Guide_29010199R003_EN_FR_ES_PT.pdf)

[image]

**Questions:**

- I tried rtl_433 -f 915M -s unknown and have attached the output - can you please tell me how to go about decoding ? I already tried running the file with -A but could not find my sensor name (was expecting something like DSC).
[g003_915M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/13194141/g003_915M_1000k.cu8.zip)

zuckschwerdt, 2023-10-28:
The recorded signal is clipping. Record at greater distance or remove the antenna.

There are actually two signals in that recording, the second looks weaker and like a response maybe.

shivamchoudhary, 2023-10-29:
I removed the antenna and opened the window 4/5 times to trigger the sensor, here's the recording. 
[g010_915M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/13197055/g010_915M_1000k.cu8.zip)

zuckschwerdt, 2023-10-29:
That looks better and we can see that it is 100 µs FSK PCM.
[image]
It should work with `-X 'n=name,m=FSK_PCM,s=100,l=100,r=1000'` but you need luck to get the signal right, rtl_433 isn't good at demodulating GFSK.

shivamchoudhary, 2023-10-30:
I tried the flex decoder - any idea how to decode the data ?

[capture.txt](https://github.com/merbanan/rtl_433/files/13208272/capture.txt)

shivamchoudhary, 2024-01-05:
Did a bit of research - PG 9945 is indeed encrypted and needs POWER G hardware to decode.

closed 2024-01-05 by shivamchoudhary