merbanan/rtl_433 #2700: Not able to install last version on Pi 3 64 bits

8um1c4, 2023-10-29:
Hi guys,

I was trying to update rtl_433 with new and last changes, but it hangs on 7% compilation after 2 major errors.
I'm using a Raspberry 3 with 64 bits OS, completely updated.
Log below.
Can you advise?

Regards.

-------------------------------

pi@Pi64:~ $ git clone https://github.com/merbanan/rtl_433.git
cd rtl_433/  
mkdir build
cd build
Cloning into 'rtl_433'...
remote: Enumerating objects: 22767, done.
remote: Counting objects: 100% (4913/4913), done.
remote: Compressing objects: 100% (637/637), done.
remote: Total 22767 (delta 4421), reused 4395 (delta 4270), pack-reused 17854
Receiving objects: 100% (22767/22767), 8.47 MiB | 3.05 MiB/s, done.
Resolving deltas: 100% (17891/17891), done.
pi@Pi64:~/rtl_433/build $ cmake ../
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found Git version: refs/heads/master commit 8535ba24f675afc498dc75ca84e00625aceac06f from 2023-10-27T12:26:51+0200
-- Using Git version tag: 22.11-238-g8535ba24 on master at 202310271226
-- Using GCC Static analysis
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
-- IPv6 support enabled.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Threads support will be compiled.
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "1.1.1w")  
-- OpenSSL TLS support will be compiled. Found version 1.1.1w
-- Found LibRTLSDR: /usr/lib/aarch64-linux-gnu/librtlsdr.so (found version "0.6.0")
-- Found LibUSB: /usr/lib/aarch64-linux-gnu/[libusb-1.0.so](http://libusb-1.0.so/) (found version "1.0.24")
-- RTL-SDR device input will be compiled. Found version 0.6.0
-- libusb-1.0 error messages are available. Found version 1.0.24
-- SoapySDR device input will be compiled. Found version 0.8.1-gd2496c50
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen missing components: dot
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/rtl_433/build
pi@Pi64:~/rtl_433/build $ make
Scanning dependencies of target data
[  0%] Building C object src/CMakeFiles/data.dir/data.c.o
[  1%] Building C object src/CMakeFiles/data.dir/abuf.c.o
[  1%] Linking C static library libdata.a
[  1%] Built target data
Scanning dependencies of target r_433
[  2%] Building C object src/CMakeFiles/r_433.dir/abuf.c.o
[  2%] Building C object src/CMakeFiles/r_433.dir/am_analyze.c.o
[  3%] Building C object src/CMakeFiles/r_433.dir/baseband.c.o
[  3%] Building C object src/CMakeFiles/r_433.dir/bitbuffer.c.o
[  3%] Building C object src/CMakeFiles/r_433.dir/compat_paths.c.o
[  4%] Building C object src/CMakeFiles/r_433.dir/compat_time.c.o
[  4%] Building C object src/CMakeFiles/r_433.dir/confparse.c.o
[... 189 lines cut ...]
                  |......
                  |  150 |     ring->data = calloc(size, sizeof(void *));
                  |      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                  |      |                  |
                  |      |                  (11) ...to here
                  |      |                  (12) allocated here
                  |  151 |     if (!ring->data) {
                  |      |        ~            
                  |      |        |
                  |      |        (13) assuming ‘<unknown>’ is non-NULL
                  |      |        (14) following ‘false’ branch...
                  |......
                  |  157 |     ring->size = size;
                  |      |     ~~~~~~~~~~~~~~~~~
                  |      |                |
                  |      |                (15) ...to here
                  |  158 |     ring->tail = ring->data;
                  |      |     ~~~~~~~~~~~~~~~~~~~~~~~
                  |      |                |
                  |      |                (16) ‘<unknown>’ leaks here; was allocated at (12)
                  |
[  6%] Building C object src/CMakeFiles/r_433.dir/jsmn.c.o
[  6%] Building C object src/CMakeFiles/r_433.dir/list.c.o
[  7%] Building C object src/CMakeFiles/r_433.dir/logger.c.o
[  7%] Building C object src/CMakeFiles/r_433.dir/mongoose.c.o

zuckschwerdt, 2023-10-29:
It seems GCC 10.2.1 is handling the static analysis well. Just remove this block: https://github.com/merbanan/rtl_433/blob/8535ba24f675afc498dc75ca84e00625aceac06f/CMakeLists.txt#L103-L107

zuckschwerdt, 2023-10-29:
If restricted Static analysis to GCC 13.2+ now, the tested versions 10.2.1, 12.2.1 are too buggy/strict.

closed 2023-10-29 by zuckschwerdt

8um1c4, 2023-10-29:
Procedure follow with success.

8um1c4, 2023-10-29:
Procedure followed with success. Instalation finished and rtl_433 running. Many thanks.

commit e202e85e (2023-10-29): minor: Change Static analysis to GCC13.2+ (closes #2700)