merbanan/rtl_433 #2705: On ARMv71, crashes with first http connection

ksaye, 2023-10-31:
With most recent builds, it crashes without any error at first http:// connect on an ARM device.  I have not seen this behavior on x64

Environment:
* Raspberry Pi3
* OS Alpine 6.1.21-v7
* build: 202310311440
* command line: rtl_433 -Y autolevel -t digital_agc=1 -R 149 -R 154 -F http -F json -M level -f 913M -f 915M -H 900

Experience:

      ~ # uname -a
      Linux ce6947e4714c 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l Linux
      ~ # cat /etc/issue 
      Welcome to Alpine Linux 3.18
      Kernel \r on an \m (\l)
      
      ~ # rtl_433 -Y autolevel -t digital_agc=1 -R 149 -R 154 -F http -F json -M level -f 913M -f 915M -H 900
      rtl_433 version 22.11-243-g8fae9e18 branch master at 202310311440 inputs file rtl_tcp RTL-SDR with TLS
      HTTP server: Starting HTTP server at 0.0.0.0 port 8433
      HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
      
      New defaults active, use "-Y classic -s 250k" if you need the old defaults
      
      Use "-F log" if you want any messages, warnings, and errors in the console.
      Found Rafael Micro R820T/2 tuner
      Exact sample rate is: 1000000.026491 Hz
      Allocating 15 zero-copy buffers
      {"time" : "2023-10-31 11:28:40", "model" : "ERT-SCM", "id" : 65071088, "physical_tamper" : 3, "ert_type" : 5, "encoder_tamper" : 1, "consumption_data" : 4963668, "mic" : "CRC", "mod" : "ASK", "freq" : 913.242, "rssi" : -0.100, "snr" : 11.259, "noise" : -11.360}
      ~ # 

the web UI loads and shows the devices, but goes into disconnect because of the crash:

[image]

If I restart the process, the UI goes back into connected mode and works as expected.  If I close the browser and reopen it, the process crashes again

[image]

My Docker build process is as follows:

      FROM alpine:latest as builder
      
      RUN apk add --no-cache --virtual .buildDeps build-base libusb-dev libressl-dev librtlsdr-dev cmake git
      
      WORKDIR /build
      
      RUN git clone https://github.com/merbanan/rtl_433.git
      WORKDIR /build/rtl_433
      WORKDIR /build/rtl_433/build
      RUN cmake -DENABLE_OPENSSL=ON ..
      RUN make
      WORKDIR /build/root
      WORKDIR /build/rtl_433/build
      RUN make DESTDIR=/build/root/ install
      
      FROM alpine:latest
      
      RUN apk add --no-cache libusb librtlsdr tzdata libressl py3-pip
      
      RUN pip3 install mysql-connector-python
      WORKDIR /root
      COPY --from=builder /build/root/ /

zuckschwerdt, 2023-10-31:
I think this is SIGPIPE from a closed connection reaching the wrong thread.

zuckschwerdt, 2023-10-31:
If you change this line https://github.com/merbanan/rtl_433/blob/master/src/mongoose.c#L3930 to
` int n = (int) MG_SEND_FUNC(nc->sock, buf, len, MSG_NOSIGNAL);`
it should work. I have no idea why the SIGPIPE is not ignored -- we do install handlers but those are not invoked.

zuckschwerdt, 2023-10-31:
This should be fixed now.

closed 2023-10-31 by zuckschwerdt

ksaye, 2023-10-31:
I can confirm that the fix does in fact work.  Thanks @zuckschwerdt

commit 613eb635 (2023-10-31): minor: Fix mongoose SIGPIPE on sockets (closes #2705)