merbanan/rtl_433 #2706: RTL-SDR Blog V4

Mazufa, 2023-11-01:
Hello!

Could someone please confirm if this rtl_433 supports this particular radio dongle: https://www.ebay.com/itm/276000566513

Thanks!

zuckschwerdt, 2023-11-01:
Yes, rtl_433 supports the RTL-SDR Blog V4, but you need an uptodate librtlsdr. See e.g. #2703

closed 2023-11-01 by zuckschwerdt

tasoss, 2023-11-23:
Hello.
I have installed the dragonos (ubuntu)  on a windows machine running vmware workstation.
I am using the latest official drivers from https://github.com/rtlsdrblog/rtl-sdr-blog.
The error i get is

> [Input] Input device start failed, exiting!

Any ideas?
Thanks!

zuckschwerdt, 2023-11-23:
> dragonos (ubuntu) on a windows machine running vmware workstation

Is the USB passthrough properly configured? Check with `lsusb`
Did you properly exclude the DVB drivers? Check with `lsmod`

tasoss, 2023-11-23:
Yes everything is working properly.
I'm using other apps successfully.

zuckschwerdt, 2023-11-23:
The error you see means the device would have been found and started but is not delivering data. Hard to say why.
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1518

tasoss, 2023-11-23:
Ok thanks.
If you need help to debug it somehow please let me know.

zuckschwerdt, 2023-11-23:
I suspect a hardware or emulation problem. Did you test `rtl_sdr` and did you get a sensible cu8 file?

tasoss, 2023-11-23:
I am a total newbie on SDR.
Do you want me to use rtl_sdr and record a frequency ? 
If not could you please explain to me what you are talking about?
BTW i can hear people talking , listening to radio and finding ads-b signals.

zuckschwerdt, 2023-11-23:
Can't really help, sorry. `rtl_sdr` is the most similar program to test against. Both can record to `.cu8` files and should produce similar results. Maybe try an older or newer version of rtl_433. Or if rtl_sdr works the try `rtl_tcp` and use `rtl_433 -d rtl_tcp` to read from that.

tasoss, 2023-11-23:
Ok thank you for your help!