merbanan/rtl_433 #2707: rtl_433 not recognizing Oregon Scientific PCR800 packets

ChoosingUsernameIsDifficult, 2023-11-01:
Running `rtl_433 -R 12 -F json -M utc` rtl_433 successfully sees and decodes packets from my Oregon Scientific THGR810 temp/humidity sensor, and from my WGR800 wind sensor (anemometer). But it fails with my PCR800 rain gauge, while my Oregon Scientific console on my wall receives data from the PCR800 (so the PCR800 is working).

Running on Debian unstable, with rtl_433 version 22.11 (2022-11-19).

I've tried some troubleshooting with rtl_433 and rtl_sdr and also recorded packets using `rtl_433 -S unknown` and `rtl_433 -S known -R 12` to try and see differences, but I'm not getting anywhere.

I do see in the source that PCR800 is supported, and I don't see any bug reports, so what could be causing my problems and how do I troubleshoot it?

I can provide output from commands if anyone wants me to try anything.

zuckschwerdt, 2023-11-02:
Run `rtl_433 -Y autolevel -M level -M noise` to get best results and see the levels.

If that doesn't help then record samples of PCR800. To know you recorded the PCR800 you likely need to get very close and remove the antenna to get exclusive recordings. Check the recordings at https://triq.org/pdv/ and post a few as zip here.

ChoosingUsernameIsDifficult, 2023-11-13:
Sorry for the late reply.

I shipped the DVB-T-dongle to the location where it's intended to be used (400 kilometers from here), it's now connected to a Raspberry Pi 4 running Raspbian 11, version of installed rtl-433 package is 20.11-1.

In this new location I'm seeing packets from two THGR810 sensors as expected. But instead of seeing packets from the WGR800 wind sensor, I'm now getting packets from a PCR800. These sensors are not the same as the ones I tested with at home, but they're the same models.

I'm confused. Perhaps the WGR800 at this new location has bad batteries, I'll check that in a month or so. But I can't explain why I'm seeing PCR800 packets at this new location, when I couldn't see them at home while my Oregon Scientific console received the PCR800 packets just fine.

I've ordered a new DVB-T-dongle and will do more tests here at home when it's set up, to see what packets I can receive.

gdt, 2023-11-26:
Feedback timer started as of November 13th.    This seems so for not to be evidence of a bug, but I'll leave it open for a bit.