merbanan/rtl_433 #2711: Wireless M-Bus length problem

rstephan, 2023-11-06:
Why is there a `-2` is the length calculation?

https://github.com/merbanan/rtl_433/blob/2a7fe211cee2c54ef6d47827324176d49a4e942d/src/devices/m_bus.c#L882

Without the "2" and with also removed "+2" in the line next to it

https://github.com/merbanan/rtl_433/blob/2a7fe211cee2c54ef6d47827324176d49a4e942d/src/devices/m_bus.c#L883

The driver is working with the https://github.com/wmbusmeters/wmbusmeters project no problems at all.

According to the OMS spec, and my meter-data, the "2" is a bug.
Am I missing something?

gdt, 2023-11-26:
If after reading the code history you think there is a bug, you could submit a PR with comments justifying the change.   This isn't really a bug report, so perhaps ask on the list?

gdt, 2024-06-23:
Status?    Does #2749 have all the useful info and we should close this?

rstephan, 2024-07-17:
From my point of view, you have all the data. Do you need more from me?

zuckschwerdt, 2024-07-17:
I'll take a look at this.

Note that `out` is the application layer representation (with checksums removed), while `in` is the link layer data.
That might be the reason to adjust the length indication and possibly you'd want to use the `in` data?

zuckschwerdt, 2024-07-18:
Ok one point down, `out->length+2` is wrong as the the two bytes beyond length are the link-layer CRC.

zuckschwerdt, 2024-07-18:
Other point, the length with `out->data[0]-2)` seems short. The `L` field should count all following bytes (for Telegram type A excluding the CRCs).
The `out->length = block1->L-9 + BLOCK1A_SIZE-2;` covers the true buffer length, `data[0]` should stay the `L` value.

Maybe this was written with just a single data block in mind as `2` would be the length of a trailing CRC, I don't know.

Where is the `data` output used anyway? Do people copy that to https://wmbusmeters.org/analyze/ ? That won't work with the missing CRCs -- it's already half decoded.

I'd like to see `data_length` gone as it's implicit in the `data` output. And even `data` might be superfluous if the decode works. We should rather output the link-layer data (with CRCs), maybe as log message.

merbanan, 2024-07-18:
The telegram data output is used as input for wmbusmeters. The data is supposed to be compliant with IMST 871a (im871a). IIRC I did not agree with how the telegrams where decoded but it was changed to match the output of said dongle.

If this is not what you are referring to I do remember that the parsing was messy and there might still be issues with it.

zuckschwerdt, 2024-07-19:
Thanks, yes that's a good explanation. I'm still confused on how broken that IMST 871a must be.

Is passing input for wmbusmeters the same as using https://wmbusmeters.org/analyze/ ?
The analyze tool seems to disregard the L field, also it can process the true value (without the -2).

And it does parse the trailing CRC bytes as data, which is very likely wrong. We can remove those and get a cleaner output.

The second example "Water meter 2 (not encrypted)" on https://wmbusmeters.org/ show this same `L-2` and trailing CRC decoded as garbage. How strange.

Maybe @weetmuts can help to clear this up?

Another question would be how wmbusmeters detects wether there are CRCs in pasted data or not? It seems to just work in both cases without needing any hint.
Maybe needing IMST 871a formatted data is legacy for wmbusmeters by now?

merbanan, 2024-07-19:
I think wmbusmeter and the analyzer is very relaxed with what you feed it. I think the IMST hardware only pass on real data without the block crc and then at the same time adjusting the L field.

But I think you are correct in that the telegram output is not really IMST "compliant".

weetmuts, 2024-07-19:
The commercial dongles that receive wmbus telegrams all remove the CRC blocs and adjust the length byte, before sending the telegram over the serial wires to the computer. I suppose this makes sense, the big risk of data corruption is in the radioways, not for 255 bytes over a slow serial port, and the CRC bytes are just annoying when decoding the content.

I had to implement removal of  crc bytes for the CUL dongle, which left them in.
The rtlwmbus software always removed the crc bytes, but did not adjust the length byte, which wmbusmeters
had to accept. This is now fixed, so the length byte is corrected after the crc bytes are removed.

So now, when you feed a telegram to wmbusmeters on the commandline (or using the analyze command)
it will check if the crc:s are valid and then remove them. If the crcs:s do not checkout, ie they are not there, or
they are broken. Then no bytes will be removed and decoding proceeds anyway.

For the i871a, amb8465 etc, wmbusmeters will not check/remove any CRC bytes in the incoming data, since they are expected to removed already. This is also the output expected from rtl_433. 

I am not sure the crc-bytes length removal is as easy as -2 since there are more or less crc bytes depending on the length of the telegram and the mode A,B.

weetmuts, 2024-07-19:
Can you explain again for me how the output from rtl_433 differs from the standard: remove crc bytes and adjust length?

zuckschwerdt, 2024-07-19:
> it will check if the crc:s are valid and then remove them.

Smart!

I was only looking at Mode A, perhaps the L-2 adjustment is meant or Mode B, as in Mode A the L fields is excluding any CRC fields, right? I.e.

> Telegram type A - L-Field:
> The first byte of the first block is the length field.
> The field specifies the number of subsequent user bytes, including control and address bytes, but not CRC bytes.
> Telegram type B - L-Field:
> The first byte of the first block is the length field. The field sets the number of all subsequent bytes, including all CRC bytes.

So we might need to check for Mode B before trimming L by 2.

The `data` output from rtl_433 will:
- Remove all but the last CRC, which for Mode A, will look like an erroneous DIF+VIF
- Trim 2 from the L byte, which for Mode A is bogus as quoted above.

weetmuts, 2024-07-19:
If the length byte should be useful at all, then it should represent how many bytes you expect after it. :-)

So, you can always do as rtl_wmbus does it now, just overwrite the length byte with the new length after the crc bytes are removed. Works for both A and B mode.

weetmuts, 2024-07-19:
Now, if you receive encrypted telegrams, the extra bytes might cause wmbusmeters to throw away the telegram. 

For example if AES_CTR encryption is used, there is a higher level checksum in the ELL layer (we have so far talked about DLL layer checksums) for the payload. If this fails, wmbusmeters will always throw away the telegram since such a failed checksum can mean an attack.

Then you have the AES_CBC that will check that the first decrypted block decrypts to 2f2f.... if the telegram is very short, then this might fail.

Then you have security mode 7 with an additional MAC (a more advanced checksum) which of course also will cause wmbusmeters to throw away the telegram if there are too many bytes.

zuckschwerdt, 2024-07-19:
Thanks, I then think we should check what Mode B needs, but for Mode A we should not falsify the L byte (just to be on the safe side) and we should not append the last CRC.

This is seen in your second example "Water meter 2 (not encrypted)" on https://wmbusmeters.org/
It show this same short L-2 and trailing CRC decoded as garbage. Worth to check again I guess.

weetmuts, 2024-07-19:
Thanks for noticing the extra garbage bytes in the example telegram! This is such an old telegram that I do not know how it was collected. There are some other fun reasons for trailing 2 garbage bytes, for example im871a and other dongles can add 2 byte crc at the end to make sure the transfer over the serial line is checked. But this crc is only added if a special bit in the settings is set. In the old days wmbusmeters did not check this bit to know if it should remove the final 2 bytes or not. :-)

If you have a framed text output (like rtl_433 output or rtl_wmbus output or a command line string of hex bytes) then
the length byte is truly irrelevant. You can just check how many bytes you have, no need to care about the length byte.

The length byte is nessary for rtl_433 to be really sure that the telegram checks out when decoding the actual radio waves.

zuckschwerdt, 2024-07-19:
Perfekt! We'll then implement what this issue requested: don't fudge with the L byte in Mode A, and drop the trailing garbage CRC -- wmbusmeters will still work fine!

gdt, 2024-08-04:
Given #2749 merged, is this resolved?

closed 2024-08-04 by zuckschwerdt