merbanan/rtl_433 #2714: Help decoding doorbell Wireless WiFi Tuya

GaetanNandelec, 2023-11-07:
Hello,

I got this wireless doorbell off aliexpress https://www.aliexpress.com/item/1005004524467677.html and i am trying to decode the signal to have event sent to mqtt when the button is pressed. I recorded the following event but i can't get my decoder to work property

[doorbell.zip](https://github.com/merbanan/rtl_433/files/13285178/doorbell.zip)

Could i have some help please ?

zuckschwerdt, 2023-11-07:
Drop a sample on https://triq.org/pdv/ -- in the pulse demod you see a stream of packets, zoom into one packet.
All gaps are the same size -> data is in the pulses. The pulses are of varying width, it's not plain PWM.

PCM will work if we assume a width of 80 µs (set for short and long), set the maximum gap to e.g. 500.
Not sure what the coding here is, but the raw data of `D5 55 B5 EA D5 60` should suffice for a doorbell.

rtl_433's demod will lock onto the 56 µs short pulses though which then adds a `1` bit to each longer pulse section. I'd say this is wrong but the resulting code should be stable and that's all you might need for a doorbell.

I.e. use a decoder of `rtl_433 -R 0 -X 'n=name,m=OOK_PCM,s=60,l=60,g=500,r=2000'`
or to just get this specific doorbell `rtl_433 -R 0 -X 'n=name,m=OOK_PCM,s=60,l=60,g=500,r=2000,match=eaaaeebeaeab80,countonly'`

closed 2023-11-07 by zuckschwerdt

GaetanNandelec, 2023-11-08:
Thank you for the explanation. Can i ask you how do you find the value for the match ?

zuckschwerdt, 2023-11-08:
> how do you find the value for the match ?

Just run the decoder and note the output code.