merbanan/rtl_433 #2715: Revolt-NC5462 has too many false decodes

gdt, 2023-11-08:
```
{"tst":"2023-11-07T15:20:12.431485-0500","topic":"sensor/foo/rtl_433/Revolt-NC5462/0","qos":0,"retain":0,"payloadlen":273,"payload":{"time":"2023-11-07 15:20:11","protocol":239,"model":"Revolt-NC5462","id":0,"voltage_V":0,"current_A":0,"frequency_Hz":0,"power_W":0,"power_factor_VA":0,"energy_kWh":0,"button":0,"mic":"CHECKSUM","mod":"ASK","freq":433.91875,"rssi":-6.30842,"snr":14.98329,"noise":-21.2917}}
```
This has voltage and frequency both 0, and also energy_kWh.    Clearly the checksum is not strong enough.    A suggested fix is to omit if V=0 or f=0, for now, as highly unlikely to remove good data.

zuckschwerdt, 2023-11-08:
Yes, excluding all-zero (or all-one) messages is common in our decoders.

TheButterZone, 2026-09-07:
Just in:

> time      : 2026-09-07 11:29:40
> model     : Revolt-NC5462 House Code: 0
> Voltage   : 0 V          Current   : 327.68 A      Frequency : 0 Hz          Power     : 3276.80 W     Power factor: 0.00 VA     Energy    : 0.00 kWh      Button    : 0             Integrity : CHECKSUM

gdt, 2026-09-07:
@TheButterZone Could you analzye the code and figure out if those values correspond to all 1s in some fields and all 0s elsewhere?  Are they all present in the message, or are some calculated?   0 Hz does not make sense, and Power exceeding VA also does not make sense.

If they are all present, we need  more sanity checks.   But really, I think we need to make this not enabled by default.

TheButterZone, 2026-09-07:
Not without unlimited HD space for cu8 collection until I get a decode for this for the second time ever; [it's been an absolute nightmare to do that even for another model that I kept seeing](https://github.com/merbanan/rtl_433/issues/2384#issuecomment-5564576275) & ["Saving signal to file" doesn't happen](https://github.com/merbanan/rtl_433/issues/2384#issuecomment-5567683409).

gdt, 2026-09-07:
If this driver is only validated on a single decode, it should probably not be default enabled.

TheButterZone, 2026-09-23:
Just got the second one ever, another falsey?

{"time" : "2026-09-23 07:17:28", "model" : "Revolt-NC5462", "id" : 0, "voltage_V" : 0, "current_A" : 0.000, "frequency_Hz" : 0, "power_W" : 819.200, "power_factor_VA" : 0.000, "energy_kWh" : 0.000, "button" : 0, "mic" : "CHECKSUM"}

ProfBoc75, 2026-09-24:
> Just got the second one ever, another falsey?
> 
> {"time" : "2026-09-23 07:17:28", "model" : "Revolt-NC5462", "id" : 0, "voltage_V" : 0, "current_A" : 0.000, "frequency_Hz" : 0, "power_W" : 819.200, "power_factor_VA" : 0.000, "energy_kWh" : 0.000, "button" : 0, "mic" : "CHECKSUM"}

Electrical checks can be done:
- P = V * I * cos(phi)    single phase active power in Watts, W
- power factor = cos(phi)
- power_W = voltage_V * current_A * power_factor
- power_VA = voltage_V * current_A  (apparent power Volt-Amperes, VA)
- power_VAR = voltage_V * current_A * sin(phi)    (reactive power Volt-Amperes Reactive, VAR)
- if power cut, so all zeros values except energy but does the revolt meter even able to transmit something ?

Notice an error in the decoder field power_factor_VA , VA should not be here. Power factor is the result of cos between voltage and current. Value is therefore decimal between -1 and 1. power factor is without unit.

zuckschwerdt, 2026-09-24:
> Notice an error in the decoder field power_factor_VA

And strangely the field is an unsigned byte, scale 100 (0–2.55). I guess it should rather be signed, i.e. -1.28–1.27.

Did anyone read a power factor greater `1.00` with this sensor maybe? @nicolaihess did you spot out-of-range values for the power factor?

ProfBoc75, 2026-09-24:
> And strangely the field is an unsigned byte, scale 100 (0–2.55). I guess it should rather be signed, i.e. -1.28–1.27.
> 
> Did anyone read a power factor greater `1.00` with this sensor maybe? [@nicolaihess](https://github.com/nicolaihess) did you spot out-of-range values for the power factor?

From 2 samples in https://github.com/merbanan/rtl_433_tests/tree/master/tests/revolt_nc_5462  power factor is classic 1 and 0.9

The standard range is between 0.85 ( current and voltage not in phase because of inductive load like ac motors) and 1 (current and voltage same phase when resistive load, like electric heater).
Could be as low as 0.7 for big industrial motors.
Lower values than 0.7 to -1 could be due to electrical production like solar panels, not sure this energy meter is even compatible, as negative means electrical production, positive means consumption.

zuckschwerdt, 2026-09-24:
You are perfectly right, I should have considered that reverse current flow would be needed. I'm just used to seeing negative phi with solar ;) Still the MSB is not needed and might sign…

commit 1e02cdc0 (2023-11-08): minor: Add all-zero sanity check to Revolt-NC5462 (closes #2715)