merbanan/rtl_433 #2718: Fineoffset-WH0290 Produces Duplicate Messages to the same MQTT Topic

cwawak, 2023-11-12:
### Description
The `rtl_433` application sends duplicate messages from a [Fineoffset-WH0290](https://ambientweather.com/pm25-wireless-outdoor-particulate-monitor) sensor to mosquitto, where only a single message is expected.

### Steps to Reproduce
1. Configure `rtl_433.conf` with the following settings:

```
output json
output mqtt://hassio.homenet.wawak.org:1883,user=xxx,pass=xxx,retain=0,events=rtl_433[/model][/id]
report_meta time:utc
output log
```

2. Observe the messages being sent from the Fineoffset-WH0290 sensor.

### Expected Result
A single message should be sent, for example:

```json
{
  "time":"2023-11-12 12:25:10",
  "model":"Fineoffset-WH0290",
  "id":100,
  "battery_ok":0.8,
  "pm2_5_ug_m3":6,
  "estimated_pm10_0_ug_m3":6,
  "family":66,
  "unknown1":0,
  "mic":"CRC"
}
```

## Actual Result
Two messages are being sent, here are the logs:

```
Nov 12 07:25:10 shackzero rtl_433[730]: {"time" : "2023-11-12 12:25:10", "model" : "Fineoffset-WH0290", "id" : 100, "battery_ok" : 0.800, "pm2_5_ug_m3" : 6, "estimated_pm10_0_ug_m3" : 6, "family" : 66, "unknown1" : 0, "mic" : "CRC"}
Nov 12 07:25:10 shackzero rtl_433[730]: {"time" : "2023-11-12 12:25:10", "model" : "Fineoffset-WH0290", "id" : 100, "battery_ok" : 0.800, "pm2_5_ug_m3" : 6, "estimated_pm10_0_ug_m3" : 6, "family" : 66, "unknown1" : 0, "mic" : "CRC"}
```

## Additional Information
Mosquitto detects the second message 0.05 seconds after the first one. The message content doesn't change at all - there is a single `id` in the message. All messages are consumed from the same topic.

This sensor is also known as the Ambient Weather PM2.5 Wireless Outdoor Particulate Monitor.

Environment: 
- Sensor: Fineoffset-WH0290
- Broker: mosquitto
- Application: rtl_433

cwawak, 2023-11-12:
It looks like the device is sending two messages. Should I be filtering these at the mqtt broker level, or should `rtl_433` only send one message? 

I can share the `g001_915M_250k.cu8` file if needed.

```
$ rtl_433 -vvvv -S known
rtl_433 version 22.11-248-g1e02cdc0 branch master at 202311082200 inputs file rtl_tcp RTL-SDR
Reading conf from "rtl_433.conf".

New defaults active, use "-Y classic -s 250k" if you need the old defaults

MQTT: Publishing MQTT data to hassio.homenet.wawak.org port 1883
MQTT: Publishing events info to MQTT topic "rtl_433[/model][/id]".
Protocols: Registered 211 out of 250 device decoding protocols
Input: The internals of input handling changed, read about and report problems on PR #1978
SDR: Found 1 device(s)
SDR: trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
SDR: Sample rate set to 250000 S/s.
Input: Bit detection level set to 0.0 (Auto).
SDR: Tuner gain set to Auto.
Input: Reading samples in async mode...
SDR: rtlsdr_set_center_freq 915000000 = 0
SDR: Tuned to 915.000MHz.
acquire_thread: acquire_thread enter...
Allocating 15 zero-copy buffers
MQTT: MQTT Connected...
MQTT: MQTT Connection established.
Baseband: low pass filter for 250000 Hz at cutoff 50000 Hz, 20.0 us
{"time" : "2023-11-12 12:45:10", "model" : "Fineoffset-WH0290", "id" : 100, "battery_ok" : 0.800, "pm2_5_ug_m3" : 3, "estimated_pm10_0_ug_m3" : 2, "family" : 66, "unknown1" : 0, "mic" : "CRC"}
Pulse data: 49 pulses
[  0] Pulse:    0, Gap:    2, Period:    2
[  1] Pulse:   13, Gap:   16, Period:   29
[  2] Pulse:   14, Gap:   15, Period:   29
[  3] Pulse:   14, Gap:   15, Period:   29
[  4] Pulse:   15, Gap:   14, Period:   29
[  5] Pulse:   15, Gap:   14, Period:   29
[  6] Pulse:   15, Gap:   14, Period:   29
[  7] Pulse:   15, Gap:   14, Period:   29
[  8] Pulse:   15, Gap:   14, Period:   29
[  9] Pulse:   15, Gap:   14, Period:   29
[ 10] Pulse:   15, Gap:   14, Period:   29
[ 11] Pulse:   15, Gap:   14, Period:   29
[ 12] Pulse:   15, Gap:   14, Period:   29
[ 13] Pulse:   15, Gap:   15, Period:   30
[ 14] Pulse:   14, Gap:   15, Period:   29
[ 15] Pulse:   14, Gap:   15, Period:   29
[ 16] Pulse:   14, Gap:   15, Period:   29
[ 17] Pulse:   14, Gap:   15, Period:   29
[ 18] Pulse:   14, Gap:   15, Period:   29
[ 19] Pulse:   14, Gap:   15, Period:   29
[ 20] Pulse:   14, Gap:   15, Period:   29
[ 21] Pulse:   14, Gap:   44, Period:   58
[ 22] Pulse:   15, Gap:   14, Period:   29
[ 23] Pulse:   29, Gap:   14, Period:   43
[ 24] Pulse:   44, Gap:   15, Period:   59
[ 25] Pulse:   14, Gap:   15, Period:   29
[ 26] Pulse:   14, Gap:   44, Period:   58
[ 27] Pulse:   14, Gap:   58, Period:   72
[ 28] Pulse:   15, Gap:   29, Period:   44
[ 29] Pulse:   29, Gap:   29, Period:   58
[ 30] Pulse:   15, Gap:   43, Period:   58
[ 31] Pulse:   15, Gap:  131, Period:  146
[ 32] Pulse:   58, Gap:  174, Period:  232
[ 33] Pulse:   14, Gap:   15, Period:   29
[ 34] Pulse:   15, Gap:   29, Period:   44
[... 54 lines cut ...]
[ 38] Pulse:   43, Gap:   44, Period:   87
[ 39] Pulse:   14, Gap:   22, Period:   36
[ 40] Pulse:    1, Gap:    9, Period:   10
[ 41] Pulse:    2, Gap:    6, Period:    8
[ 42] Pulse:   14, Gap:   45, Period:   59
[ 43] Pulse:    2, Gap:    5, Period:    7
[ 44] Pulse:    4, Gap:    1, Period:    5
[ 45] Pulse:    2, Gap:   21, Period:   23
[ 46] Pulse:    2, Gap:    1, Period:    3
[ 47] Pulse:   12, Gap:   14, Period:   26
[ 48] Pulse:    5, Gap:    1, Period:    6
[ 49] Pulse:    3, Gap:    1, Period:    4
*** Saving signal to file g001_915M_250k.cu8 (44859 samples, 131072 bytes)
^CSignal caught, exiting!
rtl_433: stopping...
sdr_stop: EXITING...
sdr_stop: JOINING...
rtlsdr_read_loop: rtlsdr_read_async done
acquire_thread: acquire_thread async stop...
acquire_thread: acquire_thread done...
sdr_stop: EXITED.
sdr_stop: EXITING...
sdr_stop: Already exiting.
Reattached kernel driver
```

gdt, 2023-11-12:
Currently, this is not a bug, and the behavior is well known.   rtl_433, by doctrine, emits a decode for each over-the-air signal.    See #2640 for an enhancement request to change this.   For now, you can filter them whereever you like, or not.   I tend to think that sending json to an external script and processing there is the most straightforward approach.   see rtl_433_mqtt_relay.py, which can probably filter adjacent dups pretty easily.   If you'd like to ask for help with that, please open a Discussion or use the list.

closed 2023-11-12 by gdt