merbanan/rtl_433 #2719: Add support for Mueller Systems Hot Rod transmitter (water meter)

howardtopher, 2023-11-12:
FCC info: [https://fcc.report/FCC-ID/SM6-HOTRODV2ML](https://fcc.report/FCC-ID/SM6-HOTRODV2ML)
FCC test report: [https://fcc.report/FCC-ID/SM6-HOTRODV2ML/1840648.pdf](https://fcc.report/FCC-ID/SM6-HOTRODV2ML/1840648.pdf)

I've been running rtl_433 for a bit and see almost 40 Neptune R900 meters, but none of them are mine.  After opening up the meter box, it turns out my street has a different brand meter (Hersey Meters) with a Hot Rod v2 transmitter attached.  

Based on the FCC test report, this meter is in 905-925MHz and uses GFSK modulation.  Researching rtl_433 it seems it's not really great at GFSK.  I've had rtl_433 output unknown sample files, but I get a lot of them (about 100 files in 30 seconds of time) and not sure if they are these meters or not.  Since there's at least a whole street of them, there could be a lot of noise here.

I have the ID of my meter and can go out and look at the cubic feet measurement when ever I grab signals.  According to the documentation this transmitter is supposed to transmit every 3 seconds over a range of frequencies so it should hopefully be quite talkative, but not really able to specifically trigger a broadcast.

I'm willing to provide whatever I can, but I may need guidance with the CLI parameters to get rtl_433 to properly see the GFSK signal.

klohner, 2024-01-08:
Is this the same device as discussed in this other issue thread?

[Decoder for Mueller Hot Rod V1 Water Meter Transmitter #2752](https://github.com/merbanan/rtl_433/issues/2752)

howardtopher, 2024-01-08:
Yes, that looks to be the same device.

klohner, 2024-01-09:
Well, to investigate further, we'll need to look at some .cu8 files.

@howardtopher , would you please generate some .cu8 files from your device, zip, and post here?

howardtopher, 2024-01-11:
Even though I have a hotrodv2, I believe it's using the same protocol as the v1 in the other issue.  The first 8 digits in the codes are my meter ID.  I've attached samples from my meter.  I ran the water for a while and only got the consumption number to change one time.  I can do this again if we need more samples.

codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {75}14426616042f1201f80
codes     : {76}1442661604978900fc0
codes     : {75}14426616092f1201f80
codes     : {75}14426614092f1201f80
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {76}1442661604978900fc0
codes     : {75}1442661604978900f80
codes     : {76}1442661604978900fc0
codes     : {77}1442661604978900fc00
codes     : {76}1442661604978900fc0
codes     : {76}1442661604979500570
codes     : {76}1442661604979500570
codes     : {76}1442661604979500570

[hotrod-samples.zip](https://github.com/merbanan/rtl_433/files/13897086/hotrod-samples.zip)

howardtopher, 2024-01-12:
> You got a lot of high-quality codes. Did you use the same rtl_433 command that I did, or something different? How far were you from the transmitter? I'm wondering how I can improve the consistency of my reads.

Yes, I used the same command.  I have my antenna in the attic and it happens to be on the same side of the house as the water meter.  Meter is about 40 ft from the house, in the ground, just under a lid.

ProfBoc75, 2024-03-28:
Hi @howardtopher :

As discussed in the other issue #2752
I started to write the decoder, I would like some more codes / more samples.

Notice that from the cu8 files I analyzed, the pulse is more around 26 µs than 20.
Can you try this:

    rtl_433 -f 909M -R 0 -X 'n=hotrod,m=FSK_PCM,s=26,l=26,r=2500,preamble={24}feb100,bits>=72'

I would like to confirm the data layout at the volume level, with 5, 6 or 7 nibbles after the ID.

Possible [bitbenchs](https://triq.net/bitbench#c=%7B73%7D1259651402670905c30&c=%7B74%7D1259651402670905c30&c=%7B75%7D1259651402670905c30&c=%7B73%7D1259651402671905940&c=%7B74%7D1259651402671905940&c=%7B75%7D1259651402671905940&c=%7B75%7D1259651402672105c50&f=ID%2032h%20VOLUME_GAL%2020h%20FLAGONE%3A4h%20FLAGTWO%3A8h%20CRC%3A8h%208x%0AID%2032h%20VOLUME_GAL%2024h%20_________FLAGTWO%3A8h%20CRC%3A8h%208x%0AID%2032h%20VOLUME_GAL%2028h%20________FLAGTWO%3A%204h%20CRC%3A8h%208x%0A----------------------------------------------------------%0A&z=1&cl=56&cw=4)

Let me know.

I will pull a request soon, starting with volume based on 5 nibbles / 20 bit.

howardtopher, 2024-03-31:
> Can you try this:
> 
> ```
> rtl_433 -f 909M -R 0 -X 'n=hotrod,m=FSK_PCM,s=26,l=26,r=2500,preamble={24}feb100,bits>=72'
> ```
> 
> I would like to confirm the data layout at the volume level, with 5, 6 or 7 nibbles after the ID.
> 

@ProfBoc75 

I've attached a bunch new samples using the command you provided.  Hope these help!

[hotrod-samples-pulse-26.zip](https://github.com/merbanan/rtl_433/files/14817028/hotrod-samples-pulse-26.zip)

ProfBoc75, 2024-03-31:
Thanks for your feedback, sounds like the water meter is based on 7 nibbles and not the 5 nibbles, from your samples, you have only 2 values, and always BCD values (binary coded decimal, i.e. base 10, never hexa values like a, b, c, d, e, or f):

    0512080
    0512228

Each time the 2 last nibbles from the 7 are numbers, never letter from Hexa coding. So, they are BCD values that belongs to the volume of gallons.

I'm waiting samples after @dolai1 will receive a new sdr device.

My decoder is ready, I commented the 7 nibbles decoding, just have to uncomment lines 98, 99 and to comment the 5 nibbles part , lines 88, 89, 90 and test.

ProfBoc75, 2024-04-01:
Yes, I merged my PR, the other issue was automatically closed by the PR.

 Remaining this issue, @howardtopher if the last version of rtl_433 is working properly for you, we can close it.

howardtopher, 2024-04-03:
I'm not sure what's going on here, it's probably something I'm doing wrong.  I've updated to the most recent commit.  I'm not getting anything from the meter.  What's weird is even using the commands from earlier when I was gathering data captures I still am not receiving anything.  I've not changed anything in my setup and can still receive other things with rtl_433.  Surely the battery in the hotrod transmitter didn't just happen to die.  That would be really coincidental.  Any other ideas beyond that why I can't hear it all of a sudden?

ProfBoc75, 2024-04-04:
@howardtopher : could be an antenna issue, may be you moved it.

 I put some verbose information into the decoder, try this:

    rtl_433 -f 909M -R 255:vv -M level

Play with the gain option and try with `-Y minmax` option also.

gdt, 2024-06-03:
Seems like we think this is fixed and are not 100% sure.   I am therefore closing, and if that's wrong please add a request to reopen with rationale.

closed 2024-06-03 by gdt