merbanan/rtl_433 #2720: Different data for Apollo Ultrasonic Oil Tank Sensor

matt5mith, 2023-11-13:
Hello

I have an Apollo Ultrasonic Oil Tank Sensor and was hoping to take advantage of the Oil Smart decoder but I seem to be getting slightly different data which means it doesn't get decoded.  Here is an example of what's coming back:

{158}555558a966955956a6aa55a996a656a966aa9a3c

As per https://github.com/merbanan/rtl_433/discussions/2244, if I run:

rtl_433 -K FILE -R 0 -X 'n=name,m=FSK_PCM,s=500,l=500,g=2000,r=9000,preamble=5558'

... then I get back this format:

{134}a966955956a6aa55a996a656a966aa9a3c

If I add this value into the existing [BitBench](https://triq.net/bitbench#c=%7B130%7D9999996a6559aaa99995a65aa665596a0%20%5BFirst%20test%5D&c=%7B134%7D9999996a6559aaa999969aa6aa9a6995fc%5B0%5D&c=%7B134%7D9999996a6559aaa99996a66aa9a5a59a3c%5B10cm%5D&c=%7B134%7D9999996a6559aaa99996a666a9a56699fc%5B10cm%5D&c=%7B134%7D9999996a6559aaa99996a65aaa66aa563c%5B11cm%5D&c=%7B130%7D9999996a6559aaa99996a65aaa66aa560%5B11cm%5D&c=%7B134%7D9999996a6559aaa99996a65aaa6599aa3c%5B12cm%5D&c=%7B134%7D9999996a6559aaa99996a65aaa5a59a5fc%5B12cm%5D&c=%7B134%7D9999996a6559aaa99996a656aa655aa9fc%5B12cm%5D&c=%7B134%7D9999996a6559aaa99996a656aa59a95a3c%5B13cm%5D&c=%7B134%7D9999996a6559aaa99996a656aa5655563c%5B14cm%5D&c=%7B134%7D9999996a6559aaa99996a5aaaa565a563c%5B14cm%5D&c=%7B134%7D9999996a6559aaa99996a5a6aa55aaa9fc%5B15cm%5D&c=%7B134%7D9999996a6559aaa99996a5a6a9a5a555fc%5B20cm%5D&c=%7B134%7D9999996a6559aaa99996a59aa9a56999fc%5B20cm%5D&c=%7B134%7D9999996a6559aaa99996a596a959a965fc%5B30cm%5D&c=%7B134%7D9999996a6559aaa99996a59aa9596a663c%5B30cm%5D&c=%7B134%7D9999996a6559aaa99996a566a69569963c%5B40cm%5D&c=%7B134%7D9999996a6559aaa99996a566a5aa6a6a3c%5B50cm%5D&c=%7B134%7D9999996a6559aaa99996a55aa5aaa6a63c%5B50cm%5D&c=%7B134%7D9999996a6559aaa99996a55aa5a9955a3c%5B50cm%5D&c=%7B134%7D9999996a6559aaa99996a55696a9a5963c%5B100cm%5D&c=%7B134%7D9999996a6559aaa99996a55696a6599a3c%5B100cm%5D&c=%7B134%7D9999996a6559aaa99996a556aaaaaa59fc%5Berr%5D&c=%7B134%7D9999996a6559aaa99996a55669965aa9fc%5Berr%5D&c=%7B134%7D9999996a6559aaa99996a5566999a6a5fc%5Berr%5D&c=%7B134%7D9999996a6559aaa99996a556665aaaa9fc%5Berr%5D&c=%7B134%7D9999996a6559aaa999969aaaaaaa9695fc%5Berr%5D&f=PRE%3F8h8h%20ID%3F8h8h%208h%20CTR%3F8d%20CM%3A8d%20CRC8h&d=MCI&cw=4) then it appears to decode to 26cm properly.

So, it appears that the existing decoder will work, but for whatever reason, it's not being used.

Any help or guidance would be hugely appreciated.  The end goal is to get the data into Home Assistant via MQTT, but I'm quite comfortable with that side of things.  It's just the decoding which I'm struggling with.

zuckschwerdt, 2023-11-13:
After decoding we expect the first bytes (perhaps a model code) to be `55 58`. In your sample it's `e582`.
You can disable (comment) this line https://github.com/merbanan/rtl_433/blob/master/src/devices/oil_smart.c#L71

It would be great if you can watch the codes you get and report maybe after a week a list.
Run with `-R 235:v` to see raw codes.

zuckschwerdt, 2023-11-13:
Or to get the "model number" along the normal report insert these lines below line 98:
```
            "unknown_0",        "Unknown 0",    DATA_FORMAT, "%02x", DATA_INT, (b[0] << 8) | b[1],
```

matt5mith, 2023-11-13:
Thank you @zuckschwerdt.  I will report back my findings.

matt5mith, 2023-11-13:
Here is the result of doing both suggestions together:

[image]

klohner, 2024-01-08:
Here's some info that may be useful:

[Previous issue thread regarding this device](https://github.com/merbanan/rtl_433/discussions/2244) (started by @StarMonkey1)

[BitBench](https://triq.net/bitbench#c=%7B158%7D555558a966955956a6aa55a996a656a966aa9a3c%20%5Bmatt5mith%202023-11-23%5D&c=%7B158%7D555558a966955956a6aa55a996a566a96659a5fc%20%5Bmatt5mith%202023-11-23%2017%3A17%3A27%5D&c=%7B158%7D5555589999996a6559aaa99996a55696a9a5963c%20%5Boil_smart.c%20example%2001%20(raw)%5D&c=%7B158%7D5555589999996a6559aaa999969aa6aa9a6995fc%20%5Boil_smart.c%20example%2002%20(aligned)%5D&f=PREAMBLE%3F%3A16h%20SENSOR_ID%3A16h%20FLAGS%3F%3A8b%20COUNTER%3F%3A8d%20DEPTH_CM%3A8d%20CRC%3Ahh&a=Preamble&m=%7B16%7D5558&d=MCI&cw=4) (including examples in oil_smart.c)

CRC confirmation:
```
reveng -w 8 -s 1a7de20f162e1a04 1a7de20f163a1ad3 5558bd01563e6136 5558bd0156420497
width=8  poly=0x31  init=0x00  refin=true  refout=true  xorout=0x00  check=0xa1  residue=0x00  name="CRC-8/MAXIM-DOW"
```

Flex decoder that doesn't enforce the `0x5558` "Preamble" like oil_smart.c does:
`rtl_433 -R 0 -X "n=OIL,m=FSK_PCM,s=500,l=500,g=2000,r=9000,bits=158,preamble={16}5558,symbol_one={2}4,symbol_zero={2}8,get=PREAMBLE?:@0:{16}:%04x,get=ID:@16:{16}:%04x,get=FLAGS?:@32:{8}:%02x,get=COUNTER?:@40:{8}:,get=DEPTH_CM:@48:{8}:,get=CRC:@56:{8}:%02x" -y {158}555558a966955956a6aa55a996a656a966aa9a3c`

or suitable for rtl_433.conf:

```
decoder {
    name=Apollo_Ultrasonic_Oil,
    modulation=FSK_PCM,
    short=500,
    long=500,
    gap=2000,
    reset=9000,
    bits=158,
    preamble={16}5558,
    symbol_one={2}4,
    symbol_zero={2}8,
    get=PREAMBLE?:@0:{16}:%04x,
    get=ID:@16:{16}:%04x,
    get=FLAGS?:@32:{8}:%02x,
    get=COUNTER?:@40:{8}:,
    get=DEPTH_CM:@48:{8}:,
    get=CRC:@56:{8}:%02x
}

```

SalimSwift, 2025-01-07:
@klohner Thanks for the .conf file, the oil_smart.c doesnt see/decode the transmitter data but your .conf file works. Would you know how I can implement this into the oil_smart.c ? Regards

gdt, 2025-07-04:
@matt5mith You've been provided a flex decoder conf.  Please test it, and if it is ok, make a PR to store it in the sources.