merbanan/rtl_433 #2721: Specify git version manually to cmake

hertzg, 2023-11-15:
Hello,

When trying to build the latest master (for example) withot the `.git` folder present, the CMakeList defaults to extracting the version from changelog.md, which results in compilted binaries outputing the wrong verson when the `-v` flag is passed.

How do you set the GIT_VERSION, BRANCH and required cmake definitions **manually** during the build process to avoid this situation withot having to keep the .git folder?

zuckschwerdt, 2023-11-15:
The usual way will produce collision warnings:
`mkdir build ; cd build ; CFLAGS="-DGIT_VERSION=1.2.3 -DGIT_BRANCH=foo -DGIT_TIMESTAMP=20230101" cmake -GNinja .. && cmake --build . -j 10`

I think we need to add those as proper options to support automation.

hertzg, 2023-11-15:
I can confirm that "the usual way" with `CFLAGS` works and does build a binary with correct output when `-V` flag is passed during runtime.

```
(base) ➜  build git:(master) ./src/rtl_433 -V
rtl_433 version 99.99.99 branch my-branch-name at 22221222 inputs file rtl_tcp RTL-SDR with TLS
```

But, like you said during build a bunch of `macro redefined [-Wmacro-redefined]` warnings are produced, I tried by deleting `changelog.md` file to avoid re-definition of those `#define` and it seems to worked, not sure if that's a correct way to go though 🤔

zuckschwerdt, 2023-11-15:
If anyone has suggestions how to integrate this in a cleaner way please tell :) Otherwise we'll close this as "good enough" in a few days.

(I said "usual" because to apply defines in C you'd use CFLAGS traditionally, e.g. `CFLAGS="-DFOO" make`. Not commonly used with cmake though.)

hertzg, 2023-11-15:
I'll check how this would work with code from older tags (eg: [0.1 from 2016](https://github.com/merbanan/rtl_433/releases/tag/0.1)) and report back if the method is really "good enough for now".

Let me know if there's gonna be a way to manually set it in the future and I would happily migrate to it.

zuckschwerdt, 2023-11-15:
`18.12` would be the oldest Version with something like this. I.e. 18.05 and 0.1 dont support this.

gdt, 2023-11-26:
About older tags: generally, projects fix the current version, and fixing old versions is basically out of scope.

@hertzg It sounds like you have arrived at instructions for someone who wants a git version number but for some reason wants to delete .git.  To me that's odd, but a very concise hint added to build instructions as a PR could be ok.

hertzg, 2023-11-27:
> About older tags: generally, projects fix the current version, and fixing old versions is basically out of scope.
I'm not asking to fix old versions, since I'm maintianing docker builds for rtl_433, I was thinking if this workaround would be "backwards compatible".

> It sounds like you have arrived at instructions for someone who wants a git version number but for some reason wants to delete .git. To me that's odd, but a very concise hint added to build instructions as a PR could be ok.

When cloning the repo normally .git would contian all blobs, therefore that layer that cloned the repo will have a different hash every time there's a new commit to the repo, no matter if you only need the first ever commit or the latest.

This was one avenue I was exploring but it's a dead end without separating binary building and image packaging...

https://github.com/hertzg/rtl_433_docker/issues/69#issuecomment-1666492451

gdt, 2024-05-30:
Looks like we've concluded, and the other bug is closed and we're still in feedback state, so I'm going to assume that people that care have a workaround for both deleting git and wanting it.  If that's off please explain why and we can reopen.

closed 2024-05-30 by gdt

hertzg, 2024-06-02:
Yes, this was resolved, but I opted in for leaving the .git folder instead fiddling with cmake and CFLAGS to try to pass those values as they would be passed by git module for cmake.
 
https://github.com/hertzg/rtl_433_docker/issues/69#issuecomment-2143923371