merbanan/rtl_433 #2724: Abode/AbodeRF 433Mhz sensors support (possible Climax rebrand)

kiler129, 2023-11-21:
### Background
[Abode](https://goabode.com) used to be a great platform to build alarm systems that work offline, with decently priced hardware. They also offered a fair semi-managed option. In late 2022 the company decided to pull a rug on users by [making all notifications useless](https://www.reddit.com/r/Abode/comments/y22yh3/getting_generic_notifications_all_of_a_sudden/), in addition to [paywalling previously free](https://www.reddit.com/r/Abode/comments/12l2pqf/notification_text_generic_and_confusing_no/) detection details, effectively [making self-monitoring next to useless](https://www.reddit.com/r/Abode/comments/117670a/theres_no_such_thing_as_selfmonitoring_here/). Naturally [users got pissed](https://www.reddit.com/r/Abode/comments/y2cpvv/f_abode_they_are_paywalling_their_customers_now/) but [nothing has changed](https://www.reddit.com/r/Abode/comments/124lel4/abode_review_stay_away/). While they couldn't kill HomeKit, probably due to Apple certification, they [ensured it's nerfed to basics](https://www.reddit.com/r/Abode/comments/y944wr/any_way_to_get_abode_cam_2_in_homekit_or_google/). Building into the system also isn't possible as [alarm can only be triggered by official sensors](https://www.reddit.com/r/Abode/comments/j2zzdn/only_official_sensors_can_trigger_alarm/).

BUT... the sensors are pretty decent, small, and many people in the US have a lot of them installed. Thus, the only sensible option is DIYing the alarm and getting rid of Abode gateway :)

### Are sensors reusable?
There's very little information about that. There are [some posts](https://www.reddit.com/r/Abode/comments/set6rb/what_can_i_switch_to_and_reuse_devices/) asking about reusing sensors but nothing more than that. Abode itself is pretty tight-lipped about how each component connects. While the gateway supports Zigbee and Z-Wave, all sensors appear to be 433Mhz ones using """proprietary AbodeRF""".

Digging through FCC database I found that Abode Systems has very little in it. Beyond WiFi products, the only thing is an [alarm keypad](https://fcc.report/FCC-ID/2ARGFKP2/) that runs on 433Mhz. Then, I found [a Reddit post](https://www.reddit.com/r/Abode/comments/jm81mp/abode_is_actually_climax_why_doesnt_it_do_more/) suggesting Abode is a Climax rebrand + a nice coat of paint. The FCC search brought me to the same [line of contact sensors](https://fccid.io/GX9MDC3R2/External-Photos/External-Photos-6637367) from Climax *(minus Abode labels)*. Looking at [the RF report](https://fccid.io/GX9MDC3R2/Test-Report/Test-Report-6637358) these indeed run at 433Mhz.

### Reusing sensors
I looked at the [contributing guidelines](https://github.com/merbanan/rtl_433/blob/master/docs/CONTRIBUTING.md), but it seems a bit over my head to write a decoder. Did anyone maybe tried to touch these sensors? They offer pairing and allegedly "jamming protection" ([which was allegedly proven false](https://www.reddit.com/r/Abode/comments/t5y8fh/local_news_piece_on_overcoming_abode_system_via/)), they don't seem to send any periodic messages nor offer battery monitoring. Just a simple open/close status broadcast.

If nobody tried, would posting a capture itself of pairing and repeated open/close help decode and possibly support these sensors?

zuckschwerdt, 2023-11-21:
Yes, sample files would be great and a quick way to see what the protocol is like.

The [FCC internal photos](https://fccid.io/GX9MDC3R2/Internal-Photos/Internal-Photos-6637365) show two QFP chips (U2, U4) -- can you maybe open one sender and identify the markings on those?

jwjenkin, 2024-01-06:
I have a ton of these, would be phenomenal to use them without the alarm necessarily

klohner, 2024-01-08:
@kiler129 & @jwjenkin Would you be able to provide several .cu8 files of these devices with descriptions of the device and conditions (open/close) that generated the signal?  See "Supporting Additional Devices and Test Data" in [CONTRIBUTING](https://github.com/merbanan/rtl_433/blob/master/docs/CONTRIBUTING.md) for info.

Essentially, `rtl_433 -S unknown` should get you started (assuming 433 MHz) and if you want to be more discriminating and provide quality samples, maybe disconnect the antenna on your RTL and hold the Abode/Climax device a few centimeters from you RTL device to get those samples.

kiler129, 2024-02-12:
@klohner Sorry for a huge delay - life got away a bit, with my SDR getting killed in the process and being replaced with genuine RTL-SDRv3 :)

I collected a few samples from the device *(`Abode Mini Door/Window Sensor`, [specs from manufacturer](https://support.goabode.com/docs/mini-doorwindow-sensor))*. Let me know if I can collect any other samples. I described the current ones below.

@zuckschwerdt As for the electronics, I took high-res pictures of both sides of the board, with an additional macro shot of the U8 and its pesky laser markings:
[image]
[image]
[image]

----

I think the table with summary will be the best. All sample files are attached ([samples.tgz](https://github.com/merbanan/rtl_433/files/14235799/samples.tgz)):
- `Smp#` represents file number e.g. `1` is `g001_433.92M_250k.cu8`
- `Action` is what I physically did to the sensor
- `Door`/`Tamper`/`Pair` columns represent state of a button/door reed switch when the action was performed
  - `-` was put in a column when action was performed on a given input
  - `?` designates unknown state (i.e. where I cannot be 100% certain what was the state)
  - `0` designates open circuit/not pressed/not triggered by magnet; while `1` represents the opposite state
- All samples collected, as recommended, without antenna attached with the receiver's SMA ~1-2mm from the sensor's antenna
 
| Smp#  | Action         | Door | Tamper | Pair |
|-------|----------------|------|--------|------|
| 1     | Press tamper   | 0    | -      | 0    |
| 2     | Release tamper | 0    | -      | 0    |
| 3     | Press tamper   | 0    | -      | 0    |
| 4     | Release tamper | 0    | -      | 0    |
| 5-8   | Press pair     | 0    | 0      | -    |
| 9-12  | Press pair     | 0    | 1      | -    |
| 13    | Unknown[1]     | 0    | ?      | 0    |
| 14    | Press tamper   | 1    | -      | 0    |
| 15    | Release tamper | 1    | -      | 0    |
| 16    | Press tamper   | 1    | -      | 0    |
| 17    | Release tamper | 1    | -      | 0    |
| 18-21 | Press pair     | 1    | 0      | -    |
| 22-25 | Press pair     | 1    | 1      | -    |
| 26    | Close door     | -    | 1      | 0    |
| 27    | Open door      | -    | 1      | 0    |
| 28    | Close door     | -    | 1      | 0    |
| 29    | Open door      | -    | 1      | 0    |
| 30    | Close door     | -    | 1      | 0    |
| 31    | Unknown[2]     | 1    | 1      | 0    |
| 32    | Open door      | -    | 1      | 0    |
| 33    | Close door     | -    | 1      | 0    |
| 34    | Open door      | -    | 1      | 0    |
| 35-38 | Unknown[3]     | ?    | ?      | ?    |
| 39    | Close door     | -    | 0      | 0    |
| 40    | Open door      | -    | 0      | 0    |
| 41    | Close door     | -    | 0      | 0    |
| 42    | Open door      | -    | 0      | 0    |
| 43    | Close door     | -    | 0      | 0    |
| 44    | Open door      | -    | 0      | 0    |
| 45    | Close door     | -    | 0      | 0    |
| 46    | Open door      | -    | 0      | 0    | 

References for "unknown" from table:
1. Spurious message when sensor was on the table
2. Message sent without state change when door was closed for a few seconds; doesn't happen all the time
3. Sensor flew away when I was taking notes :D Attached for completeness of series in case there's some counter in the packets

----

Full capture log from `rtl_433`:
```
# rtl_433 -S unknown
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[... 41 lines cut ...]
*** Saving signal to file g027_433.92M_250k.cu8 (572059 samples, 1179648 bytes)
*** Saving signal to file g028_433.92M_250k.cu8 (564179 samples, 1179648 bytes)
*** Saving signal to file g029_433.92M_250k.cu8 (564097 samples, 1179648 bytes)
*** Saving signal to file g030_433.92M_250k.cu8 (564157 samples, 1179648 bytes)
*** Saving signal to file g031_433.92M_250k.cu8 (270793 samples, 655360 bytes)
*** Saving signal to file g032_433.92M_250k.cu8 (595562 samples, 1310720 bytes)
*** Saving signal to file g033_433.92M_250k.cu8 (572024 samples, 1179648 bytes)
*** Saving signal to file g034_433.92M_250k.cu8 (587744 samples, 1179648 bytes)

*** Saving signal to file g035_433.92M_250k.cu8 (619221 samples, 1310720 bytes)
*** Saving signal to file g036_433.92M_250k.cu8 (627110 samples, 1310720 bytes)
*** Saving signal to file g037_433.92M_250k.cu8 (587847 samples, 1179648 bytes)
*** Saving signal to file g038_433.92M_250k.cu8 (619279 samples, 1310720 bytes)

*** Saving signal to file g039_433.92M_250k.cu8 (587769 samples, 1179648 bytes)
*** Saving signal to file g040_433.92M_250k.cu8 (595709 samples, 1310720 bytes)
*** Saving signal to file g041_433.92M_250k.cu8 (587796 samples, 1179648 bytes)
*** Saving signal to file g042_433.92M_250k.cu8 (564162 samples, 1179648 bytes)
*** Saving signal to file g043_433.92M_250k.cu8 (564184 samples, 1179648 bytes)
*** Saving signal to file g044_433.92M_250k.cu8 (587745 samples, 1179648 bytes)
*** Saving signal to file g045_433.92M_250k.cu8 (611379 samples, 1310720 bytes)
*** Saving signal to file g046_433.92M_250k.cu8 (548506 samples, 1179648 bytes)
^CSignal caught, exiting!
Reattached kernel driver
```

zuckschwerdt, 2024-02-12:
That is a very nice and detailed analysis! Re chips: the bigger one is a [TI MSP430G2553](https://www.ti.com/product/de-de/MSP430G2553) MCU -- the sensor seems capable and well designed, the idle messages, perhaps with battery info a great and not found in cheaper ones.
No idea what that "119B D904 125" part is.

zuckschwerdt, 2024-02-12:
The signals are clipping (too loud) but it looks like FSK with 100 µs PCM. This will get data:
`rtl_433 -Y minmax -R 0 -X 'n=name,m=FSK_PCM,s=100,l=100,r=4000'`

There is a consistent prefix of `aaaaaaaa29cd29cd0a0000` but the payload seems encrypted.
Have a look through all the codes to see if any correlation pops out.

<details>
<summary><b>Raw codes</b></summary>
<p>

```ini
[g001_433.92M_250k.cu8]
{173}d555555514e694e685000009287de27bd6ceb9cadb80[@0.1s]
{172}aaaaaaaa29cd29cd0a0000311eaeefd3a6698391360[@0.6s]
{173}d555555514e694e6850000086fdd1a065756398d4c00[@0.9s]
{172}aaaaaaaa29cd29cd0a00003cfff053ef58ca8fbec10[@1.3s]
{172}aaaaaaaa29cd29cd0a00002980db3426f0acbca6a40[@1.7s]
{172}aaaaaaaa29cd29cd0a000017a72c0d8de6272e24f50[@2.1s]
[g002_433.92M_250k.cu8]
{171}aaaaaaaa29cd29cd0a00001642448bdeb594f5d3840[@0.2s]
{172}aaaaaaaa29cd29cd0a000027c5a21efb8c8dffd6030[@0.7s]
{172}aaaaaaaa29cd29cd0a000028b0a2c2f2b0b5ffd4e20[@1.1s]
{173}d555555514e694e685000014bb5177720034fffd3700[@1.5s]
{172}aaaaaaaa29cd29cd0a00002aa8319d34afe0fff9770[@2.0s]
{172}aaaaaaaa29cd29cd0a00002b9c3121682b22ff5ed60[@2.4s]
[g003_433.92M_250k.cu8]
{173}d555555514e694e685000004d2912e471e4f06d5b580[@0.1s]
{172}aaaaaaaa29cd29cd0a000019cd4c503c06ac2d50790[@0.6s]
{173}d555555514e694e685000016a735a2083b0e8092c880[@0.9s]
{173}d555555514e694e68500000fa79028066d46b989ad00[@1.3s]
{172}aaaaaaaa29cd29cd0a0000132e2c45f7fbc67cf6ed0[@1.7s]
{173}d555555514e694e6850000045991670e3b3486a46000[@2.1s]
[g004_433.92M_250k.cu8]
{171}aaaaaaaa29cd29cd0a0000164bf72bb8b38d1b4da20[@0.1s]
{172}aaaaaaaa29cd29cd0a00001ed9db1e6f2b9d2b990c0[@0.6s]
{172}aaaaaaaa29cd29cd0a000032e3b65c8e8cac2c9c010[@0.9s]
{173}d555555514e694e68500000bd0c92230cfc6bd321900[@1.3s]
{172}aaaaaaaa29cd29cd0a00001bb44d45c027227a67b90[@1.7s]
{172}aaaaaaaa29cd29cd0a00000f692c7d5b27470b072b0[@2.1s]
[g005_433.92M_250k.cu8]
{173}d555555514e694e685000007144de9e05f6ac759c500[@0.1s]
{173}d555555514e694e685000013e8b5e9b693ea965c8b80[@0.4s]
{173}d555555514e694e685000007144de9e05f6ac759c500[@0.8s]
[g006_433.92M_250k.cu8]
{173}d555555514e694e68500001114d729b1cfd58637f280[@0.2s]
{173}d555555514e694e68500001bc1e5a86061d4391a2900[@0.6s]
{173}d555555514e694e68500000612cdc936e3f56db0ca00[@1.1s]
[g007_433.92M_250k.cu8]
{172}aaaaaaaa29cd29cd0a0000237d6b95bd399d8e32aa0[@0.2s]
{173}d555555514e694e68500000fae57611a44ce47357280[@0.6s]
{173}d555555514e694e6850000131b11284713bb06192000[@1.1s]
[g008_433.92M_250k.cu8]
{173}d555555514e694e6850000199ce5c38e0034066f8900[@0.1s]
{173}d555555514e694e68500000ceb510869df7acd4a6900[@0.6s]
{173}d555555514e694e685000017830f3eaf9cb28601d700[@1.1s]
[g009_433.92M_250k.cu8]
{172}aaaaaaaa29cd29cd0a00001c78b60d6dc1df5065650[@0.2s]
{173}d555555514e694e68500001aee7286b096efce562880[@0.6s]
{173}d555555514e694e685000000bc2664b69cd4638d9780[@1.1s]
[g010_433.92M_250k.cu8]
{172}aaaaaaaa29cd29cd0a00001c80f050dc2d9d93ceb50[@0.2s]
{173}d555555514e694e68500000de5db49476d0e7e503c80[@0.7s]
{172}aaaaaaaa29cd29cd0a000015947cc2e3b921fc5e460[@1.1s]
[g011_433.92M_250k.cu8]
{172}aaaaaaaa29cd29cd0a000029842787d3b9d593a1640[@0.1s]
{173}d555555514e694e6850000057d98c97bbcfaeaf0e280[@0.4s]
{173}d555555514e694e6850000131793f75eeb4e49c58a80[@0.8s]
[g012_433.92M_250k.cu8]
{172}aaaaaaaa29cd29cd0a00002d5ce57dc00614931e860[@0.2s]
{172}aaaaaaaa29cd29cd0a00001eebcbceef2b9c8588650[@0.6s]
[... 466 lines cut ...]
[g044_433.92M_250k.cu8]
aaaaaaaa29cd29cd0a000020676e928df29ad2dc330[@0.0s]
aaaaaaaa29cd29cd0a000016eca387c0d866988e430[@0.4s]
aaaaaaaa29cd29cd0a00001bf11a34105a9a980c070[@0.9s]
aaaaaaaa29cd29cd0a00002d87ae5c8e00930046060[@1.4s]
aaaaaaaa29cd29cd0a000024de2250f22f8ab14ff90[@1.8s]
aaaaaaaa29cd29cd0a00003acd4d0d61afe8adb44e0[@2.1s]
[g045_433.92M_250k.cu8]
aaaaaaaa29cd29cd0a0000136fe53dd32f9a56889e0[@0.2s]
aaaaaaaa29cd29cd0a00002536ba215bc393ca014e0[@0.7s]
aaaaaaaa29cd29cd0a00001bf9b1ee5fa729fc81fc0[@1.1s]
aaaaaaaa29cd29cd0a00002c902c50e3fa18a9e37e0[@1.6s]
aaaaaaaa29cd29cd0a000014fda3c46d0844fc87fc0[@1.9s]
aaaaaaaa29cd29cd0a0000266afb4526f0b2ca61cb0[@2.4s]
[g046_433.92M_250k.cu8]
aaaaaaaa29cd29cd0a0000252cfb10f23e4498a1a60[@0.2s]
aaaaaaaa29cd29cd0a00002ae72c7df77be89844f90[@0.6s]
aaaaaaaa29cd29cd0a00001186e544722fc74d00560[@1.0s]
aaaaaaaa29cd29cd0a00001d6a7f5363c7930434500[@1.4s]
aaaaaaaa29cd29cd0a00001285a3341cbe744d4ce10[@1.8s]
aaaaaaaa29cd29cd0a00002704aec9349f77984ff40[@2.1s]
```

</p>
</details>

ProfBoc75, 2024-02-12:
@zuckschwerdt : I found something interesting [here](https://github.com/AxlLind/master-thesis/blob/dcef6c577b44c5d4488dd2cb7b6cce37b82f29b6/chapters/pentesting-tasks/2-rf-reverse-engineering.tex#L83)

In their analysis, the code is encrypted but it can be replayed and it will work. I also notice that the payload part is always starting by niblle 0 , 1 or 2.

zuckschwerdt, 2024-02-12:
That is indeed a very nice find! I didn't spot the sync-word but it's obvious now, `29cd`.
In summary `aaaa aaaa 29cd 29cd 0a00 00 …` -- so we know the alignment is correct.

ProfBoc75, 2024-02-12:
The last 0 should be ignore.

Data layout : 8 byte + 2 byte CRC-16 , poly=0x8005, init=0x8e8b, for 94% from the 222 samples

    reveng-3.0.5$ ./reveng -w 16 -s 1250fbc4f7ad9d7395b7 311eaeefd3a669839136 10dfba340caeac731a98 3cfff053ef58ca8fbec1 2980db3426f0acbca6a4
    width=16  poly=0x8005  init=0x8e8b  refin=false  refout=false  xorout=0x0000  check=0xdbab  residue=0x0000  name=(none)
    
    reveng -w 16 -s 2aa8319d34afe0fff977 2b9c3121682b22ff5ed6 09a5225c8e3c9e0dab6b 19cd4c503c06ac2d5079 2d4e6b4410761d012591
    width=16  poly=0x8005  init=0x8e8b  refin=false  refout=false  xorout=0x0000  check=0xdbab  residue=0x0000  name=(none)

or poly=0x8005, init=0x9d13, for the rest of 6% from the samples

    reveng -w 16 -s 07dad785a6895350aad8 14139e88de743793d38e 1642448bdeb594f5d384 164bf72bb8b38d1b4da2 1685447a680a34a942f4 187747784d6a8e168b04 18916ddcc7b0cd096d1c 1ca324fbf76c532d206a 33d7f6695ebfdb6be3b8 3ee862fade5ba62156a4 497698a7dfdd5111379e 58ed9724b667bdac7712 5e309a891c6bebf853bc 78629985f667159be3fc
    width=16  poly=0x8005  init=0x9d13  refin=false  refout=false  xorout=0x0000  check=0xb46e  residue=0x0000  name=(none)

2 CRC formulas, strange or false positive or ?

kiler129, 2024-02-12:
> (...) the sensor seems capable and well designed, the idle messages, perhaps with battery info a great and not found in cheaper ones.

Indeed. If supported, I would actually look for OEM version of these as these sensors are legitimately good and do last for long on a battery.

> The signals are clipping (too loud) but it looks like FSK with 100 µs PCM. This will get data: `rtl_433 -Y minmax -R 0 -X 'n=name,m=FSK_PCM,s=100,l=100,r=4000'`

Would you like me to re-record some with greater distance?

> I found something interesting [here](https://github.com/AxlLind/master-thesis/blob/dcef6c577b44c5d4488dd2cb7b6cce37b82f29b6/chapters/pentesting-tasks/2-rf-reverse-engineering.tex#L83)

@ProfBoc75 the encryption would make sense, as they have the whole pairing song-and-dance. In addition, they advertise them as not being able to be spoofed (I know I know :D). I also have their motion sensor and quite a few door sensors, so maybe grabbing recordings from at least another door sensor would aid comparison?

Also, this slide is interesting from the thesis:
[image]

Maybe @AxlLind himself could chime in with more details? :)

zuckschwerdt, 2024-02-12:
Re that picture: we know that synwords are very commonly 16-bit and repeated once. Also we can sensibly expect a length byte. This is more likely:
```
Preamble Sync-Word Len Type? Payload (10 bytes)   Demod-Junk
aaaaaaaa 29cd 29cd 0a  0000  311eaeefd3a669839136 0
```
And even more likely with @ProfBoc75's finding of the checksum :
```
Preamble Sync-Word Len Payload (10 bytes)   Checksum Demod-Junk
aaaaaaaa 29cd 29cd 0a  0000311eaeefd3a66983 9136     0
```

And then using that assumption and computing the CRC-16 starting with the `0a` length byte we get a very nice poly=0x8005 init=0xffff.

gdt, 2024-06-03:
Any status?   This seems like a start but not close to landing something.  Is it still useful to have an issue?

kiler129, 2024-06-04:
@gdt I played a bit with it and wasn't able to get anywhere. However, I'm not very good at RE protocols like that. Maybe if we had a dump of the gateway or at least one of the sensors.....

I didn't try, but I can almost bet the chip is read-protected and I'm far away from decaping for the mere sensors ;)

gdt, 2024-06-04:
Does it make sense for you to make a PR which adds a decoder which just has the hex of the encrypted bits?  That would let more people gather data, merged or not, and move this along from issue.  It seems there is good info here so I'm not thrilled about just closing this, but OTOH it's hard to make a case that anything useful will happen.  If you think it's not worth the time to make a PR with the preamble/crc, then that feels like a clue (not saying you are wrong - just that it's a clue).

klohner, 2024-06-04:
FWIW, this is the flex decoder I put together to get the encrypted data:

```
decoder {
    name=AbodeRF,
    modulation=FSK_PCM,
    short=100,
    long=100,
    reset=2200,
    bits>=170,
    bits<=180,
    match={64}aaaaaaaa29cd29cd,
    preamble={64}aaaaaaaa29cd29cd,
    get=LEN:@0:{8}:%d,
    get=ZEROS:@8:{16}:%04x,
    get=ENCRYPTED_1:@24:{32}:%08x,
    get=ENCRYPTED_2:@56:{32}:%08x,
    get=CRC-16_8005_CMS:@88:{16}:%04x
}
```

`$  rtl_433 -Y minmax -r g006_433.92M_250k.cu8`
[image]

```
$ reveng -w 16 -s 0a00000c259b926dc7eadb61940 0a00003783cb50c0c3a87234520 0a00002229ae53639fab0c6fe50
width=16  poly=0x8005  init=0xffff  refin=false  refout=false  xorout=0x0000  check=0xaee7  residue=0x0000  name="CRC-16/CMS"
```

gdt, 2024-06-05:
PR to add, and we can close?

gdt, 2024-06-05:
Closing then.  A PR to add a flex decoder, or even more progress, can happen regardless of this being closed - there's no requirement for an issue.

closed 2024-06-05 by gdt

Wetzel402, 2026-03-13:
Are these sensors supported now?

zuckschwerdt, 2026-03-14:
You can use the flex decoder in a conf file