merbanan/rtl_433 #2732: WS90 with new firmware not decoded

sombrero2003, 2023-11-28:
Hello,
I just upgraded my WS90 to the latest firmware (1.3.8). I recompiled rtl_433 to take care of issue #2702 but the sensor does not get decoded using
`rtl_433 -R 244 -f 915M`

Here is some more verbose output via `rtl_433 -R 244:vv -f 915M`

```
[fineoffset_ws90_decode] abort length
codes     : {371}aaaaaaaaaaaa2dd490004714000085824338056906003fff0000000000170000ffdffc00008ae57f0000000000000
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {371}aaaaaaaaaaaa2dd490004714000085824338056906003fff0000000000170000ffdffc00008ae57f0000000000000
^[[O^[[I[fineoffset_ws90_decode] abort length
codes     : {62}303f1024a553e438
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {62}303f1024a553e438
[fineoffset_ws90_decode] abort length
codes     : {6}30
[fineoffset_ws90_decode] abort length
codes     : {371}aaaaaaaaaaaa2dd490004714000085824338066106003fff0000000000170000ffdffc00008a48db0000000000000
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {371}aaaaaaaaaaaa2dd490004714000085824338066106003fff0000000000170000ffdffc00008a48db0000000000000
```

When running 
`rtl_433  -f 915M  -X 'n=Ecowitt-WS90,prio=9,m=FSK_PCM,s=58,l=58,r=3000,preamble=aaaa2dd4'`

I get
```
time      : 2023-11-27 21:19:48
model     : Ecowitt-WS90 count     : 1             num_rows  : 1             rows      : 
len       : 305          data      : 90004714000085824338056406003fff0000000000170000ffdffc00008a9f340000000000000
codes     : {305}90004714000085824338056406003fff0000000000170000ffdffc00008a9f340000000000000
```

But I am not sure what to make of it. Any suggestion on how to get this sensor to be decoded properly.
Thanks a lot.

jpochmara, 2023-11-28:
Try adding '-Y minmax' to your rtl_433 command to set the FSK pulse detector mode and see if that makes a differences.

sombrero2003, 2023-11-28:
Unfortunately it does not change, i.e., no output from  `rtl_433 -R 244 -f 915M -Y minmax`

jpochmara, 2023-11-28:
In the file src/devices/fineoffset_ws90.c at line 66, change the 350 to 400 and recompile.  Then try again. 

This is where the length of the packet is checked.  You seem to be getting packets that are 371 bits long.  On my WS90 when I upgraded to 1.3.8 I was seeing packets that are 342 bits long.  Seems like there might be some variation in the packet length.

zuckschwerdt, 2023-11-28:
@jpochmara might be due to the abnormally wide reset limit, 3000 µs / 58 µs might give up to 52 trailing "phantom" bits.

sombrero2003, 2023-11-28:
Dear @jpochmara  and @zuckschwerdt , 
thank you both.  Changing the package length to 400 does the trick. The sensor now gets decoded properly!
Out of curiosity, why do you need to specify an upper limit for the package length?
all the best.

zuckschwerdt, 2023-11-29:
We should find a proper the reset limit.

jpochmara, 2023-11-29:
In the code the reset limit is set to 1500, which seems to work well for me.

@sombrero2003 what reset limit are you using now?

After spending sometime with the Ecowitt decode code in rtl_433, I am beginning to think that a lot of the packet length checking is not really the best thing.  There are places where the packet length is used to tell what device the packet is coming from.  I have an older WH32B which transmit a ~900 bit packet, and a newer version that produces a ~170 bit packet. Both provide the same information.  And the packet length checking in rtl_433 caused the new version to fail to decode.

gdt, 2023-11-29:
A lot of the point of the length checks is to avoid trying to decode something that is not in fact the format expected by the decoder.  These checks help prevent bad decodes on things that aren't from the intended devices.  Rather than throwing the checks out, consider checking for old and new ranges, and then only decoding the old and new formats.   Perhaps that's not it, but I suspect this can be understood better.

zuckschwerdt, 2023-11-29:
The reset limit should be maximum number of expected 0's multiplied with symbol length then rounded up.
The longest run of 0's we see in the examples is two bytes plus surrounding bits, let's say 24 bit, so 1500 µs seems a good value, the current 3000 µs looks excessive. But wind, gust, dir and uv could reasonably all be 0 and that would need 32 or more 0-bits. Really hard to get right :/

jpochmara, 2023-11-29:
The problem I ran into with packet length check is around the Ecotwitt WH32B device.  I purchased a WH32B about 9 months ago, and it just worked.  No problems.  About a month ago I purchased another WH32B, now labeled WN32BP, and that device did not decode.  The new device transmits two 170 bit packets back to back, whereas the old device transmitted one large packet (971 bits) with two packets of data in it.  And then found that the decode routine for the WH32B/WH25 tries to determine packets not just for the WH32B/WH25 but for two other devices too.

It seems like you should be able to look at the model ID at the start of the packet,  which will tell you where the checksum and CRC are located, check those and if they are correct say we got the correct decode.  In fact I reworked the WH32B/WH25 routine to do just that and have not had any issues so far.  But I do not have the other sensors to test against, so I am a little hesitant  put the code up for inclusion.

sombrero2003, 2023-11-30:
> In the code the reset limit is set to 1500, which seems to work well for me.
> 
> @sombrero2003 what reset limit are you using now?
> 

3000. I didn't have to change this part of the code to get the sensor to be decoded.

MiltonBrian, 2023-12-02:
We might open any window by using it and acquiring its proficient highlights. Hence, it is a standard program all around the globe. [https://ziapc.org/aact-portable-crack-serial-keygen/](https://ziapc.org/aact-portable-crack-serial-keygen/)

jpochmara, 2023-12-14:
@sombrero2003 can you try a reset limit of 1500 and see if that affects the packet length for the WS90?

victorhooi, 2023-12-30:
I have a Ecowitt WS90.

Out of the box, it was on firmware 1.3.3.

I'm using a Macbook, and installed rtl_433 via homebrew.

I was able to run rtl_433 with no options using a Nooelec RTL-SDR V5, and it picked up the WS90 data successfully - I was super impressed! =):
```

victorhooi@Victors-MacBook-Pro-3 ~/D/Ablage> rtl_433
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR with TLS
Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-31 00:28:37
model     : Fineoffset-WS90                        ID        : 004d95
Battery   : 0.520        Battery Voltage: 2240 mV  Temperature: 21.2 C       Humidity  : 54 %          Wind direction: 0         Wind speed: 0.0 m/s       Gust speed: 0.0 m/s       UVI       : 0.0           Light     : 170.0 lux     Flags     : 02
Total Rain: 0.0 mm       Supercap Voltage: 0.3 V   Firmware Version: 133     Extra Data: 3fff000000------0000ffbffb0000          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-31 00:28:37
model     : Fineoffset-WS90                        ID        : 004d95
Battery   : 0.520        Battery Voltage: 2240 mV  Temperature: 21.2 C       Humidity  : 54 %          Wind direction: 0         Wind speed: 0.0 m/s       Gust speed: 0.0 m/s       UVI       : 0.0           Light     : 170.0 lux     Flags     : 02
Total Rain: 0.0 mm       Supercap Voltage: 0.3 V   Firmware Version: 133     Extra Data: 3fff000000------0000ffbffb0000          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-31 00:28:46
model     : Fineoffset-WS90                        ID        : 004d95
Battery   : 0.520        Battery Voltage: 2240 mV  Temperature: 21.2 C       Humidity  : 52 %          Wind direction: 337       Wind speed: 0.0 m/s       Gust speed: 1.7 m/s       UVI       : 0.0           Light     : 180.0 lux     Flags     : aa
Total Rain: 0.0 mm       Supercap Voltage: 0.3 V   Firmware Version: 133     Extra Data: bfff000000------0000ffbffb0000          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

However, I then updated my WS90 to the current firmware (1.3.8) from the Ecowitt website. rtl_433 then stopped picking up any data, and I was quite flummoxed (I thought my RTL-SDR dongle had gone bad) - then I saw https://www.wxforum.net/index.php?topic=46182.0, which linked me to this issue.

So it seems I'm affected by this issue as well. Let me know if there's any testing you'd like me to do.

sombrero2003, 2023-12-30:
> @sombrero2003 can you try a reset limit of 1500 and see if that affects the packet length for the WS90?
Sorry for the delay @jpochmara 

Setting the reset limit to 1500 seems to reduce the package length.

Here is what I am seeing with a reset limit of 1500. I let it to run for a little while and I getting package lengths < 350.
Would this be a more robust solution? I guess this another way to ask what reset limit does ;-)

```
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
[fineoffset_ws90_decode] abort length
codes     : {570}33000000000000000000000000000000000000000000000000000000000000000000000000000ffff00000000000000000000000000000000000000000000000000000000000000
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {570}33000000000000000000000000000000000000000000000000000000000000000000000000000ffff00000000000000000000000000000000000000000000000000000000000000
[fineoffset_ws90_decode] abort length
codes     : {519}f326cccd9b33333330cd9999807ffffe19b1b333f0000cc06066046006393fff3867e03c0c01870cf3f01fcc0c7e079ec030f99e03086060018803333f0fe0c020
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {519}f326cccd9b33333330cd9999807ffffe19b1b333f0000cc06066046006393fff3867e03c0c01870cf3f01fcc0c7e079ec030f99e03086060018803333f0fe0c020
[fineoffset_ws90_decode] short package at 60 (182)
codes     : {182}aaaaaaaaaaa2dd4510d18140d7f00f83affffff3d82208
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {182}aaaaaaaaaaa2dd4510d18140d7f00f83affffff3d82208
[fineoffset_ws90_decode] short package at 60 (182)
codes     : {182}aaaaaaaaaaa2dd4510d18140d7f00f83affffff3d82208
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {182}aaaaaaaaaaa2dd4510d18140d7f00f83affffff3d82208
[fineoffset_ws90_decode] abort length
codes     : {32}dc370a18
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {32}dc370a18
[fineoffset_ws90_decode] WS90 detected, buffer is 341 bits length
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-30 17:15:57
model     : Fineoffset-WS90                        ID        : 004714
Battery   : 0.700        Battery Voltage: 2520 mV  Temperature: 5.5 C        Humidity  : 77 %          Wind direction: 99
Wind speed: 0.7 m/s      Gust speed: 0.8 m/s       UVI       : 0.0           Light     : 0.0 lux       Flags     : 81
Total Rain: 154.5 mm     Supercap Voltage: 2.9 V   Firmware Version: 138     Extra Data: 3fff823ef9------12c5ffbffb0000
Integrity : CRC
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
```

--------

For a reset limit of 3000, I am getting the following

```
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
[fineoffset_ws90_decode] abort length
codes     : {120}001800200120200000000000000000
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {120}001800200120200000000000000000
[fineoffset_ws90_decode] short package at 231 (199)
codes     : {199}82212108004cc1e3001fccc83bf820000056116185141375bc
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {199}82212108004cc1e3001fccc83bf820000056116185141375bc
[fineoffset_ws90_decode] WS90 detected, buffer is 371 bits length
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-30 17:20:02
model     : Fineoffset-WS90                        ID        : 004714
Battery   : 0.710        Battery Voltage: 2540 mV  Temperature: 5.4 C        Humidity  : 78 %          Wind direction: 118
Wind speed: 0.7 m/s      Gust speed: 0.9 m/s       UVI       : 0.0           Light     : 0.0 lux       Flags     : 81
Total Rain: 154.5 mm     Supercap Voltage: 2.9 V   Firmware Version: 138     Extra Data: 3fff823ef9------12c5ffbffb0000
Integrity : CRC
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {371}aaaaaaaaaaaa2dd49000471400007f81c64e077609003fff823ef906091d12c5ffbffb00008a01c30000000000000
[fineoffset_ws90_decode] short package at 71 (169)
codes     : {169}ab5555555555545ba86f11045476ec38d8000400000
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {169}ab5555555555545ba86f11045476ec38d8000400000
[fineoffset_ws90_decode] short package at 71 (169)
codes     : {169}ab5555555555545ba86f11045476ec38d8000400000
[pulse_slicer_pcm] Fine Offset Electronics WS90 weather station
codes     : {169}ab5555555555545ba86f11045476ec38d8000400000
[fineoffset_ws90_decode] abort length
codes     : {9}a00
```

zuckschwerdt, 2023-12-30:
Using a smaller reset limit will potentially reduce the package length. This might help here but allowing longer packets is more robust.

The "reset limit" effectivly limits the number of consecutive 0-bits by ending a packet when there is that long of gap/silence.
A small reset limit is a good sanity check, but with raw PCM(NRZ) there might many 0's in succession legitimately -- raw PCM(NRZ) is just a bad choice as protocol…

S.a. https://triq.org/rtl_433/PULSE_FORMATS.html on NRZ

I'd avocate for a reset limit as small as possible, but here the mentioned change of "src/devices/fineoffset_ws90.c at line 66, change the 350 to 400" (i.e. allow overly long packets and just ignore the trailing phantom 0's) seems better suited.

Can you test that?

sombrero2003, 2023-12-30:
Thanks for the explanation
As far as I can tell, both of these fixes work equally well
1. setting the reset limit to 1500 with a package length of 350
2. setting the package length to 400 with a reset limit of 400.

I will go with 2 as you suggested.
For whatever it’s worth, after I updated the WS90 firmware and applying fix 2, WS90 sometimes disappears (from rtl_433) for hours. Restarting rtl_433 fixes the issue but I didn’t experience this behavior with the previous firmware. 
This may or may not be related. Thanks

zuckschwerdt, 2023-12-30:
Can you grab a sample (`-S unknown`) when the WS90 disappears? Might be difficult as this will write a lot of samples when running for a long time.

victorhooi, 2024-01-01:
Is there a quick and easy way of getting my WS90 (on the current firmware release - 1.3.8) to work with the current rtl_433?

As in, is there anything that I could set at say, runtime?

Or will I need to setup a build environment, and recompile rtl_433 packages myself? If so, what is the optimal change to make here?

The [comment above](https://github.com/merbanan/rtl_433/issues/2732#issuecomment-1872619969) mentions this line:

https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_ws90.c#L66

and to change `350` to `400`. Is that the only change I should make?

Or if there's going to be a new release of rtl_433 cut soon, with a properly vetted fix, I'd be happy to just wait till then =).

(It's just I got this WS90 recently, and was super excited to try it out...haha).

jpochmara, 2024-01-01:
I think that a reset limit of 400 is to low,  I have never used that myself.  With firmware 1.3.8 I am getting solid decodes with the packet length check set to 400, reset limit 1500, and using the minmax FSK pulse detector (-Y minmax).
Before the 1.3.8 firmware I saw a lot of short packets, by 2 or 3 bits,  And I could never tune rtl_433 to fix it.

sombrero2003, 2024-01-02:
Sorry I fixed my post. For option 2, I meant to use the default reset limit of 3000. Glad you caught the typo!

zuckschwerdt, 2024-01-02:
Thanks for the feedback. I've increased the packet length check to 400 now. Let's see how this works out.

closed 2024-01-02 by zuckschwerdt

wgrant, 2024-05-30:
My Ecowitt WS90 running firmware 1.4.3 is consistently showing even longer packets that otherwise decode fine. e.g.:

```
codes     : {423}5555555555516ea48002cb801a752512e9a0326a1001fffc000000000148b807fdffe0000479341a5dffe787e00000000000000000
codes     : {421}5555555555516ea48002cb801de52412e9a06eb0f001fffc000000000148b807feffe800047dc7825dffe625500000000000000000
codes     : {414}5555555555516ea48002cb801da52512e9a0f9e14801fffc000000000148b807feffe80004795ff25dffe68e9800000000000000
codes     : {453}5555555555516ea48002cb8020652512e99890e92001fffc000000000148b807feffe800047e05d25dffe1fdc8000000000000000000000000
```

I'm running with the limit increased locally to 500 for now.

kr105, 2024-07-22:
> My Ecowitt WS90 running firmware 1.4.3 is consistently showing even longer packets that otherwise decode fine. e.g.:
>
> ```
> codes     : [...]

Same here, after a bunch of digging I found this issue and I tested with limit 500 and it worked right away!

commit 0bffc9d8 (2024-01-02): Fix Fineoffset-WS90 to support newer firmware (closes #2732)