merbanan/rtl_433 #2733: no output on 23.11 with same cmd args

engineeror, 2023-11-28:
Hello,
dropping this here just in case this is something on your side. myself am fine with the previous release, it just works! :)

this used to work great in previous 22.11 release:  
```
$ rtl_433 -f 868M -s 2M -R 173 -vvv                                                                                                                                                     
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR with TLS                                                                                                                    
Use -h for usage help and see https://triq.org/ for documentation.                                                                                                                                  
Trying conf file at "rtl_433.conf"...                                                                                                                                                               
Trying conf file at "/home/user/.config/rtl_433/rtl_433.conf"...                                                                                                                                    
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...                                                                                                                                        
Trying conf file at "/etc/rtl_433/rtl_433.conf"...                                                                                                                                                  
                                                                                                                                                                                                    
New defaults active, use "-Y classic -s 250k" for the old defaults!                                                                                                                                 
                                                                                                                                                                                                    
Registered 1 out of 223 device decoding protocols                                                                                                                                                   
Found 1 device(s)                                                                                                                                                                                   
                                                                                                                                                                                                    
trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001                          
Found Rafael Micro R820T tuner            
Using device 0: Generic RTL2832U OEM      
Exact sample rate is: 2000000.052982 Hz   
[R82XX] PLL not locked!                   
Sample rate set to 2000000 S/s.                                                                   
Bit detection level set to 0.0 (Auto).    
Tuner gain set to Auto.                   
Reading samples in async mode...                                                                  
Tuned to 868.000MHz.                      
baseband_demod_FM: low pass filter for 2000000 Hz at cutoff 400000 Hz, 2.5 us                     
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-11-28 20:48:33           
model     : Bresser-7in1 id        : 50313
Temperature: -6.6 C      Humidity  : 91            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 0             Rain      : 0.0 mm        Light     : 0.000 klx                    
Light     : 0.000 lux    UV Index  : 0.0           Battery   : 1             Integrity : CRC      

```
in the latest 23.11 release, i just get no output at all:
```
$ rtl_433 -f 868M -s 2M -R 173 -vvv
rtl_433 version -128-NOTFOUND branch master at 202311281827 inputs file rtl_tcp RTL-SDR SoapySDR with TLS

New defaults active, use "-Y classic -s 250k" if you need the old defaults

[Protocols] Registered 1 out of 250 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 2000000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 868000000 = 0
[SDR] Tuned to 868.000MHz.
[acquire_thread] acquire_thread enter...
[Baseband] low pass filter for 2000000 Hz at cutoff 400000 Hz, 2.5 us
```
same with `-Y classic`, also tried +-5000Hz, same nothingness  
unless i do `-vvvv`:
```
$ rtl_433 -f 868M -s 2M -R 173 -vvvv                                                                                                                                                    
rtl_433 version -128-NOTFOUND branch master at 202311281827 inputs file rtl_tcp RTL-SDR SoapySDR with TLS                                                                                           
                                                                                                                                                                                                    
New defaults active, use "-Y classic -s 250k" if you need the old defaults                                                                                                                          
                                                                                                                                                                                                    
[Protocols] Registered 1 out of 250 device decoding protocols                                                                                                                                       
[Input] The internals of input handling changed, read about and report problems on PR #1978                                                                                                         
[... 18 lines cut ...]
Pulse data: 172 pulses                                                                                                                                                                              
[  0] Pulse:    0, Gap:    2, Period:    2                                                                                                                                                          
[  1] Pulse:    2, Gap:    1, Period:    3                                                                                                                                                          
[  2] Pulse:    2, Gap:    1, Period:    3                                                        
[  3] Pulse:    6, Gap:    2, Period:    8    
[  4] Pulse:   26, Gap:    6, Period:   32                                                        
[  5] Pulse:   16, Gap:    2, Period:   18   
[  6] Pulse:    5, Gap:    7, Period:   12       
[  7] Pulse:    3, Gap:    9, Period:   12
[  8] Pulse:    2, Gap:    2, Period:    4    
[  9] Pulse:    5, Gap:    2, Period:    7                                                        
[ 10] Pulse:    2, Gap:    2, Period:    4
[ 11] Pulse:    5, Gap:    7, Period:   12 
[ 12] Pulse:    6, Gap:    6, Period:   12  
[ 13] Pulse:    5, Gap:    2, Period:    7                                                        
[ 14] Pulse:   13, Gap:   10, Period:   23                                                        
[ 15] Pulse:    3, Gap:    9, Period:   12
[ 16] Pulse:   13, Gap:    2, Period:   15
[ 17] Pulse:    5, Gap:    7, Period:   12
[ 18] Pulse:    6, Gap:    6, Period:   12
[ 19] Pulse:    5, Gap:    2, Period:    7
[ 20] Pulse:    2, Gap:    2, Period:    4   

```
but sorry this is beyond me

zuckschwerdt, 2023-11-28:
Oh, and what is the reason for `-s 2M`? The default rate is recommended for e.g. Bresser 7-in-1.

engineeror, 2023-11-28:
`-s 2M` was just a mindless copy paste from somewhere.  
unfortunately, neither removing `-s 2M` nor `-R 173`, makes the Bresser WS appear:
```
3 version -128-NOTFOUND branch master at 202311281827 inputs file rtl_tcp RTL-SDR SoapySDR with TLS

New defaults active, use "-Y classic -s 250k" if you need the old defaults

[Protocols] Registered 1 out of 250 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 1000000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 868000000 = 0
[SDR] Tuned to 868.000MHz.
[acquire_thread] acquire_thread enter...
[Auto Level] Estimated noise level is -16.8 dB, adjusting minimum detection level to -13.8 dB
[Baseband] low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
[Auto Level] Estimated noise level is -18.3 dB, adjusting minimum detection level to -15.3 dB
[Auto Level] Estimated noise level is -19.6 dB, adjusting minimum detection level to -16.6 dB
[Auto Level] Estimated noise level is -20.7 dB, adjusting minimum detection level to -17.7 dB
[Auto Level] Estimated noise level is -22.6 dB, adjusting minimum detection level to -19.6 dB
[Auto Level] Estimated noise level is -24.0 dB, adjusting minimum detection level to -21.0 dB
[Auto Level] Estimated noise level is -25.1 dB, adjusting minimum detection level to -22.1 dB
[Auto Level] Estimated noise level is -26.4 dB, adjusting minimum detection level to -23.4 dB
[Auto Level] Estimated noise level is -27.4 dB, adjusting minimum detection level to -24.4 dB
[Auto Level] Estimated noise level is -28.4 dB, adjusting minimum detection level to -25.4 dB
[Auto Level] Current noise level -29.1 dB, estimated noise -29.0 dB
[Auto Level] Current noise level -29.1 dB, estimated noise -28.9 dB
[Auto Level] Current noise level -29.2 dB, estimated noise -29.1 dB
[Auto Level] Current noise level -29.1 dB, estimated noise -29.1 dB
[Auto Level] Current noise level -29.3 dB, estimated noise -28.8 dB
[Auto Level] Current noise level -29.3 dB, estimated noise -29.2 dB
[Auto Level] Current noise level -29.1 dB, estimated noise -29.0 dB
[Auto Level] Current noise level -29.2 dB, estimated noise -29.2 dB
[Auto Level] Current noise level -29.2 dB, estimated noise -29.2 dB
[Auto Level] Current noise level -29.2 dB, estimated noise -28.9 dB

```

zuckschwerdt, 2023-11-29:
Levels look good, but also like the band is empty. Try `-f 868.3M`
Otherwise capture a sample, check on https://triq.org/pdv/ and upload as zip here. https://triq.org/rtl_433/ANALYZE.html

Daniel-WW, 2023-11-30:
Same problem here. For me it looks like the conf file with the band settings will be ignored. Do I have to change or renew the conf file in the new version?

Gollam, 2023-11-30:
@zuckschwerdt I just ran into the same issue today. My conf file was at "/etc/rtl_433/rtl_433.conf" and upgrading silently broke things. Maybe it would be helpful to show a message e.g. "No config file found at the search locations."

Daniel-WW, 2023-12-09:
Thanks for the advices. I'm running rtl433 as docker container with docker-compose. I had to change the mounted volume with new directory path, like this: 

```yaml
volumes:
      # - $DOCKERDIR/rtl-433tomqtt/rtl_433.conf:/etc/rtl_433/rtl_433.conf:ro
      - $DOCKERDIR/rtl-433tomqtt/rtl_433.conf:/usr/local/etc/rtl_433/rtl_433.conf:ro
```
Now it works again.

closed 2024-01-21 by zuckschwerdt