merbanan/rtl_433 #2750: weather station Altronics-X7064

adminy, 2023-12-08:
I'm getting reports of:
```bash
root in ~ λ rtl_433
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-12-08 10:48:14
model     : Altronics-X7064                        id        : 697
Channel   : 4            Battery_OK: 1             Temperature_F: 48.3       Humidity  : 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-12-08 10:48:44
model     : Altronics-X7064                        id        : 697
Channel   : 4            Battery_OK: 1             Temperature_F: 48.3       Humidity  : 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-12-08 10:49:14
model     : Altronics-X7064                        id        : 697
Channel   : 4            Battery_OK: 1             Temperature_F: 48.3       Humidity  : 1             Integrity : CHECKSUM
```

However the screen of the weather station is also getting wind direction and speed. Which are properties that seem to be left out of here.

zuckschwerdt, 2023-12-08:
You uploaded 4 files, the 2nd is just a faint TPMS signal. The other 3 work fine:
```
[Input] Test mode active. Reading samples from file: g001_433.92M_250k.cu8
model     : Emax-W6      id        : 697
Channel   : 4            Battery_OK: 1             Temperature: 48.5 F       Humidity  : 86 %
Wind avg speed: 11.2 km/h       Wind Direction: 178
Total rainfall: 16.0 mm  UV Index  : 4             Lux       : 0             Integrity : CHECKSUM

[Input] Test mode active. Reading samples from file: g004_433.92M_250k.cu8
model     : Emax-W6      id        : 697
Channel   : 4            Battery_OK: 1             Temperature: 48.3 F       Humidity  : 87 %
Wind avg speed: 4.0 km/h         Wind Direction: 185
Total rainfall: 16.0 mm  UV Index  : 4             Lux       : 0             Integrity : CHECKSUM

[Input] Test mode active. Reading samples from file: g005_433.92M_250k.cu8
model     : Emax-W6      id        : 697
Channel   : 4            Battery_OK: 1             Temperature: 48.3 F       Humidity  : 87 %
Wind avg speed: 11.0 km/h       Wind Direction: 138
Total rainfall: 16.0 mm  UV Index  : 4             Lux       : 0             Integrity : CHECKSUM
```

zuckschwerdt, 2023-12-08:
Also note that ambient pressure is always measured at the base station /head unit, it's by principle not an "outdoor" value.

adminy, 2023-12-08:
thank you. seems like I had a cached bin