merbanan/rtl_433 #2751: Govee water leak detector not detecting

bkenobi, 2023-12-13:
I have a number of Acurite sensors (thermo-hygrometers and 5in1 weather station) and a Blueline power meter reader working in rtl_433.  I just acquired a set of Govee H5054 water leak detectors and added protocol 192 to my launch string but I'm not seeing any activity.  This is a new sensor to me so perhaps it doesn't respond the way I'm used to.  Pushing the button to make it chime seems to suggest in protocol that it should send a signal which resets the "wet" status to 0.  Also, putting the contacts in liquid causes it to alert but I don't see anything in the rtl__433 output.  I have not found any induction in the documentation or other issues I've reviewed that this should work any different than other sensors (set protocol and it should detect), so I'm not clear if I'm missing something.  These appear to be the older version so I would have expected them to work out of the box.

zuckschwerdt, 2023-12-13:
Support for the newer Govee H5054 model was recently added. Is your rtl_433 version up to date (23.11)?

bkenobi, 2023-12-14:
I updated my installation to the latest release this morning. For some reason it's still on 21.12 so I'll try again.

bkenobi, 2023-12-14:
I was using a branch that had a test code and I forget how to switch back to the master.  I'll look around to find the command.  I'm currently sitting on a version from 2022 or earlier from the dates of the files.

bkenobi, 2023-12-14:
I grabbed the repository again and made/installed.  I'm getting an error now so if this is the wrong way to ask about that I can create a new issue.

When I run the binary I get the following:
[image]

I tried the test binary and it looks like the interface isn't loading.
`pi@RPi3:~ $ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN:

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.
`

bkenobi, 2023-12-14:
Ok, never mind.  I unplugged and re-plugged the SDR in and now it seems to be working better (the SDR detects and the existing modules I have set up are being received).  I still don't see any traffic from the Govee but at least I'm up to date with the current version of rtl_433.  Is there a debug set I can provide that might show what I need to change?

zuckschwerdt, 2023-12-14:
Did you try protocol 231, the newer Govee model?
Get debugging info with `-R 192:vv -R 231:vv`

bkenobi, 2023-12-14:
I have been testing with protocol 192 (I didn't notice 231).  I tried both by calling 192 and letting rtl_433 use all available.  When I enable 231 it does appear to detect.

```
pi@RPi3:~ $ rtl_433 -R 231:vv
rtl_433 version 23.11-12-g8950dc41 branch master at 202312052040 inputs file rtl_tcp RTL-SDR
Found Fitipower FC0013 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: , "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {207}fffffffffffffffffffffffffffffffffffffffffffffffffffe
[govee_h5054_decode] Original Bytes: 04383054b701
[govee_h5054_decode] id=0438
[govee_h5054_decode] unk16=3
[govee_h5054_decode] event=0
[govee_h5054_decode] event_data=54
[govee_h5054_decode] crc_sum=b701
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-14 08:14:24
model     : Govee-Water  id        : 1080
event     : Button Press detect_wet: 0             Raw Code  : 04383054b701
Integrity : CRC
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {2}c, {1}0, {48}04383054b701, {1}0, {48}04383054b701, {1}0, {48}04383054b701, {1}0, {48}04383054b701, {1}0, {249}04383054bf00040000078c7c2c76f20000526cf38b30048000021c182a5b808, {1}0, {1}0, {48}04383054b701, {1}0, {48}04383054b701, {1}0, {48}04383054b701, {1}0, {48}04383054b701, {1}0, {48}04383054b701
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[govee_h5054_decode] Original Bytes: 0438320f3afd
[govee_h5054_decode] id=0438
[govee_h5054_decode] unk16=3
[govee_h5054_decode] event=2
[govee_h5054_decode] event_data=0f
[govee_h5054_decode] crc_sum=3afd
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-14 08:14:33
model     : Govee-Water  id        : 1080
event     : Water Leak   detect_wet: 1             Leak Num  : 15
Raw Code  : 0438320f3afd Integrity : CRC
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {2}c, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {21}043830
[govee_h5054_decode] Original Bytes: 0438320f3afd
[govee_h5054_decode] id=0438
[govee_h5054_decode] unk16=3
[govee_h5054_decode] event=2
[govee_h5054_decode] event_data=0f
[govee_h5054_decode] crc_sum=3afd
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-14 08:14:34
model     : Govee-Water  id        : 1080
event     : Water Leak   detect_wet: 1             Leak Num  : 15
Raw Code  : 0438320f3afd Integrity : CRC
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {27}41e75fa, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd, {1}0, {48}0438320f3afd
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {61}fffffffffffffff8, {61}fffffffffffffff8, {61}fffffffffffffff8
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {207}fffffffffffffffffffffffffffffffffffffffffffffffffffe, {232}fffffffffffdede1ffffffffffffffe1ffffffffffc7ffffffffffffff
^CSignal caught, exiting!
```
And when I switch back to non-verbose it looks correct.  Looks like I just had the wrong protocol selected.  Thanks!

closed 2023-12-14 by bkenobi