merbanan/rtl_433 #2753: Support for Weather Station IMETEO X6

fermosan, 2023-12-13:
Hello,

Information about product is here: 
https://www.technisat.com/en_XX/IMETEO-X6/352-22343-23994/

Looks similar (or very close) to Emax-W6. 
Wind Speed (Avg) is always 0. 
It reports erroneus values for Lux and UV (i think there are no sensors for that on the weather station)
rtl_433] time      : 2023-12-13 23:23:51
[rtl_433] model     : Emax-W6
[rtl_433] id        : 319
[rtl_433] Channel   : 4
[rtl_433] Battery_OK: 1
[rtl_433] Temperature: 13.8 C
[rtl_433] Humidity  : 84 %
[rtl_433] Wind avg speed: 0.0 km/h
[rtl_433] Wind Direction: 133
[rtl_433] Total rainfall: 18.6 mm
[rtl_433] UV Index  : 24
[rtl_433] Lux       : 0
[rtl_433] Integrity : CHECKSUM
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2023-12-13 23:24:21
[rtl_433] model     : Emax-W6
[rtl_433] id        : 319
[rtl_433] Channel   : 4
[rtl_433] Battery_OK: 1
[rtl_433] Temperature: 13.8 C
[rtl_433] Humidity  : 84 %
[rtl_433] Wind avg speed: 0.0 km/h
[rtl_433] Wind Direction: 132
[rtl_433] Total rainfall: 18.6 mm
[rtl_433] UV Index  : 24
[rtl_433] Lux       : 0
[rtl_433] Integrity : CHECKSUM
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2023-12-13 23:24:51
[rtl_433] model     : Emax-W6
[rtl_433] id        : 319
[rtl_433] Channel   : 4
[rtl_433] Battery_OK: 1
[rtl_433] Temperature: 13.9 C
[rtl_433] Humidity  : 84 %
[rtl_433] Wind avg speed: 0.0 km/h
[rtl_433] Wind Direction: 170
[rtl_433] Total rainfall: 18.6 mm
[rtl_433] UV Index  : 24
[rtl_433] Lux       : 0
[rtl_433] Integrity : CHECKSUM

[2023-12-13T23:24:21+0200] DEBUG:root:MQTT message: "{\"time\":\"2023-12-13 23:24:21\",\"model\":\"Emax-W6\",\"id\":793,\"channel\":4,\"battery_ok\":1,\"temperature_C\":13.77778,\"humidity\":84,\"wind_avg_km_h\":0,\"wind_dir_deg\":132,\"rain_mm\":18.6,\"uv\":24,\"light_lux\":0,\"mic\":\"CHECKSUM\"}"
[2023-12-13T23:24:21+0200] DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Emax-W6-4-793/Emax-W6-4-793-UTC/config
[2023-12-13T23:24:21+0200] DEBUG:root:Discovery timeout in the future for: homeassistant/device_automation/Emax-W6-4-793/Emax-W6-4-793-CH/config

Another device "Altronics-X7064-4-793" is also discovered with correct Temperature and wrong Humidity
[2023-12-13T23:28:44+0200] DEBUG:root:homeassistant/sensor/Altronics-X7064-4-793/Altronics-X7064-4-793-UTC/config:{"device_class": "timestamp", "name": "Timestamp", "entity_category": "diagnostic", "enabled_by_default": false, "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/Altronics-X7064/4/793/time", "unique_id": "Altronics-X7064-4-793-UTC", "device": {"identifiers": ["Altronics-X7064-4-793"], "name": "Altronics-X7064-4-793", "model": "Altronics-X7064", "manufacturer": "rtl_433"}, "force_update": "true"}
[2023-12-13T23:28:44+0200] DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r1, m12), 'b'homeassistant/sensor/Altronics-X7064-4-793/Altronics-X7064-4-793-UTC/config'', ... (421 bytes)
[2023-12-13T23:28:44+0200] DEBUG:root:homeassistant/device_automation/Altronics-X7064-4-793/Altronics-X7064-4-793-CH/config:{"automation_type": "trigger", "type": "button_short_release", "subtype": "button_1", "topic": "rtl_433/9b13b3f4-rtl433/devices/Altronics-X7064/4/793/channel", "platform": "mqtt", "device": {"identifiers": ["Altronics-X7064-4-793"], "name": "Altronics-X7064-4-793", "model": "Altronics-X7064", "manufacturer": "rtl_433"}, "force_update": "true"}
[2023-12-13T23:28:44+0200] DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r1, m13), 'b'homeassistant/device_automation/Altronics-X7064-4-793/Altronics-X7064-4-793-CH/config'', ... (345 bytes)
[2023-12-13T23:28:44+0200] DEBUG:root:homeassistant/sensor/Altronics-X7064-4-793/Altronics-X7064-4-793-B/config:{"device_class": "battery", "name": "Battery", "unit_of_measurement": "%", "value_template": "{{ float(value) * 99 + 1 }}", "state_class": "measurement", "entity_category": "diagnostic", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/Altronics-

zuckschwerdt, 2023-12-14:
Can you grab a sample, check with https://triq.org/pdv/ and upload as zip here? https://triq.org/rtl_433/ANALYZE.html

@ProfBoc75 can you then take a look?

fermosan, 2023-12-21:
[cu8 files.zip](https://github.com/merbanan/rtl_433/files/13733685/cu8.files.zip)

Hello, sorry for the delay. Find attached the samples.

ProfBoc75, 2023-12-21:
Hi , sorry for this late answer, nowadays I have less time to work on rtl 433.

@fermosan : can you give me the values related to your samples, like temp, humidity, wind direction, wind speed, lux , uv ... If you don't have these information, please capture again few samples, write down the information you have on the station display and share here into a table with the file name and the related figures.

Your both samples are decoded and yes, not normal to get UV index = 24 and a LUX at 0.

We are already able to decode another model without UV and LUX, but your model is coded like with UV and LUX but the byte which contains the UV information is = 0xD9 , so 0XD8 and UV is the 5 right bits  = 0x18 = 24, 
we should have values from 0x01 ( = 0 ) to 0x15 , where UVI > 11 is extreme in very few regions on the Earth.

So, your station may be coded differently and need to be taken into account as without UV and LUX, little update is need to the device decoder, but I need more sample to confirm that, please.

With my station, I got UVI = 13 with LUX = 100 K last May.

fermosan, 2023-12-21:
Hello, 
you will find the measurements reported by the  rtl (verified by the monitor) and some new samples. 
[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/13744127/g001_433.92M_250k.zip)

I think that this station does not have a UV sensor (i might be wrong, the only real way to be sure it's to open it). The manual also does not mention something about UV and the there is no UV measurement in the station (see attached photo). 

The value reported for the UV sensor is constant and never changes. 

Also, from within MQTT there are reported two different temperatures. 
emax_w6_4_793_temperature
emax_w6_4_793_temperature_2 
[image]
Running rtl_433 next and rtl_433 MQTT Auto Discovery (next)
RTL_433 reports only one temperature. 
[rtl_433] time      : 2023-12-21 18:50:43
[rtl_433] model     : Emax-W6
[rtl_433] id        : 319
[rtl_433] Channel   : 4
[rtl_433] Battery_OK: 1
[rtl_433] Temperature: 12.7 C
[rtl_433] Humidity  : 62 %
[rtl_433] Wind avg speed: 0.0 km/h
[rtl_433] Wind Direction: 74
[rtl_433] Total rainfall: 43.8 mm
[rtl_433] UV Index  : 24
[rtl_433] Lux       : 0
[rtl_433] Integrity : CHECKSUM

ProfBoc75, 2023-12-21:
From your station screenshot, we can see on the bottom right you have one "big" cell with Baro information (Baro 996 hpa), on my display station the same cell is split into 2 smaller cells, one with UV information and the other with Baro information, this Baro can also be LUX information (long press on the up/thermometer button on my station). So this confirm that you don't have UV and LUX.

On the external weather station itself, the UV LUX sensor is located into the outgrowth above the top left corner of the solar panel,  if yours is all white / sealed without transparent glass , it confirms that there's no sensor here either. On mine , I can see the sensor protected with a glass.

your UV byte is always = 0xD9  ( = 0xD8 ) and Lux Bytes always = 0x0101 ( = 0 ) .
We can update the decoder , add a condition to UV and LUX metrics, like provide information only if LUX > UVI , or better extract the 2 first left bits from the UV byte (byte 13) which looks like a tag = 11 (binary) for your station and on my station, these 2 bits are = 00,  

In emax.c : [line 203](https://github.com/merbanan/rtl_433/blob/57c22d7c73b88c7ed31cee303182d43a8606f772/src/devices/emax.c#L203)

add just before clang format, line 203

```
                int tag           = ((b[13] - 1) & 0xC0)>>6;
```
Then update [line 215](https://github.com/merbanan/rtl_433/blob/57c22d7c73b88c7ed31cee303182d43a8606f772/src/devices/emax.c#L215)

```
                        "uv",               "UV Index",         DATA_COND, tag !=3, DATA_FORMAT, "%u",       DATA_INT,    uv_index,
                        "light_lux",        "Lux",              DATA_COND, tag !=3, DATA_FORMAT, "%u",       DATA_INT,    light_lux,
```

About the Wind avg always at 0 Km/h , I don't understand why, since we can get the good values from your samples, and I guess it's related to your installation with MQTT in the middle ? If you're using the HA rtl433 MQTT plugin, I can't help , I don't know this part.

Notice that the weather station is sending / refreshing the information every 30 seconds, if no wind during that time, wind avg speed is 0 Km/h , this is the avg value between 2 frames, so within / during these 30 seconds.

fermosan, 2023-12-22:
Sounds perfect. I could test the changes when you commit.  tyvm

ProfBoc75, 2023-12-22:
Done, I also took the opportunity to add a small improvement to another weather station.

Please, try with my PR and let us know.

fermosan, 2023-12-24:
Hello, it works like a charm. Merry xmas. 
time      : 2023-12-24 01:50:41
model     : IMETEO-X6    id        : 319
Channel   : 4            Battery_OK: 1             Temperature: 66.7 F       Humidity  : 47 %          Wind avg speed: 1.6 km/h
Wind Direction: 60       Total rainfall: 43.8 mm   Integrity : CHECKSUM

ProfBoc75, 2023-12-24:
@fermosan : Thanks a lot, same to you,

@zuckschwerdt : can you merge my PR, please.

And Merry xmas to all rtl_433 Fans and Contributors !

gdt, 2024-06-03:
calling this fixed.

closed 2024-06-03 by gdt