merbanan/rtl_433 #2754: Contribution Question: Adding Multiple Decoders

eshaz, 2023-12-15:
I recently purchased a box of 50 or so old car keyfobs, and I also have a few other devices that I plan on using for various home automation things. I quickly tried these devices with rtl_433, and only a handful of them are implemented. I think it would be nice to contribute back to the community by creating decoders in this library that will support the various protocols implemented by these devices.

What would be the best way for me to contribute these additions? Would separate PRs for each protocol be OK, or is there some other way that would be easier for maintainers to review and merge a bunch of new devices?

zuckschwerdt, 2023-12-16:
Usually this type of sender (doorbell style) sends only a fixed code, likely EV1527 protocol. This is best implemented with flex decoders. See https://github.com/merbanan/rtl_433/tree/master/conf
A combined PR is fine for these. You likely want to `match` on your senders IDs, a commented example as template is still appreciated.

klohner, 2024-01-08:
@eshaz Would you mind posting some .cu8 files of these samples with descriptions here?

I took a look through your audiovox_car_remote.c code in the PR and I'm concerned that without any matching of a product ID code or CRC verification in the samples, it will find itself matching other kinds of devices out in the wild too readily.  A flex decoder might be more appropriate.

eshaz, 2024-01-09:
@klohner I was thinking it would be a good idea to include sample data and docs for these in the [rtl_433_tests](https://github.com/merbanan/rtl_433_tests) repo. I can open a PR there with the samples / docs.

Regarding the Nutek remotes, (I renamed `audiovox_car_remote.c` to `nutek_car_remote.c` in a later commit to match the actual manufacturer), I'll go ahead and move this one into a flex decoder. I was getting a few false positives with this as I was adding the other devices.

eshaz, 2024-01-09:
I've added some samples / docs / pictures here: merbanan/rtl_433_tests#462 if you want to take a look. This has about half of the devices. I plan on adding the rest tomorrow.

eshaz, 2024-01-10:
Everything is added now into the tests PR. Also, I've moved that Nutek remote into a flex decoder.

gdt, 2024-06-03:
Seems like this was a question about process, answered, and some progress.  Future contributions welcome, but I don't see this issue being open as helpful to anyone.   If I got that wrong please tell me.

closed 2024-06-03 by gdt