merbanan/rtl_433 #2756: Support for Ecowitt WH55 Leak Detector (Currently Uses Incorrect Protocol)

cdavis289, 2023-12-18:
I just got some WH55 sensors and they are not officially supported by rtl-433, however, they show under the incorrect protocol (246). Output data is attached. The only value that changes is the total rainfall and the wind gust (2.45 when not triggered, 1.22 when triggered). The sensor should be reporting a measured capacitance.

From the [manual]((https://osswww.ecowitt.net/uploads/20220803/WH55%20Manual.pdf): 
High sensitivity: The water leakage alarm will be triggered when the detected water leak capacitance value equals to or over 60PF and be muted when the value equals to or lower 50PF.

Low sensitivity: The water leakage alarm will be triggered when the detected water leak capacitance value equals to or over 90PF and be muted when the value equals to or lower 80PF._

I used the command from this [forum](https://groups.google.com/g/rtl_433/c/dEz2UEVBses)

Data generated using command: rtl_433 -R 246  -X 'n=name,m=FSK_PCM,s=60,l=60,g=1000,r=2500,preamble=aaaa2dd4'
[wh55 data.txt](https://github.com/merbanan/rtl_433/files/13708045/wh55.data.txt)

I'm happy to do more tests/research but need recommendations. This is my first foray into decoding a signal.

zuckschwerdt, 2023-12-18:
Interesting that such a simple "leak" / "no-leak" sensor would transfer that complex data. A false positive is also strange as the "Fineoffset-WH1050" / "TFA-303151" protocol uses a CRC-8 over 8 bytes -- random collisions will be extremely unlikely.

Can you also grab a cu8 sample and upload as zip here? https://triq.org/rtl_433/ANALYZE.html

Your posted codes for reference:

`{96}5510cf5503 0292 be67 4590 49` No alarm (high sensitivity, CH 1)
`{96}5510cf5503 010b fe1e b418 04` Alarm  (high sensitivity, CH 1)
`{95}5510cf5503 029a be50 3610 08` No alarm (high sensitivity, CH 1)
`{95}5510cf5503 0293 3ee9 4810 08` No Alarm (low sensitivity, CH 1)
`{95}5510cf5503 010b 7e64 7a10 48` Alarm (low sensitivity, CH 1)

cdavis289, 2023-12-18:
[g007_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/13708843/g007_433.92M_250k.cu8.zip)

Attached is the cu8 file. I uploaded it to https://triq.org/rtl_433/ANALYZE.html but I wasn't quite sure what to be looking for. Should I grab more data? I have a few sensors so I can get dumps from a few sensors if that helps to decode the data.

gdt, 2023-12-19:
Wildly speculating, they might use a standard CRC and there are id/flag bits that give the model.

zuckschwerdt, 2023-12-19:
I does look like proper 60 µs PCM.

Can you play around with the channel setting (which hopefully changes single bits) and report the codes found?

This will help to find the checksum fields and determine the type of check. It could be 7 or 8 bytes payload and then 3-4 bytes of different checks possibly.

cdavis289, 2023-12-19:
Here are the codes for various channels and high and low sensitivity. I used bitbench and think I located the channel using this:

hhhhhhhhhhhhhhhhh ID bbbbb hhhhhhhhhh

5555555555516ea2a8067aa81814cdf7963c804 CH 1, High
5555555555516ea2a8067aa81814c1f3e681008 CH 1, Low
5555555555516ea2a8867aa81814a5f1e8ec804 CH 2, High
5555555555516ea2a8867aa81814b9f3566c804 CH 2, Low
5555555555516ea2a9067aa81814adf34ad0804 CH 3, High
5555555555516ea2a9867aa81814cdf292bc804 CH 4, High

cdavis289, 2023-12-19:
Made some more progress. I now think I have the CH, Dev ID, and a threshold for the alarm (ie <3028 alarm sounds)

[Bitbench](https://triq.net/bitbench#c=5555555555516ea2a8067aa81814fdf5441c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e801814fdf6b96c8048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa81814c1f3e681008%20%5BCh%201%2C%20Low%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814b9f3566c804%20%5BCh%202%2C%20Low%2C%20No%20Alarm%5D&c=&c=&c=5555555555516ea2a8067aa81814cdf7963c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814a5f1e8ec804%20%5BCh%202%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9067aa81814adf34ad0804%20%5BCh%203%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9867aa81814cdf292bc804%20%5BCH%204%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa8180bdff55500804%20%5BCh%201%2C%20High%2C%20Alarm%5D&c=5545555555516ea2a8067e80180ae7f53fc9048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20Alarm%5D&c=5555555555516ea2a8067aa8181275f4c111048%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e8018141df06a3c804%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&f=PREAMBLE%20hhhhhhhhhhhhhhhhh%20CH%20bbbbb%20ID%20hhhhh%20ALARM%20THRESHOLD%3F%20hhhhh%20HMMMM%20hhhhhhhhhhhhhhhh&cw=4)

zuckschwerdt, 2023-12-20:
You can sync the preamble like [this BitBench](https://triq.net/bitbench#c=5555555555516ea2a8067aa81814fdf5441c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e801814fdf6b96c8048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa81814c1f3e681008%20%5BCh%201%2C%20Low%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814b9f3566c804%20%5BCh%202%2C%20Low%2C%20No%20Alarm%5D&c=&c=&c=5555555555516ea2a8067aa81814cdf7963c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814a5f1e8ec804%20%5BCh%202%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9067aa81814adf34ad0804%20%5BCh%203%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9867aa81814cdf292bc804%20%5BCH%204%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa8180bdff55500804%20%5BCh%201%2C%20High%2C%20Alarm%5D&c=5545555555516ea2a8067e80180ae7f53fc9048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20Alarm%5D&c=5555555555516ea2a8067aa8181275f4c111048%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e8018141df06a3c804%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&f=PREAMBLE%208h%20FLAGS%3F%202b%20CH%202d%20ID%20hhhhh%20ALARM%20THRESHOLD%3F%20hhhhh%20HMMMM%20hhhhhhhhhhhhhhhh&a=Preamble&m=2dd4&cw=4).

AhrBee, 2023-12-20:
If you need another test case, I have WH55 (ecowitt) sensor (among others) and am motivated to get off the 'phone home' gateways in order to read and process sensor outputs directly. Let me know what, if anything, I can add to the decoding/testing effort.

cdavis289, 2023-12-20:
@AhrBee do you happen to have one with a low battery? I haven't been able to get any decodes for a unit with a full and a low battery for comparison.

AhrBee, 2023-12-20:
The lowest rechargeable AAA batteries I have are at 1.26VDC and they show 40% battery level (two bars on the WSView app connected to the Ecowitt GW1000). Directly monitoring the WH55 shows battery: 1 (which means still good). Going to root around and drain the batteries on something and report back, By the way, my WH55 transmits at 915 MHz.

What is the exact command syntax you'll need to save the output?

cdavis289, 2023-12-20:
@AhrBee I actually found some lower charged batteries and was able to correlate the output with the battery bar in the ecowitt software. I've matched 2, 3, and 4 bars. Charging some batteries now to check for fully charged. I think I have now found everything. I have:

> Family Code (55 for this sensor)
> Channel (0,1,2,3)
> ID (this matches with ecowitt software)
> Bat (number correlates with number of bars in ecowitt software, ie 2 = 2 bars)
> Alarm (1 = alarm, 2 = off)

[Bitbench](https://triq.net/bitbench#c=5555555555516ea2a8067aa81808b7f5fa80c110%20%5BAlarm%5D&c=5555555555516ea2a8067aa818143df78da1008%20%5B%2B3s%5D&c=5555555555516ea2a8067aa81814adf230bc804%20%5B%2B6s%5D&c=5555555555516ea2a8067aa81814b5f359ec804%20%5B%2B60s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B61s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B60s%2011%3A55%3A59%5D&c=5555555555516ea2a8067aa818149df0e759008%20%5Bchanged%20batteries%2C%203bars%5D&c=5555555555516ea2a8067aa818149df0e75c804&c=5555555555516ea2a8067aa81814adf230bc804%20&c=5555555555516ea2a8067aa810147df2e93c804&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b90804%20%5Bbatteries%202%20bars%5D&c=5555555555516ea2a8067aa81814adf230bc8040%20%5Bbatteries%203%20bars%5D&c=5555555555516ea2a8067aa82014cdf301b0805%20%5Bbatteries%204%20bars%5D&c=5555555555516ea2a8067aa82014ddf5cc8c804%20%20%5Bbatteries%204%20bars%5D&f=FAMILY_CODE%208h%20FLAGS%3F%202b%20CH%202d%20ID%2020h%20BAT%20d%20ALARM%20d%20HMMMM%20hhhhhhhhhhhhhhhh&a=Preamble&m=2dd4&cw=4)

The command I have been using is 

> rtl_433 -R 246  -X 'n=WH55_CH1,m=FSK_PCM,s=60,l=60,g=1000,r=2500'

but you will need to add a flag for your frequency since my sensors are 433 MHz. This device currently shows up at protocol 246 which is why I'm using the -R flag. It would be good to get some data for 915 MHz to make sure everything works for that frequency.

zuckschwerdt, 2023-12-20:
> Family Code (55 for this sensor)

This is very likely, Fine Offset codes most of their products like this (Ecowitt is a rebrand).

cdavis289, 2023-12-20:
Found some non-rechargeable batteries and was able to get a reporting of 5 corresponding to a fully charged battery.

Family Code (55 for this sensor)
Channel (0,1,2,3)
ID (this matches with ecowitt software)
Bat (number correlates with number of bars in ecowitt software, 1,2,3,4,5)
Alarm (1 = alarm, 2 = off)

[Bitbench](https://triq.net/bitbench#c=5555555555516ea2a8067aa81808b7f5fa80c110%20%5BAlarm%5D&c=5555555555516ea2a8067aa818143df78da1008%20%5B%2B3s%5D&c=5555555555516ea2a8067aa81814adf230bc804%20%5B%2B6s%5D&c=5555555555516ea2a8067aa81814b5f359ec804%20%5B%2B60s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B61s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B60s%2011%3A55%3A59%5D&c=5555555555516ea2a8067aa818149df0e759008%20%5Bchanged%20batteries%2C%203bars%5D&c=5555555555516ea2a8067aa818149df0e75c804&c=5555555555516ea2a8067aa81814adf230bc804%20&c=5555555555516ea2a8067aa810147df2e93c804&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b90804%20%5Bbatteries%202%20bars%5D&c=5555555555516ea2a8067aa81814adf230bc8040%20%5Bbatteries%203%20bars%5D&c=5555555555516ea2a8067aa82014cdf301b0805%20%5Bbatteries%204%20bars%5D&c=5555555555516ea2a8067aa82014ddf5cc8c804%20%20%5Bbatteries%204%20bars%5D&c=aaaaaaaaaaa8b75154033d54140a6ef88bee4020%20%5Bbatteries%205%20bars%5D&c=5555555555516ea2a8067aa82814e5f46330804%20%5Bbatteries%205%20bars%5D&f=FAMILY_CODE%208h%20FLAGS%3F%202b%20CH%202d%20ID%2020h%20BAT%20d%20ALARM%20d%20HMMMM%20hhhhhhhhhhhhhhhh&a=Preamble&m=2dd4&cw=4)

AhrBee, 2023-12-20:
Clearly I don't know what I am doing, but here are a few samples that I copied and pasted from the terminal outpt of the command
`rtl_433 -f 915M -R 246 -X 'n=WH55_CH1,m=FSK_PCM,s=60,l=60,g=1000,r=2500'`

The battery level is bouncing between low (20% 1 bar) and (40% 'normal'). I forced a leak detection (wet the back of the WH55) to trigger the alarm. The Ecowitt WH55 is set to channel 1 at 915MHz. 

[Bitbench](https://triq.net/bitbench#c=aaaaaaaaaaa2dd45500c82d0102e13e6cb0204%20%5Bbattery%201%20bar%5D&c=aaaaaaaaaaa2dd45500c82d0202e03e04701010%20%5Bbattery%202%20bars%5D&c=aaaaaaaaaaa2dd45500c82d0202f09f782e900&c=aaaaaaaaaaa2dd45500c05a000b09f782e9810%20&c=aaaaaaaa528b742a8064168100803fa1540404%20%5Bbattery%202%20bars%20leak%20detected%5D&c=aaaaaaaaaaa1ba8540320b4080665f82dbc6020%20%5Bbattery%202%20bars%20leak%20detected%5D&c=aaaaaaaaaaa2da8aa01905a040330f905a0408%20%5Bbattery%202%20bars%20back%20to%20normal%5D&c=aaa555555545ba8aa01905a0403b27d7e47200&f=FAMILY_CODE%208h%20FLAGS%3F%202b%20CH%202d%20ID%2020h%20BAT%20d%20ALARM%20d%20HMMMM%20hhhhhhhhhhhhhhhh&a=Preamble&m=2dd4&cw=4)

cdavis289, 2023-12-20:
Those match up except for lines 4-7. Line 4 showed a different device ID and lines 5-7 didn't match the format. Can you take out the batteries of your leak detector and run the rtl_433 command again and see if anything is showing? If you have a weather station you may also be seeing data from your weather station. 

When I run the rtl_433 command I have to wait until I see a device pop up showing temp, humidity, and wind speed and then the output directly below (same time stamp) shows the un-decoded leak detector message.

[image]

> aaaaaaaaaaa2dd45500c05a000b09f782e9810 
> aaaaaaaa528b742a8064168100803fa1540404 [battery 2 bars leak detected]
> aaaaaaaaaaa1ba8540320b4080665f82dbc6020 [battery 2 bars leak detected]
> aaaaaaaaaaa2da8aa01905a040330f905a0408 [battery 2 bars back to normal]

AhrBee, 2023-12-20:
> Those match up except for lines 4-7. Line 4 showed a different device ID and lines 5-7 didn't match the format. Can you take out the batteries of your leak detector and run the rtl_433 command again [...]

Pretty sure nothing else is reaching the receiver: no antenna connected, WH55 right up against the dongle. I ran a baseline at 915MHz with no protocol filter, and only the WH55 showed up. In any event, here is the bitbench with data collected only after I see the WH55 put something out (as you have referenced). Here's a couple of snippets to see if I am on track:

```
rtl_433 -f 914.95M -R 246 -X 'n=WH55_CH1,m=FSK_PCM,s=60,l=60,g=1000,r=2500'

Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
time      : 2023-12-20 15:46:21
model     : TFA-303151   Station ID: 50
Msg type  : 5            Battery   : 1             Temperature: 20.0 C       Humidity  : 45 %          Wind avg speed: 4.90 km/h Wind gust : 2.45 km/h
Total rainfall: 30623.0 mm                         Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 15:46:21
model     : WH55_CH1     count     : 1             num_rows  : 1             rows      :
len       : 151          data      : aaaaaaaaaaa2dd45500c82d0402ef3e9411204
codes     : {151}aaaaaaaaaaa2dd45500c82d0402ef3e9411204
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 15:47:22
model     : TFA-303151   Station ID: 50
Msg type  : 5            Battery   : 1             Temperature: 20.0 C       Humidity  : 45 %          Wind avg speed: 4.90 km/h Wind gust : 2.45 km/h
Total rainfall: 30623.0 mm                         Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 15:47:22
model     : WH55_CH1     count     : 1             num_rows  : 1             rows      :
len       : 150          data      : aaaaaaaaaaa2dd45500c82d0402ef3e9411408
codes     : {150}aaaaaaaaaaa2dd45500c82d0402ef3e9411408
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

```

Installed the low-voltage batteries (battery 1 bar, reading less than 1.2VDC each) and re-ran the same above:
```
time      : 2023-12-20 15:57:59
model     : TFA-303151   Station ID: 50
Msg type  : 5            Battery   : 1             Temperature: 20.0 C       Humidity  : 45 %          Wind avg speed: 1.22 km/h Wind gust : 2.45 km/h
Total rainfall: 27679.0 mm                         Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 15:57:59
model     : WH55_CH1     count     : 1             num_rows  : 1             rows      :
len       : 151          data      : aaaaaaaaaaa2dd45500c82d0102d83e1d80204
codes     : {151}aaaaaaaaaaa2dd45500c82d0102d83e1d80204
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 15:59:00
model     : TFA-303151   Station ID: 50
Msg type  : 5            Battery   : 1             Temperature: 20.0 C       Humidity  : 45 %          Wind avg speed: 1.22 km/h Wind gust : 2.45 km/h
Total rainfall: 27807.0 mm                         Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 15:59:00
model     : WH55_CH1     count     : 1             num_rows  : 1             rows      :
len       : 151          data      : aaaaaaaaaaa2dd45500c82d0102d93ee94d204
codes     : {151}aaaaaaaaaaa2dd45500c82d0102d93ee94d204
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 16:00:00
model     : TFA-303151   Station ID: 50
Msg type  : 5            Battery   : 1             Temperature: 20.0 C       Humidity  : 45 %          Wind avg speed: 1.22 km/h Wind gust : 2.45 km/h
Total rainfall: 27935.0 mm                         Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-12-20 16:00:00
model     : WH55_CH1     count     : 1             num_rows  : 1             rows      :
len       : 151          data      : aaaaaaaaaaa2dd45500c82d0102da3ec429204
codes     : {151}aaaaaaaaaaa2dd45500c82d0102da3ec429204

```

cdavis289, 2023-12-20:
Thanks @AhrBee I'll take a look a little later at your data.  I decoded a little more of the signal:

> Family Code (55 for this sensor)
> Channel (0,1,2,3)
> ID (this matches with ecowitt software)
> Bat (number correlates with number of bars in ecowitt software, 1,2,3,4,5)
> Alarm (1 = alarm, 2 = off)
> Sensitivity (254 = alarm on, 190 = high, 062 = low)

[Bitbench](https://triq.net/bitbench#c=5555555555516ea2a8067aa81808b7f5fa80c110%20%5BAlarm%5D&c=5555555555516ea2a8067aa818143df78da1008%20%5B%2B3s%5D&c=5555555555516ea2a8067aa81814adf230bc804%20%5B%2B6s%5D&c=5555555555516ea2a8067aa81814b5f359ec804%20%5B%2B60s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B61s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B60s%2011%3A55%3A59%5D&c=aaaaaaaaaaa8b75154c33d54100a28fbe5c0804%20%5BCH4%20Low%5D&c=aaaaaaaaaaa8b75154c33d54100a30fa8c704020%20%5BCH4%20Low%5D&c=aaaaaaaaaaa8b75154c33d54100a32fb67484120%20%5BCH4%20High%5D&c=5555555555516ea2a9867aa820088ff11f008268%20%5BCH%204%20High%20Alarm%5D&c=&c=&c=&c=&c=&c=5555555555516ea2a8067aa818149df0e759008%20%5Bchanged%20batteries%2C%203bars%5D&c=5555555555516ea2a8067aa818149df0e75c804&c=5555555555516ea2a8067aa81814adf230bc804%20&c=5555555555516ea2a8067aa810147df2e93c804&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b90804%20%5Bbatteries%202%20bars%5D&c=5555555555516ea2a8067aa81814adf230bc8040%20%5Bbatteries%203%20bars%5D&c=5555555555516ea2a8067aa82014cdf301b0805%20%5Bbatteries%204%20bars%5D&c=5555555555516ea2a8067aa82014ddf5cc8c804%20%20%5Bbatteries%204%20bars%5D&c=aaaaaaaaaaa8b75154033d54140a6ef88bee4020%20%5Bbatteries%205%20bars%5D&c=5555555555516ea2a8067aa82814e5f46330804%20%5Bbatteries%205%20bars%5D&c=5555555555516ea2a8067aa81814fdf5441c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e801814fdf6b96c8048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa81814c1f3e681008%20%5BCh%201%2C%20Low%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814b9f3566c804%20%5BCh%202%2C%20Low%2C%20No%20Alarm%5D&c=&c=&c=5555555555516ea2a8067aa81814cdf7963c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814a5f1e8ec804%20%5BCh%202%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9067aa81814adf34ad0804%20%5BCh%203%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9867aa81814cdf292bc804%20%5BCH%204%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa8180bdff55500804%20%5BCh%201%2C%20High%2C%20Alarm%5D&c=5545555555516ea2a8067e80180ae7f53fc9048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20Alarm%5D&c=5555555555516ea2a8067aa8181275f4c111048%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e8018141df06a3c804%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&c=&c=5555555555516ea2a8067aa81814cdf7963c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa8180bdff55500804%20%5BCh%201%2C%20High%2C%20Alarm%5D&f=FAMILY_CODE%20hh%20FLAG%3F%202b%20CH%202d%20ID%2020h%20BAT%20d%20ALARM%20d%20XXX%20d%20SENSITIVITY%20d%20XXX%20hh%20hh%20hh%20hh%20hh&a=Preamble&m=2dd4&cw=4)

cdavis289, 2023-12-21:
@AhrBee I added your data to [bitbench](https://triq.net/bitbench#c=aaaaaaaaaaa2dd45500c82d0402ef3e9411204%20%5BAhrBee%5D&c=aaaaaaaaaaa2dd45500c82d0402ef3e9411408%20%5BAhrBee%5D&c=aaaaaaaaaaa2dd45500c82d0102d83e1d80204%20%5BAhrBee%5D&c=aaaaaaaaaaa2dd45500c82d0102d93ee94d204%20%5BAhrBee%5D&c=aaaaaaaaaaa2dd45500c82d0102da3ec429204%20%5BAhrBee%5D&c=&c=5555555555516ea2a8067aa81808b7f5fa80c110%20%5BAlarm%5D&c=5555555555516ea2a8067aa818143df78da1008%20%5B%2B3s%5D&c=5555555555516ea2a8067aa81814adf230bc804%20%5B%2B6s%5D&c=5555555555516ea2a8067aa81814b5f359ec804%20%5B%2B60s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B61s%5D&c=5555555555516ea2a8067aa81814b5f359ec8048%20%5B%2B60s%2011%3A55%3A59%5D&c=aaaaaaaaaaa8b75154c33d54100a28fbe5c0804%20%5BCH4%20Low%5D&c=aaaaaaaaaaa8b75154c33d54100a30fa8c704020%20%5BCH4%20Low%5D&c=aaaaaaaaaaa8b75154c33d54100a32fb67484120%20%5BCH4%20High%5D&c=5555555555516ea2a9867aa820088ff11f008268%20%5BCH%204%20High%20Alarm%202303.0%20mm%5D&c=5555555555516ea2a9867aa82014c5f1a1cc804%20%20%5BCH4%20High%2019551%20mm%5D&c=&c=&c=&c=&c=&c=&c=5555555555516ea2a8067aa818149df0e759008%20%5Bchanged%20batteries%2C%203bars%5D&c=5555555555516ea2a8067aa818149df0e75c804&c=5555555555516ea2a8067aa81814adf230bc804%20&c=5555555555516ea2a8067aa810147df2e93c804&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b91008&c=5555555555516ea2a8067aa8101475f54b90804%20%5Bbatteries%202%20bars%5D&c=5555555555516ea2a8067aa81814adf230bc8040%20%5Bbatteries%203%20bars%5D&c=5555555555516ea2a8067aa82014cdf301b0805%20%5Bbatteries%204%20bars%5D&c=5555555555516ea2a8067aa82014ddf5cc8c804%20%20%5Bbatteries%204%20bars%5D&c=aaaaaaaaaaa8b75154033d54140a6ef88bee4020%20%5Bbatteries%205%20bars%5D&c=5555555555516ea2a8067aa82814e5f46330804%20%5Bbatteries%205%20bars%5D&c=5555555555516ea2a8067aa81814fdf5441c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e801814fdf6b96c8048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa81814c1f3e681008%20%5BCh%201%2C%20Low%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814b9f3566c804%20%5BCh%202%2C%20Low%2C%20No%20Alarm%5D&c=&c=&c=5555555555516ea2a8067aa81814cdf7963c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8867aa81814a5f1e8ec804%20%5BCh%202%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9067aa81814adf34ad0804%20%5BCh%203%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a9867aa81814cdf292bc804%20%5BCH%204%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa8180bdff55500804%20%5BCh%201%2C%20High%2C%20Alarm%5D&c=5545555555516ea2a8067e80180ae7f53fc9048%20%5BDevice%202%2C%20CH1%2C%20High%2C%20Alarm%5D&c=5555555555516ea2a8067aa8181275f4c111048%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067e8018141df06a3c804%20%5BDevice%202%2C%20CH1%2C%20High%2C%20No%20Alarm%5D&c=&c=5555555555516ea2a8067aa81814cdf7963c804%20%5BCh%201%2C%20High%2C%20No%20Alarm%5D&c=5555555555516ea2a8067aa8180bdff55500804%20%5BCh%201%2C%20High%2C%20Alarm%5D&f=FAMILY_CODE%20hh%20FLAG%3F%202b%20CH%202d%20ID%2020h%20BAT%20d%20ALARM%20d%20XX%20bbbbbbbb%20SENSITIVITY%20d%20XX%20bbbbbbbb%20bbbbbbbb%20bbbbbbbb%20d%20d%20&a=Preamble&m=2dd4&cw=4) and it matches up. Could you paste your output for the sensor with the alarm going off similar to your last post?

AhrBee, 2023-12-21:
@cdavis289 Attached are the details and condensed captures under no alarm conditions - varied channel, sensitivity, battery level. Will try to get to the same captures under alarm conditions after I get sufficient hearing protection (alarm is extremely LOUD) and the household has gone about their day elsewhere in order to not disrupt them. I seemed to have trouble with consistent battery level at low voltage. Fully charged batteries only got to 4 bars, not 5, don't have access to regulated DC power supply. It seems it takes 4 transmits for the WH55 signal to reach a '4' value, starting at 0 (none), then 1,2,3,4 (as noticed in the raw output of the [weewx GW1000 driver](https://github.com/gjr80/weewx-gw1000) used here) 

If any of the conditions need to be re-run, let me know. 

Method:
1. remove batteries, set switches
2. replace batteries, replace WH55 near dongle
3. wait for at least one red flash from WH55, then run command (I varied name)
`rtl_433 -f 914.95M -R 246 -X 'n=WH55_CH1,m=FSK_PCM,s=60,l=60,g=1000,r=2500'`
4. record results

Steps 1-4 repeated for each type of measurement. 

[wh55b_capture_2023-12-21_no-alarm_detail.txt](https://github.com/merbanan/rtl_433/files/13742001/wh55b_capture_2023-12-21_no-alarm_detail.txt)
[wh55b_capture_2023-12-21_no-alarm_condensed.txt](https://github.com/merbanan/rtl_433/files/13742002/wh55b_capture_2023-12-21_no-alarm_condensed.txt)

AhrBee, 2023-12-21:
I've attached the alarm state captures - all except the last one were with freshly charged batteries, last one was with the weak batteries (1 bar). Had to stop, the alarm is driving the dogs nuts, and I have a headache... Will get around to consolidating later, maybe. 

Note that when the alarm triggers, the WH55 immediately transmits - that's the first record in each batch. 

[wh55b_capture_2023-12-21_alarm_detail.txt](https://github.com/merbanan/rtl_433/files/13743910/wh55b_capture_2023-12-21_alarm_detail.txt)

cdavis289, 2023-12-21:
@AhrBee thanks for the data! I consolidated so no need for you to manipulate it. I would say 98% of the values matched. There were a few channel numbers off/battery numbers off. I think for the battery there may have been a delay between the bars changing in the ecowitt website vs the sensor data.

**Changes:**
- Sensitivity: Alarm on value is different when the sensitivity is set to low vs high. When alarm triggers, sensitivity value is increased by 64.
- Alarm: In your data, when the alarm was triggered the value was 0 or 1. When the alarm was off, the value was 2 ([ alarm bitbench](https://triq.net/bitbench#c=%5B%20AhrBee%20ch%201%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45500c82d0400d3fe59781048&c=aaaaaaaaaaa2dd45500c82d0400ccfeaec61824&c=aaaaaaaaaaa2dd45500c82d0400cafef40a1804&c=aaaaaaaaaaa2dd45500c82d0400cafef40a1822&c=&c=%5B%20AhrBee%20ch%201%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45500c82d04011a7e0cf31048&c=aaaaaaaaaaa2dd45500c82d0401177e4c301808&c=aaaaaaaaaaa2dd45500c82d0401167eb89b1824&c=&c=%5B%20AhrBee%20ch%202%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400ee7e5e281048&c=aaaaaaaaaaa2dd45510c82d0400ec7e874f1824&c=aaaaaaaaaaa2dd45510c82d0400ea7edda31822&c=&c=%5B%20AhrBee%20ch%202%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400d4fe78a82010&c=aaaaaaaaaaa2dd45510c82d0400d6fea1d31010&c=aaaaaaaaaaa2dd45510c82d0400cdfed5fe1010&c=&c=%5B%20AhrBee%20ch%203%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400e0fe498510a0&c=aaaaaaaaaaa2dd45510c82d0400dafe154b2020&c=&c=%5B%20AhrBee%20ch%203%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400ce7e822c1048&c=aaaaaaaaaaa2dd45510c82d0400b07ed4601010&c=aaaaaaaaaaa2dd45510c82d0400a17e4ecb1010&c=aaaaaaaaaaa2dd45510c82d0402433e5031204&c=&c=%5B%20AhrBee%20ch%204%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20ON%2008%3A57%3A28%20then%201%20reading%20%20%5D&c=aaaaaaaaaaa2dd45530c82d0300a07e07a21048&c=aaaaaaaaaaa2dd45530c82d0300ca7ebc811010&c=&c=%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20ON%2008%3A59%3A33%20alarm%20OFF%2009%3A01%3A42%20%5D&c=aaaaaaaaaaa2dd45530c82d0301ddfe40991090&c=aaaaaaaaaaa2dd45530c82d0300ddfe065e1048&c=aaaaaaaaaaa2dd45530c82d0300d0fe4691180c&c=aaaaaaaaaaa2dd45530c82d03023ebeec67204%20%5Boff%5D&c=&c=%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20alarm%2C%20updates%20intermittant%20%5D&c=aaaaaaaaaaa2dd45530c82d0100dafeaf021048&c=aaaaaaaaaaa2dd45530c82d0100c3fe023e1804&f=FAMILY_CODE%20hh%20FLAG%3F%202b%20CH%202d%20ID%2020h%20BAT%20d%20ALARM%20hhxxxxxxxx%20XX%20%20SENSITIVITY%20d%20XX%20d%20d%20d%20d%20d%20d&a=Preamble&m=2dd4&cw=4)). I can't see any correlation for the alarm value being 0 vs 1 so I'm trying to see if I'm missing something. 

[no alarm bitbench](https://triq.net/bitbench#c=aaaaaaaaaaa2dd45500c82d0202ddbe8c75204%20%5B%20AhrBee%20ch%201%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45500c82d0402f4be9a9c102%20%5B%20AhrBee%20ch%201%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45500c82d0102d83e1d80102%20%5B%20AhrBee%20ch%201%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45500c82d0402ef3e9411204%20%5B%20AhrBee%20ch%201%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0102dcbe6b62102%20%5B%20AhrBee%20ch%202%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0402f2be4f5f408%20%5B%20AhrBee%20ch%202%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0102d33e88f6102%20%5B%20AhrBee%20ch%202%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0402f03eec7a208%20%5B%20AhrBee%20ch%202%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0102ddbe3f47204%20%5B%20AhrBee%20ch%203%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0402f6be6c90204%20%5B%20AhrBee%20ch%203%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0102d63e5fe0102%20%5B%20AhrBee%20ch%203%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0402fa3ea24a102%20%5B%20AhrBee%20ch%203%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0102d9be3347204%20%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0402f2be6090204%20%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0102dc3e3ed5204%20%5B%20AhrBee%20ch%204%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0402ea3e43eb408%20%5B%20AhrBee%20ch%204%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&f=FAMILY_CODE%20hh%20FLAG%3F%202b%20CH%202d%20ID%2020h%20BAT%20d%20ALARM%20hhhh%20XX%20%20SENSITIVITY%20d%20XX%20d%20d%20d%20d%20d%20d&a=Preamble&m=2dd4&cw=4)

**Current Summary**
Family Code (55 for this sensor)
Channel (0,1,2,3) corresponds to (1,2,3,4)
ID (this matches with ecowitt software)
Bat (number correlates with number of bars in ecowitt software, 1,2,3,4,5)
Alarm (0 = alarm, 1 = alarm, 2 = alarm off)
Sensitivity (when alarm is triggered, low or high value is increased by 64)
- 254 = alarm on, high sensitivity
- 190 = alarm off, high sensitivity
- 126 = alarm on, low sensitivity
- 062 = alarm off, low sensitivity

I'll play around with the data some more in bitbench, but I'm starting to run out of ideas.

cdavis289, 2023-12-21:
**Update:** I combined CH and ID so that the device ID matches what is shown in the Ecowitt web interface. For example, 30CF55 would be a device on CH 4 while 00CF55 would be the same device on CH 1.

@zuckschwerdt How completely does the signal need to be decoded to get implemented into rtl_433? I have identified all of the information shown in the Ecowitt software.

### **Current Summary**
**Family Code:** 55 for this sensor

**Channel and ID:** ABBBBB
- A is channel: 0,1,2,3 corresponds to channel 1,2,3,4
- BBBB is the remaining device ID (ie 0CF55)

**BATT:** number correlates with number of bars in ecowitt software, 1,2,3,4,5
**ALARM:** This appears to be a measurement. > 2000 equates to no alarm (2000 is not exact). When I placed the sensor on my hand to trigger the alarm, this value dropped to 1000-1200. @Ahrbee had values around 200 when the alarm was triggered. I'm assuming they placed the sensor into water.

**SENSITIVITY:** When alarm is triggered, low or high value is increased by 64
- 254 = alarm on, high sensitivity
- 190 = alarm off, high sensitivity
- 126 = alarm on, low sensitivity
- 062 = alarm off, low sensitivity

[bitbench](https://triq.net/bitbench#c=aaaaaaaaaaa2dd45500c82d0202ddbe8c75204%20%5B%20AhrBee%20ch%201%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45500c82d0402f4be9a9c102%20%5B%20AhrBee%20ch%201%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45500c82d0102d83e1d80102%20%5B%20AhrBee%20ch%201%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45500c82d0402ef3e9411204%20%5B%20AhrBee%20ch%201%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0102dcbe6b62102%20%5B%20AhrBee%20ch%202%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0402f2be4f5f408%20%5B%20AhrBee%20ch%202%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0102d33e88f6102%20%5B%20AhrBee%20ch%202%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45510c82d0402f03eec7a208%20%5B%20AhrBee%20ch%202%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0102ddbe3f47204%20%5B%20AhrBee%20ch%203%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0402f6be6c90204%20%5B%20AhrBee%20ch%203%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0102d63e5fe0102%20%5B%20AhrBee%20ch%203%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45520c82d0402fa3ea24a102%20%5B%20AhrBee%20ch%203%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0102d9be3347204%20%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0402f2be6090204%20%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0102dc3e3ed5204%20%5B%20AhrBee%20ch%204%2C%20low%20sensitivity%2C%20batt%201%20bar%2C%20normal%20%5D&c=aaaaaaaaaaa2dd45530c82d0402ea3e43eb408%20%5B%20AhrBee%20ch%204%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20normal%20%5D&c=&c=%5B%20AhrBee%20ch%201%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45500c82d0400d3fe59781048&c=aaaaaaaaaaa2dd45500c82d0400ccfeaec61824&c=aaaaaaaaaaa2dd45500c82d0400cafef40a1804&c=aaaaaaaaaaa2dd45500c82d0400cafef40a1822&c=&c=%5B%20AhrBee%20ch%201%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45500c82d04011a7e0cf31048&c=aaaaaaaaaaa2dd45500c82d0401177e4c301808&c=aaaaaaaaaaa2dd45500c82d0401167eb89b1824&c=&c=%5B%20AhrBee%20ch%202%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400ee7e5e281048&c=aaaaaaaaaaa2dd45510c82d0400ec7e874f1824&c=aaaaaaaaaaa2dd45510c82d0400ea7edda31822&c=&c=%5B%20AhrBee%20ch%202%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400d4fe78a82010&c=aaaaaaaaaaa2dd45510c82d0400d6fea1d31010&c=aaaaaaaaaaa2dd45510c82d0400cdfed5fe1010&c=&c=%5B%20AhrBee%20ch%203%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400e0fe498510a0&c=aaaaaaaaaaa2dd45510c82d0400dafe154b2020&c=&c=%5B%20AhrBee%20ch%203%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20%5D&c=aaaaaaaaaaa2dd45510c82d0400ce7e822c1048&c=aaaaaaaaaaa2dd45510c82d0400b07ed4601010&c=aaaaaaaaaaa2dd45510c82d0400a17e4ecb1010&c=aaaaaaaaaaa2dd45510c82d0402433e5031204&c=&c=%5B%20AhrBee%20ch%204%2C%20low%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20ON%2008%3A57%3A28%20then%201%20reading%20%20%5D&c=aaaaaaaaaaa2dd45530c82d0300a07e07a21048&c=aaaaaaaaaaa2dd45530c82d0300ca7ebc811010&c=&c=%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%204%20bar%2C%20alarm%20ON%2008%3A59%3A33%20alarm%20OFF%2009%3A01%3A42%20%5D&c=aaaaaaaaaaa2dd45530c82d0301ddfe40991090&c=aaaaaaaaaaa2dd45530c82d0300ddfe065e1048&c=aaaaaaaaaaa2dd45530c82d0300d0fe4691180c&c=aaaaaaaaaaa2dd45530c82d03023ebeec67204%20%5Boff%5D&c=&c=%5B%20AhrBee%20ch%204%2C%20high%20sensitivity%2C%20batt%201%20bar%2C%20alarm%2C%20updates%20intermittant%20%5D&c=aaaaaaaaaaa2dd45530c82d0100dafeaf021048&c=aaaaaaaaaaa2dd45530c82d0100c3fe023e1804&f=FAMILY_CODE%20hh%20FLAG%3F%202b%20CH-ID%202d20h%20BAT%20d%20MEASUREMENT%3F%20dd%20%20SENSITIVITY%20d%20%3F%3F%20d%20d%20d%20d%20d%20d&a=Preamble&m=2dd4&cw=4)

zuckschwerdt, 2023-12-21:
I need to take a look at the checksum, but with the collision it seems already figured out.

The alarm/sensitivity looks like a bit-field, first bit sensitivity (1 for high) and second bit alarm (1 for alarm).

Looks like good and thorough documentation so far. Do you want to try coding a decoder or should I just add one?

cdavis289, 2023-12-22:
@zuckschwerdt I'll take a shot at coding a decoder.

cdavis289, 2023-12-22:
@zuckschwerdt I started to create the decoder and quickly realized my knowledge of C is lacking so I was only able to set up the signal description part of the file.

[wh55.zip](https://github.com/merbanan/rtl_433/files/13756191/wh55.zip)

zuckschwerdt, 2023-12-24:
Can you add a cu8 recording as zip here for testing?

zuckschwerdt, 2023-12-24:
> Can you add a cu8 recording as zip here for testing?

You already did, sorry, nevermind.

closed 2023-12-24 by zuckschwerdt

AhrBee, 2023-12-24:
> Can you add a cu8 recording as zip here for testing?
Just for grins I ran a collection using `rtl_433 -f 915.05M -S all` channel 1 low sensitivity battery 4 bars, included was an alarm condition ON for about a minute then OFF.
🎄
[test-2023-12-24.zip](https://github.com/merbanan/rtl_433/files/13762174/test-2023-12-24.zip)

zuckschwerdt, 2023-12-24:
Thanks! The decoder is now available. Let me know if any problems come up.

AhrBee, 2023-12-25:
Follow up that confirms the commit appears to be working as intended. I filtered out other detected protocols. WH55 state is battery 80%, low sensitivity, no alarm, located one floor down in maintenance closet in hot water tank catch tray.

```
# rtl_433  -f 915.05M -C customary -Mbits -Mlevel -Mprotocol -T 90
rtl_433 version 23.11-18-gcf4c74de branch master at 202312241630 inputs file rtl_tcp RTL-SDR

Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz

...

time      : 2023-12-25 07:04:31                    Protocol  : 251
model     : Fineoffset-WH55                        ID        : 0C82D
Channel   : 1            Battery   : 0.800         Raw Value : 728           Sensitivity: 0            Alarm     : 0             Integrity : CRC
Modulation: FSK          Freq1     : 914.9 MHz     Freq2     : 915.0 MHz
RSSI      : -0.1 dB      SNR       : 31.3 dB       Noise     : -31.4 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

```

commit 30d7dff2 (2023-12-24): Add support for Fine Offset / Ecowitt WH55 water leak sensor (closes #2756)
commit 7b88ef5a (2023-12-24): Add support for Fine Offset / Ecowitt WH55 water leak sensor (closes #2756)
commit cf4c74de (2023-12-24): Add support for Fine Offset / Ecowitt WH55 water leak sensor (closes #2756)