merbanan/rtl_433 #2757: hackrf detection problem on windows

danyhm, 2023-12-19:
Hello,

same as #1530 , i have problems detecting hackrf in windows 10.

copying `rtl_433-rtlsdr-soapysdr.exe` to `PothosSDR\bin` installation solves the problem , however I just wanted to point out this problem so a solution can be found so we can use the `rtl_433-rtlsdr-soapysdr.exe` without PothosSDR installation.

zuckschwerdt, 2023-12-20:
We are using Pothos 2021.07.25-vc16 to build. Which version did you install? Maybe it's just a different major version and thus the modules can't be loaded.

danyhm, 2023-12-21:
I'm using the same version. I even replaced the DLL's from pothos but nothing changed

zuckschwerdt, 2023-12-21:
Placing `rtl_433-rtlsdr-soapysdr.exe` in `PothosSDR\bin` works, but copying the `bin/SoapySDR.dll` to the `rtl_433-rtlsdr-soapysdr.exe` dir does not? Strange. Not sure how to debug that.

danyhm, 2023-12-21:
yes that is exactly what is happening. if i run the default release of rtl_433 i get the following:

```
C:\Users\xxx\Desktop\rtl>rtl_433-rtlsdr-soapysdr.exe
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR
[SDR] No supported devices found.

C:\Users\xxx\Desktop\rtl>rtl_433-rtlsdr-soapysdr.exe -d "driver=hackrf"
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR
[sdr_open_soapy] Failed to open sdr device matching 'driver=hackrf'.
```

and if i copy everything from pothos/bin to rtl directory i get no errors but again nothing happens
```
C:\Users\xxx\Desktop\rtl>rtl_433-rtlsdr-soapysdr.exe -d "driver=hackrf"
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR
Reading conf from "C:\Users\xxx\Desktop\rtl\rtl_433.conf".
Use "-F log" if you want any messages, warnings, and errors in the console.

C:\Users\xxx\Desktop\rtl>
```

and if i copy from rtl_433 to pothos bin it works
```
C:\Program Files\PothosSDR\bin>rtl_433-rtlsdr-soapysdr.exe -d driver=hackrf
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR
Reading conf from "C:\Program Files\PothosSDR\bin\rtl_433.conf".
Use "-F log" if you want any messages, warnings, and errors in the console.
Using device HackRF One: clock source=internal part id=a000cb3c006c4f56 serial=0000000000000000909864c8378229cf version=2023.01.1
Found 1 antenna(s): TX/RX
Found 3 gain(s): LNA 0 - 116 (step 0) AMP 0 - 116 (step 0) VGA 0 - 116 (step 0)
Found 1 frequencies: RF
Found 1 frequency range(s): 0 - 7250000000 (step 0)
Found 20 sample rate range(s): 1000000 2000000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000 11000000 12000000 13000000 14000000 15000000 16000000 17000000 18000000 19000000 20000000
Found 16 bandwidth range(s): 1750000 - 1750000 (step 0) 2500000 - 2500000 (step 0) 3500000 - 3500000 (step 0) 5000000 - 5000000 (step 0) 5500000 - 5500000 (step 0) 6000000 - 6000000 (step 0) 7000000 - 7000000 (step 0) 8000000 - 8000000 (step 0) 9000000 - 9000000 (step 0) 10000000 - 10000000 (step 0) 12000000 - 12000000 (step 0) 14000000 - 14000000 (step 0) 15000000 - 15000000 (step 0) 20000000 - 20000000 (step 0) 24000000 - 24000000 (step 0) 28000000 - 28000000 (step 0)
Found current bandwidth 0
Found 4 stream format(s): CS8 CS16 CF32 CF64
Found native stream format: CS8 (full scale: 128.0)
Signal caught, exiting!
{"time" : "2023-12-21 14:12:31.647734", "enabled" : 8, "since" : "2023-12-21T14:12:29", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}

C:\Program Files\PothosSDR\bin>
```

zuckschwerdt, 2023-12-21:
Can you run `SoapySDRUtil --info` in both dirs (needs to be copied to rtl_433 dir)?
Is the Search Path maybe different?

danyhm, 2023-12-21:
```
C:\Users\xxx\Desktop\rtl>SoapySDRUtil.exe --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-PothosSDR-2021.07.25-vc16-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\Users\xxx\Desktop
Search path:  C:\Users\xxx\Desktop/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

C:\Userr\xxx\Desktop\rtl>
```

and

```
C:\Program Files\PothosSDR\bin>SoapySDRUtil.exe --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-PothosSDR-2021.07.25-vc16-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\Program Files\PothosSDR
Search path:  C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll (0.2.0-d682533)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll   (0.2.0-411f73e)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/audioSupport.dll    (0.1.1-91080cb)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/bladeRFSupport.dll  (0.4.1-70505a5)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll   (0.3.3-8d2e7be)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/IrisSupport.dll     (2020.02.0.1-f100723)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll     (20.10.0-a45e482d)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll     (0.2.6-585c012)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/netSDRSupport.dll   (0.1.0-51516db)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll  (0.2.6-585c012)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.2.1-c880222)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/RedPitaya.dll       (0.1.1-3d576f8)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/remoteSupport.dll   (0.6.0-c09b2f1)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll   (0.3.2-53ee8f4)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll  (0.3.0-206b241)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll      (0.4.1-9a738c3)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/volkConverters.dll[WARNING] SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.
  (0.1.0-62ac7f5)
Available factories... airspy, airspyhf, audio, bladerf, hackrf, iris, lime, miri, netsdr, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd
Available converters...
 -  CF32 -> [CF32, CF64, CS16, CS32, CS8, CU16, CU8]
 -  CF64 -> [CF32, CS16, CS32, CS8]
 -  CS16 -> [CF32, CF64, CS16, CS8, CU16, CU8]
 -  CS32 -> [CF32, CF64, CS32]
 -   CS8 -> [CF32, CF64, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
[... 40 lines cut ...]
Module found: C:\Users\xxx\Desktop/lib/SoapySDR/modules0.8/volkConverters.dll[WARNING] SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.
  (0.1.0-62ac7f5)
Available factories... airspy, airspyhf, audio, bladerf, hackrf, iris, lime, miri, netsdr, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd
Available converters...
 -  CF32 -> [CF32, CF64, CS16, CS32, CS8, CU16, CU8]
 -  CF64 -> [CF32, CS16, CS32, CS8]
 -  CS16 -> [CF32, CF64, CS16, CS8, CU16, CU8]
 -  CS32 -> [CF32, CF64, CS32]
 -   CS8 -> [CF32, CF64, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, F64, S16, S32, S8, U16, U8]
 -   F64 -> [F32, S16, S32, S8]
 -   S16 -> [F32, F64, S16, S8, U16, U8]
 -   S32 -> [F32, F64, S32]
 -    S8 -> [F32, F64, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

C:\Users\xxx\Desktop\rtl>rtl_433-rtlsdr-soapysdr.exe -d "driver=hackrf"
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR
Reading conf from "C:\Users\Danial\Desktop\rtl\rtl_433.conf".
Use "-F log" if you want any messages, warnings, and errors in the console.

```

zuckschwerdt, 2023-12-21:
So the module search path is relative to the Soapy library. That explains it. The only way then is to copy the rtl_433 executable to the Pothos bin path.

I should add a note to the bundled release note.

commit 57c22d7c (2023-12-21): docs: Add SoapySDR note to release (closes #2757)
closed 2023-12-21 by zuckschwerdt