merbanan/rtl_433 #2758: Add: Efergy emtx (E-Max transmitter)

mscottco, 2023-12-19:
Hoping to get support for the Efergy EMTX transmitter that comes as part of the E-Max energy monitor. I've uploaded some tests to the repo and submitted a pull request: https://github.com/merbanan/rtl_433_tests/pull/460

I've tried to follow some of the steps to create a flex decoder but gotten a bit stuck. 
`rtl_433 -w OOK:- g2246_433.92M_250k.cu8`
Will give me a readout of the pulses

And my attempt at a flex decoder:
`rtl_433 -X 'n=name,m=OOK_PWM,s=2152,l=5028,r=60,g=0,t=1150,y=0' g2246_433.92M_250k.cu8`

Gives the following (formatted as json):
`{"time" : "@0.136884s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 3, "data" : "6"}], "codes" : ["{3}6"]}`

But the actual specifics are beyond me. I'd expect the device to give the same or similar data readouts to the Efergy c2...

Happy to provide any further information or testing as needed.

zuckschwerdt, 2023-12-20:
Both samples contain some transmission but there is no usable carrier/data in there, just noise. Maybe try lower (`-f 433.8M`) or with higher sample rate (`-s 1024k`).

mscottco, 2023-12-20:
Thanks, I've added a couple of samples. I used the IQ Spectrogram and can see there's what looks to be a more usable signal there, does that help?

zuckschwerdt, 2023-12-20:
Yes, those signals are good. But there is clipping. Increase the distance or remove the receiver antenna for a cleaner signal. It already works though (see in pdv), looks like 84/140 µs PWM with preceeding gaps.

mscottco, 2023-12-20:
Thanks, I've added some extra tests without the antenna / greater range (more tests is better I assume...). I also set the frequency based on the specific frequency the device is supposed to use.

I've had a go at writing a flex decoder but the data output doesn't make a lot of sense to me... Any ideas if this is due to a poor sample or the decoder is written wrong? The command and output are below:

`rtl_433 -X 'n=emtx,m=FSK_PWM,s=84,l=140,r=7000' g052_433.55M_1024k.cu8`

```
time      : @0.323848s
model     : emtx         count     : 1             num_rows  : 1             rows      : 
len       : 37           data      : 0000000000
codes     : {37}0000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.323848s
model     : emtx         count     : 1             num_rows  : 1             rows      : 
len       : 65           data      : 00000000000000000
codes     : {65}00000000000000000
```

ProfBoc75, 2023-12-22:
> Yes, those signals are good. But there is clipping. Increase the distance or remove the receiver antenna for a cleaner signal. It already works though (see in pdv), looks like 84/140 µs PWM with preceeding gaps.

Hi Christian, from my side, I can see 62/140 µs (84x61.7µs short pulses) and I found more information on the web,

[Efergy e-max user guide](https://efergy.com/wp-content/uploads/2020/10/E-Max-Quick-Start-Guide-16102020-V2-Curves.pdf) confirmed it's compatible with Efergy E2 and I found information about E2 in another github project, which confirmed short = 60 µs, long = 140 µs, preamble 1800 µs LOW (= gap), and 500 µs HIGH ( = sync) very close to what we can see in pdv. Needs also to invert the data to get the good values.

So, the flex decoder is for me :

`rtl_433 -X "n=efergy,m=FSK_PWM,s=62,l=140,g=1800,sync=500,r=7000,invert" *.cu8` 

result :

[image]

Data = 00 18 d3 40 00 00 f8 23
Fixed = 0x0018d340
Current = 0x0000  --> Sounds like the sensor was not installed , so no current here ?
Exponent = 0xf8
CRC (CHECKSUM) = 0x23 , which is the exact sum of 7 previous bytes.

[bitbench](https://triq.net/bitbench#c=ffe72cbfffff07dc&f=ID%3A32h%20CURRENT%3A16h%20EXP%3A8h%20CRC%3A8h%20&i=true&cl=5&cw=8)

Source :
[MySensor Efergy E2](https://github.com/gtortone/MySensors-Efergy-e2/blob/34bda46a7fc8fba19923e39a90917ff4f5c56751/MySensors-Efergy-e2.ino#L173) :

/*
 * Efergy e2 power monitor http://efergy.com/it/products/electricity-monitors/e2-classic
 * transmission protocol features: 
 * 
 * frequency: 433.535 MHz
 * bandwith: 67 kHz
 * data rate: 10142 bps
 * 
 * preamble: ~1800 us LOW - ~500 us HIGH
 * 
 * payload: 
 * - '0' => 140 us LOW - 60 us HIGH
 * - '1' => 60 us LOW - 140 HIGH
 * 
 * payload format:
 * 
 * 8 bytes 
 * ------- 
 * bytes # 1 :  0x09    fixed value
 * bytes # 2 :  0XF0    fixed value
 * bytes # 3 :  0x2F    fixed value
 * bytes # 4 :  0x40    fixed value
 * bytes # 5 :  current MSB
 * bytes # 6 :  current LSB
 * bytes # 7 :  exponent 
 * bytes # 8 :  CRC
 * 
 * current detected = ((MSB << 8) + LSB / 32768) * pow(2, exponent)
 * 
 * example:
 * 
 * 0x09 0xF0 0x2F 0x40 0x6B 0x85 0x01 0x59
 * 
 * current = (0x6B85 / 32768) * (2^1) = 27525/32768 * 2 = 0.839 * 2 = 1.67 A
 * 
 */

ProfBoc75, 2023-12-22:
I also found other information into github, an [Arduino library for Efergy](https://github.com/hardtechnology/efergy_v2/blob/master/efergy.cpp) that could give more information, like the battery flag, signal frequency and so on ...

ProfBoc75, 2023-12-23:
I got it , little update line 93 into the decoder [efergy_e2_classic.c](https://github.com/merbanan/rtl_433/blob/57c22d7c73b88c7ed31cee303182d43a8606f772/src/devices/efergy_e2_classic.c#L93) 

Replace fact > 20 by fact > 23 and it will work.

`    if (fact < 7 || fact > 23) // full range unknown so far`

Fact (exponent) value must stay as small as possible since behind there is a power 2 exponent, higher the value, more long time it will take and could hang rtl_433 if too much.

Result with the correction :

[image]

mscottco, 2023-12-24:
Amazing, thank for your help!

I've tried installing the latest build using the instructions at https://triq.org/rtl_433/#building-installation however rtl_433 doesn't seem to recognise the transmitter. I've starting with all the logging options enabled (-v -vv -vvv -A) and get the following output (if its helpful). Have I missed something or made an error?

```
Detected FSK package	2023-12-24T15:44:25+1100
Exact bit width (in us) is 61.60 vs 99.61, 66 bit preamble
Exact bit width (in us) is 61.60 vs 104.49, 66 bit preamble
Exact bit width (in us) is 61.09 vs 54.69, 68 bit preamble
Exact bit width (in us) is 61.60 vs 57.62, 66 bit preamble
Exact bit width (in us) is 61.09 vs 51.76, 68 bit preamble
Exact bit width (in us) is 61.09 vs 51.76, 68 bit preamble
Exact bit width (in us) is 61.09 vs 51.76, 68 bit preamble
Exact bit width (in us) is 61.09 vs 51.76, 68 bit preamble
Exact bit width (in us) is 61.09 vs 51.76, 68 bit preamble
M-Bus: CRC error: Calculated 0xFFFF, Read: 0x0
M-Bus: CRC error: Calculated 0xFFFF, Read: 0x0
Exact bit width (in us) is 61.60 vs 99.61, 66 bit preamble
Exact bit width (in us) is 61.09 vs 55.66, 68 bit preamble
bresser_5in1_decode: bit_per_row 137 out of range
directv: incorrect number of bits in bitbuffer: 9 (expected between 44 and 99).
WARNING: Undeclared field "msg" in [130] "IKEA Sparsnas Energy Meter Monitor"
WARNING: Undeclared field "num_rows" in [130] "IKEA Sparsnas Energy Meter Monitor"
WARNING: Undeclared field "codes" in [130] "IKEA Sparsnas Energy Meter Monitor"
{"time" : "2023-12-24T15:44:25+1100", "msg" : "ikea_sparsnas_decode: ", "num_rows" : 4, "codes" : ["{174}99999999999999999999999999999999998000000000", "{71}c0000000c000000000", "{503}ffffc183060c183060c183067cf83060cf9f067c1833e7c19f060c183060c183060c183060c183060c183060cf9f3e7cf83060c1833e0c1833e7c000000000", "{0}"]}
ikea_sparsnas_decode: malformed package, preamble not found. (Expected 0xAAAAD201)
Exact bit width (in us) is 61.60 vs 99.61, 66 bit preamble
Exact bit width (in us) is 61.60 vs 103.52, 66 bit preamble
Exact bit width (in us) is 61.09 vs 48.83, 68 bit preamble
Exact bit width (in us) is 61.60 vs 57.62, 66 bit preamble
Fineoffset_WH51: short package. Header index: 378
bitbuffer:: Number of rows: 1 
[00] {354} 55 55 55 55 55 55 55 55 50 00 00 40 01 00 00 00 01 fe 49 24 92 49 6c 92 d9 64 b6 59 24 92 49 24 92 49 24 92 db 6c 92 4b 24 b6 00 00 00 
Exact bit width (in us) is 61.60 vs 57.62, 66 bit preamble
fineoffset_wh1080: short package. Header index: 378
bitbuffer:: Number of rows: 1 
[00] {354} 55 55 55 55 55 55 55 55 50 00 00 40 01 00 00 00 01 fe 49 24 92 49 6c 92 d9 64 b6 59 24 92 49 24 92 49 24 92 db 6c 92 4b 24 b6 00 00 00 
Exact bit width (in us) is 61.09 vs 51.76, 68 bit preamble
Exact bit width (in us) is 61.60 vs 109.38, 66 bit preamble
insteon_callback: new buffer 1 rows
insteon_callback: new buffer 2 rows
insteon_callback: new buffer 1 rows
Exact bit width (in us) is 61.60 vs 106.45, 66 bit preamble
lacrosse_breezepro_decode: Sync word not found
Exact bit width (in us) is 61.60 vs 103.52, 66 bit preamble
lacrosse_wr1_decode: Packet too long: 354 bits
Exact bit width (in us) is 61.60 vs 103.52, 66 bit preamble
lacrosse_th_decode: Packet too long: 354 bits
bitbuffer:: Number of rows: 1 
[00] {354} 55 55 55 55 55 55 55 55 50 00 00 40 01 00 00 00 01 fe 49 24 92 49 6c 92 d9 64 b6 59 24 92 49 24 92 49 24 92 db 6c 92 4b 24 b6 00 00 00 : 01010101 01010101 01010101 01010101 01010101 01010101 01010101 01010101 01010000 00000000 00000000 01000000 00000001 00000000 00000000 00000000 00000001 11111110 01001001 00100100 10010010 01001001 01101100 10010010 11011001 01100100 10110110 01011001 00100100 10010010 01001001 00100100 10010010 01001001 00100100 10010010 11011011 01101100 10010010 01001011 00100100 10110110 00000000 00000000 00
bresser_6in1_decode: bit_per_row 137 out of range
bresser_7in1_decode: to few bits (137)
ecodhome_decode: to few bits (87)
Exact bit width (in us) is 61.60 vs 103.52, 66 bit preamble
lacrosse_r1_decode: Packet too long: 354 bits
Pulse data: 103 pulses
[  0] Pulse:   29, Gap:   62, Period:   91
[  1] Pulse:   63, Gap:   63, Period:  126
[  2] Pulse:   63, Gap:   63, Period:  126
[  3] Pulse:   63, Gap:   63, Period:  126
[  4] Pulse:   63, Gap:   63, Period:  126
[  5] Pulse:   63, Gap:   63, Period:  126
[  6] Pulse:   63, Gap:   63, Period:  126
[  7] Pulse:   63, Gap:   64, Period:  127
[  8] Pulse:   63, Gap:   63, Period:  126
[  9] Pulse:   63, Gap:   63, Period:  126
[ 10] Pulse:   63, Gap:   63, Period:  126
[ 11] Pulse:   63, Gap:   64, Period:  127
[ 12] Pulse:   62, Gap:   63, Period:  125
[ 13] Pulse:   63, Gap:   64, Period:  127
[... 102 lines cut ...]
 [ 4] count:   48,  width:  139 us [139;140]	( 142 S)
Pulse period distribution:
 [ 0] count:    1,  width:   89 us [89;89]	(  91 S)
 [ 1] count:   41,  width:  123 us [122;124]	( 126 S)
 [ 2] count:    2,  width: 1346 us [1310;1382]	(1378 S)
 [ 3] count:    1,  width:  861 us [861;861]	( 882 S)
 [ 4] count:    1,  width: 1981 us [1981;1981]	(2029 S)
 [ 5] count:    1,  width:  632 us [632;632]	( 647 S)
 [ 6] count:   48,  width:  200 us [199;202]	( 205 S)
 [ 7] count:    7,  width:  279 us [279;280]	( 286 S)
Pulse timing distribution:
 [ 0] count:    1,  width:   28 us [28;28]	(  29 S)
 [ 1] count:  134,  width:   62 us [60;62]	(  63 S)
 [ 2] count:    1,  width:  492 us [492;492]	( 504 S)
 [ 3] count:   64,  width:  139 us [139;141]	( 142 S)
 [ 4] count:    1,  width:    9 us [9;9]	(   9 S)
 [ 5] count:    2,  width: 1245 us [1169;1321]	(1275 S)
 [ 6] count:    1,  width:  801 us [801;801]	( 820 S)
 [ 7] count:    1,  width: 1921 us [1921;1921]	(1967 S)
 [ 8] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  14617,     11
RSSI: -0.5 dB SNR: 31.2 dB Noise: -31.7 dB
Frequency offsets [F1, F2]:  -17914, -26971	(-279.9 kHz, -421.4 kHz)
Guessing modulation: No clue...
```

ProfBoc75, 2023-12-24:
@mscottco : strange, because I tested with your cu8 captures.

Can you share your command line ?

Try with lower frequency, because by default it's 433.92 MHz @ 250 Khz (default sample rate) , this means from 433.8 to 434MHz , and Efergy sensor sounds around 433.5 Mhz so out of scope.

Try this :
`rtl_433 -f 433.5M -M level`

Or keep 433.92 Mhz as central frequency and increase the sample rate @ 1000 KHz to capture from 433.42 to 434.42 Mhz 

`rtl_433 -s 1000K -M level`

Or mix both , provide the frequency and the sample rate if you need to capture other sensors / devices.

The `-M level` option will give you the exact frequencies, so you can adjust for your needs.

mscottco, 2023-12-24:
I just logged back onto the machine with the SDR to get the command (pasted below). I re-ran the command and it now works. Maybe just needed a fresh session for the new version to take effect.

```rtl_433 -Mtime:iso:tz:local -Fmqtt://xxx:1883,retain=1,user=xxx,pass=xxx,devices=rtl_433[/type][/model][/subtype][/channel][/id] -Fjson -d:1433 -s1024k -v -vv -vvv -A -f433.55M```

Thanks again!

gdt, 2024-06-03:
Status?

gdt, 2024-06-23:
@ProfBoc75 Submitter is not responding and I'd like to move this is closed.  Is there a new flex decoder somewhere in the discussion that should get added to the source tree, or does git master work, or ?

mscottco, 2024-06-25:
My apologies, I missed your previous message. Yes the revised version seems to be working well so I'm happy the issue can be closed.

closed 2024-06-25 by mscottco