merbanan/rtl_433 #2761: Nothing logged to mqtt when reading from stdin

pimlie, 2023-12-23:
## What Im trying to do
Im trying to read both wmbus as other 868M devices at the same time, for that I'd like to pipe the rtlsdr signal to both wmbusmeters as to rtl_433. I've got this working with the following command:

```bash
$ rtl_433 -R 0 -f 868.95M -s 1600k -w- 2>/dev/null | tee >(rtl_wmbus -p s -a | wmbusmeters --useconfig=/wmbusmeters_data/ --overridedevice=stdin:rtlwmbus) >(rtl_433 -f 868.95M -s 1600k -R -104 -R -105 -R -106 -R -238 -F "mqtt://mosquitto:1883,user=USER,pass=PASS,retain=1" -r- 2>/dev/null) >/dev/null
```

The only issue is that nothing gets ever logged to mqtt from rtl_433. The issue seems to be related to reading from stdin.

## What works

✅ Logging to mqtt works when not piping/reading from stdin rtl_433 as follows:
```bash
$ rtl_433 -f 868.95M -s 1600k -F"mqtt://mosquitto:1883,user=USER,pass=PASS,retain=1"
```
✅ Logging json to a file also works when piping rtl_433 as follows:
```bash
$ rtl_433 -f 868.95M -s 1600k -R 0 -w- | rtl_433 -f 868.95M -s 1600k -Fjson:rtl433.log -Fkv -r-
```

## What doesn't work

When I then try to also log to mqtt then nothing ever gets published to mqtt, even though there is still output in the console and in the json log file.

:x: Logs to console & json file, but **not** to mqtt:
```bash
$ rtl_433 -f 868.95M -s 1600k -R 0 -w- | rtl_433 -f 868.95M -s 1600k -Fjson:rtl433.log -Fkv -F"mqtt://mosquitto:1883,user=USER,pass=PASS,retain=1" -r-
```

Please note there is no issue with the mqtt config as the first command that doesnt read from stdin succesfully publishes to mqtt with the exact same mqtt config :)

Could this be related to the notice `[Input] Test mode active. Reading samples from file: <stdin>`?

#### Version info
```bash
$ rtl_433 -V
rtl_433 version 23.11-11-g7e6a4baf branch master at 202312031159 inputs file rtl_tcp RTL-SDR
```

zuckschwerdt, 2023-12-23:
Reduce complexity by using the rtl_tcp output:
`rtl_433 -f 868.95M -s 1600k -R -104 -R -105 -R -106 -R -238 -v -F log -F rtl_tcp -F "mqtt://mosquitto:1883,user=USER,pass=PASS,retain=1"`

and then feed wmbus from tcp somehow, e.g.
`socat tcp:localhost:1234 - | rtl_wmbus -p s -a | wmbusmeters --useconfig=/wmbusmeters_data/ --overridedevice=stdin:rtlwmbus`

closed 2023-12-23 by zuckschwerdt

pimlie, 2023-12-23:
Ah wow, overlooked that nice solution. Can confirm it works, thanks for the quick solve! 🎉