merbanan/rtl_433 #2762: Nooelec SMArt v5 - weak decoder performance

rauar, 2023-12-24:
Got lately into SDR and bought a Nooelec NESDR SMArt v5 SDR. My antenna is a stubby which I took from a Lilygo Lora board (430MHz).

I have a temperature sensor (outdoor - roughly 5 meters away) and the receiver station has no issues receiving the temperature although it's even further away than my SDR dongle (I'm physically playing man-in-the-middle with my SDR setup).

When I launch rtl_433 without any parameters I don't see any decoded messages. So I started playing around with some of the parameters like gain, level, direct_agc etc. and after a lot of trial and error I was able to randomly receive some messages. However this approach didn't feel right so I started looking at the raw data with GQRX and triq.

GQEX at 433.920MHz:
[image]

There's a signal around 433.90MHZ but what confused me is the carrier at the center frequency. So changed the frequency a bit and again the carrier appears on the selected center frequency. Am I right in the assumption that the SDR receiver has internal noise and that this could be a source for potential decoding issues? Is this "carrier" on the center frequency a known effect?

[image]

For comparison here's a sample taken with rtl_433 without any parameters:

[image]

The frequency scale looks weird but otherwise I'd say it's similar to what GQRX shows...

I know that the performance of stubby antennas is close to bad - however the antenna in the receiver station is probably not much more powerful than a stubby and the receiver station has 1 thick wall as attenuator, a plastic case and not even direct line of sight.

Noise level measurements:

rtl_433 version 23.11-14-g60bdd62d branch master at 202312181113 inputs file rtl_tcp RTL-SDR
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[Auto Level] Current noise level -38.1 dB, estimated noise -27.9 dB \
[Auto Level] Current noise level -38.2 dB, estimated noise -37.3 dB
[Auto Level] Current noise level -38.2 dB, estimated noise -38.1 dB
[Auto Level] Current noise level -38.1 dB, estimated noise -38.2 dB
[Auto Level] Current signal level -32.7 dB, estimated noise -37.2 dB
[Auto Level] Current noise level -37.7 dB, estimated noise -35.8 dB
[Auto Level] Current noise level -38.1 dB, estimated noise -37.5 dB
[Auto Level] Current noise level -38.1 dB, estimated noise -38.1 dB
[Auto Level] Current noise level -38.2 dB, estimated noise -38.1 dB
[Auto Level] Current noise level -37.7 dB, estimated noise -38.0 dB
[Auto Level] Current noise level -38.2 dB, estimated noise -38.2 dB
[Auto Level] Current noise level -38.2 dB, estimated noise -38.2 dB

Thanks in advance!

zuckschwerdt, 2023-12-24:
The DC spike is expected, it can be an artifact of processing but also present in the I/Q data. It should not be too prominent, like seen in the pdv output.
The pdv tool relies on meta data in the filename to scale the axis.

The signals seem much too weak, you usually want something over -12 dB and my NESDR SMArt reliably shows close to 0 dB. Your noise floor is exceptionally quiet, below -20 dB is a good value.

ProfBoc75, 2023-12-24:
Sounds like the antenna is not good, for 433 Mhz the length is around 17 cm / 6.7 inch for a classic 1/4 wave length.

You must have a male pin in the antenna SMA connector, some antennas are female without the central pin (RP SMA).

rauar, 2023-12-25:
@ProfBoc75  good hint - however I already double-checked this and antenna is SMA. Antenna is a standard stubby (has an internal coil and is therefore shorter than 1/4 lamda). I took a 2nd identical antenna from another Lilygo board and it makes no difference.

@zuckschwerdt do you mean by "it should not be too prominent" that the DC spike *actually is* too prominent in my samples or that it in general should not be prominent but do not harm that much?

I managed to get another SMArt v5 in my hands and it shows totally different noise levels (around -12db to -20db) compared to the first SMArt stick (like in my previous post around 35-38db). And the new one easily picks up each and every transmission of the outdoor temperature sensor while the other one does nothing - both SDRs being only 5cm apart and identical orientation.

[image]

I cross-checked again in this case with swapping both antennas and the noise levels do not change per SMArt stick.

My impression is that the first one is kind of deaf or has some internal attentuation for some reason which leads to a very low noise level but comes along with weak signals as well. Checked the eeprom settings for any parameters which might persisted there and could have an effect one noise level but there's only serial number stuff etc. Nothing interesting.

I have still 2 variables untested: the new stick is hooked up to a Macbook with the latest rtl_433 release while the old one is connected to a MacMini running rtl_433 master/head from github.

Will connect both to the MacMini tomorrow and run a parallel test and report here.

ProfBoc75, 2023-12-25:
So could be a faulty device. Since we are working with rf signals , any dirty and/or bad soldering could affect the sensibility. If it's under waranty try to exchange it first. If you have electronic knowledge try to open it and cross check the last stage : solders, dirty ...

rauar, 2023-12-25:
Yes - I think it's faulty and I've already prepared the return. I did one last cross-check and hooked up the good stick to the Macmini with rtl_433 master/head and it shows expected noise level and picks up signal quite nicely.

Thanks for the awesome support to both of you @zuckschwerdt @ProfBoc75. And thanks to all commiters/owner for the awesome project repo.

closed 2023-12-25 by rauar