merbanan/rtl_433 #2763: problem with HA core 2023.12.3 - rtl433 version 23.11-16-gf9dc6b88 branch master at 202312231145

ap20132022, 2023-12-24:
today I have problem with my home assistant core 2023.12.3 
rtl433 version 23.11-16-gf9dc6b88 branch master at 202312231145 
I can't start it. 
`[configuration] rtl_433 version 23.11-16-gf9dc6b88 branch master at 202312231145 inputs file rtl_tcp RTL-SDR
[configuration] MQTT: Publishing MQTT data to 192.168.0.151 port 1883
[configuration] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433-next/devices[/type][/model][/subtype][/channel][/id]".
[configuration] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433-next/events".
[configuration] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433-next/states".
[configuration] Use "-F log" if you want any messages, warnings, and errors in the console.
[configuration] Found Fitipower FC0012 tuner
[configuration] Exact sample rate is: 250000.000414 Hz
[configuration] Allocating 15 zero-copy buffers
[configuration] Failed to allocate zero-copy buffer for transfer 10
[configuration] Falling back to buffers in userspace
[configuration] Allocating 15 (non-zero-copy) user-space buffers`

I have tried to use my back up with version 0.4.0 without success. 
Everytime after changing the configuration file I restart rtl433 but I think it doesn't happen anything. I always have to uninstall it and reinstall. Then I have a new log. 
Do you have any idea?

gdt, 2023-12-24:
- Run rtl_433 from the command line, separately from HA, with debugging.
 - Reboot your computer, after reading the "can't allocate" error.

ap20132022, 2023-12-24:
> * Run rtl_433 from the command line, separately from HA, with debugging.
> * Reboot your computer, after reading the "can't allocate" error.

sorry for my dummy question. How can I do it?

ap20132022, 2023-12-26:
it solved. 
After rebooting (not restarting) HA, it works again.

gdt, 2024-05-30:
submitter reported they no longer have the problem

closed 2024-05-30 by gdt