merbanan/rtl_433 #2768: MQTT option to override default base topic

jhj1974, 2023-12-27:
Base topic currently defaults to /rtl_433/<hostname>. If this is to be changed one has to fully supply all three devices, events and states options. It would be a usefull feature to be able to override the default base topic without changing anything else. Especially if one is running multiple instances of rtl_433 on the same host and would like to identify the instance by topic. E.g. defining base topics as /rtl_433/receiver0 and /rtl_433/receiver1

Maybe something like this:
rtl_433 -f 868350000 -C si -F mqtt://192.168.100.100:1883,base=/rtl_433/recevier0

zuckschwerdt, 2023-12-27:
A sensible request, though I'd say specifying three topics is not that bad. Perhaps use a conf file, and maybe you don't even use/process all three?

To terse the args: `-f 868350000` is `-f 868.35M` and `-F mqtt://192.168.100.100:1883` is `-F mqtt://192.168.100.100`

Note that, as per MQTT recommendations, topics do not start with a slash.

rct, 2023-12-27:
> though I'd say specifying three topics is not that bad.

One thing that would make me more comfortable with this is if there was a way to get the **full current** list of default topics that the current version rtl_433 would use, so I'm not trying to hunt down whether I've got the right starting point. 

Also, someday I'm hoping there might be more than 3 topics.

zuckschwerdt, 2023-12-27:
Not sure if it's sufficient, but `-v` will log the topics:
```
$ rtl_433 -v -F mqtt
...
MQTT: Publishing MQTT data to localhost port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/HOSTNAME/devices[/type][/model][/subtype][/channel][/id]".
MQTT: Publishing events info to MQTT topic "rtl_433/HOSTNAME/events".
MQTT: Publishing states info to MQTT topic "rtl_433/HOSTNAME/states".
```

rct, 2023-12-27:
Good to know and helps a little. It's a bit obscure, but at least it prints that out before trying and failing to open either the sdr or the MQTT connection. More useful to get all the default values with one of the help options. 

So if you want to change the mqtt base to include a radio or an instance name you need to do:

```
-F mqtt://mqtthost:mqttport,user=mqttuser,pass=mqttpass,devices=rtl_433/HOSTNAME/INSTANCE/devices[/type][/model][/subtype][/channel][/id] \
-F mqtt://mqtthost:mqttport,user=mqttuser,pass=mqttpass,events=rtl_433/HOSTNAME/INSTANCE/events \
-F mqtt://mqtthost:mqttport,user=mqttuser,pass=mqttpass,states=rtl_433/HOSTNAME/INSTANCE/states \
```

Is that correct?

zuckschwerdt, 2023-12-27:
That would open multiple connections. Better use just one -F mqtt with all args appended.

jhj1974, 2023-12-27:
> A sensible request, though I'd say specifying three topics is not that bad. Perhaps use a conf file, and maybe you don't even use/process all three?
>
> To terse the args: `-f 868350000` is `-f [...]

Thank you for your reply.

I'm running my rtl_433 instances in Docker containers. Hence, supplying all the parameters through the command saves me from the "clutter" of volumes and conf files at the host system. By managing Docker from Portainer I don't even have to ssh into the host to edit the parameters. Just edit and re-deploy the container. So basically my request is based on laziness and an urge to de-clutter my rtl_433 command. The output is processed through rtl_433_mqtt_hass.py, so at least devices and events are required. But I like to keep as much default behaviour as possible, so supplying the states option feels right.

Also thanks for the tips to on how to tidy up my args, and the superfluous leading slash.

rct, 2023-12-28:
> That would open multiple connections. Better use just one -F mqtt with all args appended. 

Didn't realize that bit of flexibility existed.  So is this the correct way to change the base?

```
-F mqtt://mqtthost:mqttport,user=mqttuser,pass=mqttpass,devices=rtl_433/HOSTNAME/INSTANCE/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/HOSTNAME/INSTANCE/events,states=rtl_433/HOSTNAME/INSTANCE/states
```

This wasn't obvious to me from the documentation, but does make sense now. Using env vars for host & password from https://github.com/merbanan/rtl_433/commit/a863963a70ac6cf23514df312a27283ddb86bfde will help a little.  But, I agree at this point, it would be nice to have a way to just change the MQTT base part of the topic without having to recreate the full arg.

zuckschwerdt, 2023-12-28:
Yes, correct. Depending on the used shell you need to quote the whole arg so brackets don't get expanded.

IIRC all outputs can be used multiple times -- just be careful not to use stdout in outputs multiple times.
E.g. `-F json -F json:log.json` might be useful.

commit 86a39f2d (2023-12-28): Add mqtt base topic option (closes #2768)
closed 2023-12-28 by zuckschwerdt