merbanan/rtl_433 #2771: WS2032 error message

michel67260, 2023-12-31:
Hello, when decode with F 145 (WS2032) on Raspberry pi 4 i have this message (json ouput):

{"time" : "2023-12-31 01:33:35", "model" : "WS2032", "id" : 7965, "temperature_C" : 2.300, "humidity" : 20, "wind_dir_deg" : 0.000, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "maybe_flags" : 0, "maybe_rain" : 256, "mic" : "CRC"}
WARNING: Undeclared field "maybe_flags" in [145] "WS2032 weather station"
WARNING: Undeclared field "maybe_rain" in [145] "WS2032 weather station"
WARNING: Undeclared field "mic" in [145] "WS2032 weather station"

But same error for anothor format, not see battery message... 

You can help ? thk's

ProfBoc75, 2023-12-31:
Hi @michel67260 : please update to a more recent version of rtl_433 , sounds like you are using a very old one ...

michel67260, 2023-12-31:
Hell thk's for reply, how to update, you have procedure for Raspberry Pi 4
please ?

michel67260, 2023-12-31:
Hello after update it's done thk's for help

michel67260, 2023-12-31:
after update it's done

closed 2023-12-31 by michel67260