merbanan/rtl_433 #2774: Eva-Dry Safe sensor compatibility

bkenobi, 2024-01-03:
I have a thermohygrometer that I inherited in a box of miscellaneous stuff.  I was thinking it was probably junk (battery corrosion was my suspicion) so I opened it up and it looks very clean.  The device runs on 433.92 based on the crystal.  I looked for an FCCID but did not see one on the product and it does not appear to be on the mfg web site.  The boards have what appears to be a date stamped on them "20130725" so this is around a 10 year old design.

[image]

I have not tried to capture anything yet.  I was hoping to find out if this is an already included product.  I did not see it in the list so assume not.  I can capture data but I'm not sure how to convert it.  Without the FCC filing, I'm assuming it's probably not something that can be accomplished?  It seems to use standard off the shelf daughter boards for the RX/TX so I was hoping it might be straight forward but I can't even locate those in the FCC database.

zuckschwerdt, 2024-01-03:
The easiest way is too capture a sample and look at that. See https://triq.org/rtl_433/ANALYZE.html

bkenobi, 2024-01-03:
Looks like a very well put together doc!  I already have a lot of devices transmitting so I'm not sure how easy it will be to isolate a signal. I may have to build an image on an old RPi and take it to a clean location unless there is an easy way to make a Faraday cage around the sdr and device.

zuckschwerdt, 2024-01-03:
Isolating mostly seems to work well when unplugging any antenna an placing the sensor at maybe 30 cm to the receiver.

bkenobi, 2024-01-03:
Ok worth a try. Also a lot easier than making a new image or constructing a faraday cage  for a simple test.

bkenobi, 2024-01-03:
Some internal images for reference:

[image]
[image]
[image]

bkenobi, 2024-01-03:
Manual
[Eva-Dry Sensor1.pdf](https://github.com/merbanan/rtl_433/files/13821901/Eva-Dry.Sensor1.pdf)

gdt, 2024-06-03:
This seems like a request for a decoder with no actionable info.  I'll therefore close it in 7 days, unless there's restatement/progress or telling me I'm confused.

bkenobi, 2024-06-03:
That's correct. I collected the info I had about hardware etc but didn't get a chance to collect any data. This can be closed. If I get a chance to post data at some point I will update.

gdt, 2024-06-03:
Thanks for responding.   We can easily reopen this issue if you return with data someday.

closed 2024-06-03 by gdt