merbanan/rtl_433 #2775: Wildly inconsistent measurement rate

JHthe4, 2024-01-03:
Hi all,

I recently picked up a Bresser 7 in 1 weather station and an RTL-SDR dongle. RTL_433 is running on a Beaglebone Black in the latest version, but unfortunately I'm seeing inconsistent rates of data being written to my InfluxDB, with only tens of seconds between measurements sometimes (apparently the weather station transmits 5 times per minute), and sometimes gaps of multiple minutes or in extreme cases multiple hours:

[image]

Identical gaps occur in all measurements, so it's not just that temperature isn't updated, it's that nothing is written to the DB at all. Is this a hardware issue or some kind of power-saving mode that I can deactivate manually? Signal strength seems to be quite good and is also totally consistent over time (when measurements are available):

[image]

My config file is quite simple, and I've deferred to the default options on most things:

```
frequency     868.3M
sample_rate   1024k
report_meta level
report_meta noise
report_meta stats
report_meta time:usec
report_meta protocol
output mqtt://...
output influxs://...
convert si
protocol 173
```

I'm at quite a loss what to do here. When the signal is received and persisted, it's perfectly fine, but I don't know along what part of the chain things are failing. Has anybody experienced something like this before? 

Thanks for any and all advice!

merbanan, 2024-01-03:
Sounds like there is an overlapping signal or that the signal is to close to the center spike. Run rtl_433 from command line with all protocols active and -A and check if something else is transmitting at the same time.

JHthe4, 2024-01-03:
Thanks for the ideads! That definitely might be possible. The thing that mainly confuses me is that it works well sometimes and then collapses so drastically at others. Especially during the night, I wouldn't expect interference to be stronger than during normal day-to-day operation. The neighborhood where I live consists of single-family homes, so I can't imagine that there should be too much interference going on. However, there's definitely a lot going on on the airwaves, here's a pastebin to the result of running rtl_433 for about 30 seconds with the rtl_433.example.conf that seems to enable almost all relevant protocols: https://pastebin.com/gLKxzHgb . Unfortanately, I don't really know what to make of it, and I'm not really sure if there is even a transmission from the weather station included in it? Maybe you can tell?

merbanan, 2024-01-03:
Found Fitipower FC0012 tuner

This tuner is not that sensitive. Try get a stick with a R820t2 or set a fixed gain with the -g parameter. Maybe -g 20 is good.

Even with -A you should get real decodes.

JHthe4, 2024-01-03:
Good to know. I was aware that this wasn't the best tuner available, but I didn't expect it to be this bad. I played around with various fixed gain values from 10 to 50 and I was able to get a decode of the weather station, but the SNR always remained around the 11-13 dB mark, which I guess isn't ideal. Here's a paste with the fixed gain set to 20, as you suggested: https://pastebin.com/nXktQHXu . Before replacing the entire RTL_SDR, I was thinking to try a different/better antenna setup? Right now, I'm using the antenna that came with it, which clearly works most of the time but probably isn't ideal. Good to know that it's not a known issue though and probably not in the software. I'll update this issue when I try a different antenna and see how that changes things. Thanks for your time!

ProfBoc75, 2024-01-03:
> Found Fitipower FC0012 tuner
> 
> This tuner is not that sensitive. Try get a stick with a R820t2 or set a fixed gain with the -g parameter. Maybe -g 20 is good.
> 
> Even with -A you should get real decodes.

Hi all, happy new year ! ... 

Notice that the R820T2 is now replaced by R860, and the R820T is replaced by the R828D. Both are much better than the FC0012 and they cover higher frequency up to 1.7 / 1.8 GHz.

@JHthe4 : if you have trouble during the night, it could be also the battery during colder period in the weather station, in result, it could shift a little the frequency and move the signal outside the receipt range if it's already border line.
So try rtl_433 with -s 1024k will increase the sample rate and cover more frequencies, that could help also.

And using an USB extension lead to keep the RTL dongle away from the PC board (away from electronic noise / rf interference).
The cables and the antenna are key in the sensitivity and require attention, a good installation properly isolated will give good/better results.

JHthe4, 2024-01-04:
Very interesting @ProfBoc75, thank you for the input! A happy new year to you as well!

I added the fixed gain (20) to the config file, and it seems like I'm receiving a signal slightly more consistently now during the day. However, there's another big loss of signal for multiple hours during the night: 

[image]

I'm inclined to believe that your theory with the batteries might be correct. How would I compensate for this in rtl_433? Sample rate is currently set to 1024k, should I increase this further?

JHthe4, 2024-01-07:
Hi all,

just wanted to report back with a quick update: After taking your advice @ProfBoc75, I experimented a little with the antenna orientation, and I was able to find a horizontal configuration where SNR significantly improved and data could be received consistently during the night as well. It seems that this slight modification was enough to enable the antenna to receive the offset frequencies during cold periods as well and my issue is now solved:

[image]

Thanks so much to you and to @merbanan for helping me get to the bottom of this! For anyone else experiencing similar issues of intermittent reception breakdowns: Just trial and error test your antenna orientation and position and see if you can get better results in a different configuration.

closed 2024-01-07 by JHthe4