merbanan/rtl_433 #2779: How to ignore a bit in flex decoder?

schumar, 2024-01-04:
I have [this signal](https://triq.org/pdv/#AAB01D050101F403DC07DC10A84F1481818190819081908181908190819081A355+AAB00D050101F403DC07DC10A84F149355+AAB01D050101F403DC07DC10A84F1481818190819081908181908190819081A355+AAB00D050101F403DC07DC10A84F149355+AAB01D050101F403DC07DC10A84F1481818190819081908181908190819081A355+AAB00D050101F403DC07DC10A84F149355+AAB01D050101F403DC07DC10A84F1481818190819081908181908190819081A355+AAB00D050101F403DC07DC10A84F149355+AAB01D050101F403DC07DC10A84F1481818190819081908181908190819081A355+AAB00D050101F403DC07DC10A84F149355+AAB01D050101F403DC07DC10A84F1481818190819081908181908190819081A455)

It's 16 bit, repeated 6 times (bits are inverted). My issue is the second pulse that is sent between the repetitions, the 988-H 4264-L one. It gets interpreted as an additional `0` bit, thus decoded I get `EA D5 / 75 6A 80 / 75 6A 80 / 75 6A 80 / 75 6A 80 / 75 6A 80 /` instead of the `EA D5 / EA D5 / EA D5 / EA D5 / EA D5 / EA D5` I need.

Does the decoder have support for this?

zuckschwerdt, 2024-01-04:
Set the gap limit (e.g. 1500 as 1000 µs is the observed maximum) and you get sync on their own rows.

schumar, 2024-01-04:
Thanks for the quick reply, @zuckschwerdt ; but with that I get additional rows containing a `0` bit, e.g. `EA D5 / / 00 / EA D5 / / 00 / EA D5 / / 00 / EA D5 / / 00 / EA D5 / / 00 / EA D5 / /`

klohner, 2024-01-08:
Would you be able to attach a .cu8 of one of these samples?

klohner, 2024-01-08:
I've had luck with similar situations by going down the path of doing an initial PCM decode (instead of PWM) and then decoding the bits in each PWM set using symbol_one and symbol_zero.  Without looking at one of your .cu8 files, this is just a guess, but here's something that might get you started:

`-X "n=MyDevice,m=OOK_PCM,s=496,l=496,r=3000,bits=58,symbol_one={3}8,symbol_zero={3}c"`

Here's what I get trying to decode the raw data in your PDV link using the `-y` parameter.

`rtl_433 -X "n=MyDevice,m=OOK_PCM,s=496,l=496,r=3000,bits=58,symbol_one={3}8,symbol_zero={3}c" -y AAB10501F403DC07DC10A84F1481818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A455`

```
time      : 2024-01-08 12:38:42
model     : MyDevice     count     : 1             num_rows  : 1             rows      :
len       : 16           data      : ead5
codes     : {16}ead5
[...repeated 6 times...]
```

You may need to tweak the reset and bits values for your actual signal.

If this device can send different signals based on different conditions or button presses, look into the "get" methods to decode that data as well.  For example:

`rtl_433 -X "n=MyDevice,m=OOK_PCM,s=496,l=496,r=2000,bits>=51,symbol_one={3}8,symbol_zero={3}c,get=ID:@0:{8}:%02x,get=Something:@8:{4}:,get=SomethingElse:@12:{4}:[5:Button_1 6:Button_2]" -y AAB10501F403DC07DC10A84F1481818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A39381818190819081908181908190819081A455`

```
time      : 2024-01-08 12:58:48
model     : MyDevice     count     : 1             num_rows  : 1             rows      :
len       : 16           data      : ead5          ID        : ea            Something : 13            SomethingElse: Button_1
codes     : {16}ead5
```