merbanan/rtl_433 #2781: Build Instruction Using CMAKE, Appears to be Missing Sudo Cmd

rmschaller, 2024-01-05:
On Line 82 of the build instructions page for rtl_433 project, the command as is, creates a permissions error and was fixed using "sudo make install" the existing line 82 is "make install"  Unless this issue is specific to my system, you might want to revisit line 82.

gdt, 2024-01-05:
Strictly, you don't need elevated privs if installing to some prefix you own, and installing to system needing elevated privs is normal, but I agree a hint might be nice.

zuckschwerdt, 2024-01-05:
We don't consider rtl_433 beginner friendly but a note has now been added.

closed 2024-01-05 by zuckschwerdt

rmschaller, 2024-01-06:
Thanks!  I was able to get pass the issue I had rather quickly.  The note may be of help to others.

commit 10bd337d (2024-01-05): docs: Add sudo install note (closes #2781)