merbanan/rtl_433 #2784: Integrate Swann doorbell

tom-hooper, 2024-01-09:
I am proposing the following decoding for a swann RF 433 doorbell. 

```
decoder {
        name        = Swann-Doorbell,
        modulation  = OOK_PWM,
        short       = 200,
        long        = 580,
        tolerance   = 152,
        gap         = 564,
        reset       = 5824,
        bits       >= 24,
        rows       >= 25,
        unique,
        get         = @0:{24}:id,
}
```

This is based on #2277  and the following out put from rtl_433 -A

```
~/rtl_433/conf$ rtl_433 -A
rtl_433 version 21.05-41-gf4aa5af7 branch master at 202107111219 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Registered 161 out of 190 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-190 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected OOK package    2024-01-09 13:52:14
Analyzing pulses...
Total count: 1200,  width: 1157.30 ms           (289325 S)
Pulse width distribution:
 [ 0] count:  672,  width:  200 us [196;224]    (  50 S)
 [ 1] count:  528,  width:  580 us [576;588]    ( 145 S)
Gap width distribution:
 [ 0] count:  530,  width:  176 us [172;200]    (  44 S)
 [ 1] count:   48,  width: 5796 us [5784;5824]  (1449 S)
 [ 2] count:  621,  width:  552 us [544;560]    ( 138 S)
Pulse period distribution:
 [ 0] count:    2,  width:  408 us [404;412]    ( 102 S)
 [ 1] count:   48,  width: 6000 us [5988;6024]  (1500 S)
 [ 2] count: 1149,  width:  752 us [744;768]    ( 188 S)
Pulse timing distribution:
 [ 0] count: 1202,  width:  188 us [172;224]    (  47 S)
 [ 1] count: 1150,  width:  564 us [544;588]    ( 141 S)
 [ 2] count:   48,  width: 5796 us [5784;5824]  (1449 S)
Level estimates [high, low]:  15969,      1
RSSI: -0.1 dB SNR: 42.0 dB Noise: -42.1 dB
Frequency offsets [F1, F2]:    9650,      0     (+36.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB00B030100BC023416A480808255+AAB021032F00BC023416A49090909090909090908181818181908181818181908181818255+AAB01E030100BC023416A49090909090909090908181818181908181818181908155
Attempting demodulation... short_width: 200, long_width: 580, reset_limit: 5828, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=200,l=580,r=5828,g=564,t=152,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 25
[00] { 3} e0          : 111
[01] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[02] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[03] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[04] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[05] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[06] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[07] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[08] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[09] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[10] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[11] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[12] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[13] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[14] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[15] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[16] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[17] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[18] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[19] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[20] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[21] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[22] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[23] {25} 00 7d f7 80 : 00000000 01111101 11110111 1
[24] {22} 00 7d f7    : 00000000 01111101 111101
... Maximum number of rows reached. Message is likely truncated.

Detected OOK package    2024-01-09 13:52:15
Analyzing pulses...
Total count:  596,  width: 541.11 ms            (135278 S)
Pulse width distribution:
 [ 0] count:  365,  width:  200 us [196;208]    (  50 S)
 [ 1] count:  231,  width:  580 us [576;588]    ( 145 S)
Gap width distribution:
 [ 0] count:  271,  width:  552 us [548;560]    ( 138 S)
 [ 1] count:   22,  width: 5812 us [5804;5824]  (1453 S)
 [ 2] count:  302,  width:  180 us [172;204]    (  45 S)
Pulse period distribution:
 [ 0] count:  502,  width:  756 us [748;784]    ( 189 S)
 [ 1] count:   22,  width: 6012 us [6004;6024]  (1503 S)
 [ 2] count:   71,  width:  400 us [400;408]    ( 100 S)
Pulse timing distribution:
 [ 0] count:  667,  width:  192 us [172;208]    (  48 S)
 [ 1] count:  502,  width:  564 us [548;588]    ( 141 S)
 [ 2] count:   22,  width: 5812 us [5804;5824]  (1453 S)
 [ 3] count:    1,  width: 10004 us [10004;10004]       (2501 S)
Level estimates [high, low]:  15934,      1
RSSI: -0.1 dB SNR: 42.0 dB Noise: -42.1 dB
Frequency offsets [F1, F2]:   10199,      0     (+38.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB00D040100C0023416B4271481818255+AAB023041400C0023416B427149090909090909090908181818181908181818181908181818255+AAB01F040100C0023416B4271490909090909090909081818181819081818181808255+AAB052040100C0023416B4271490808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808355
Attempting demodulation... short_width: 200, long_width: 580, reset_limit: 5828, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=200,l=580,r=5828,g=564,t=152,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 23
[00] { 3} e0                         : 111
[01] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[02] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[03] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[04] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[05] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[06] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[07] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[08] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[09] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[10] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[11] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[12] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[13] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[14] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[15] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[16] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[17] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[18] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[19] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[20] {25} 00 7d f7 80                : 00000000 01111101 11110111 1
[21] {21} 00 7d f8                   : 00000000 01111101 11111
[22] {72} 7f ff ff ff ff ff ff ff ff
```

zuckschwerdt, 2024-01-09:
Document it like the files in https://github.com/merbanan/rtl_433/tree/master/conf and add a PR if you like.
Note that this protocol is EV1527 (there are already a few of these in conf but timings always differ.)

klohner, 2024-01-09:
@tom-hooper Would you be able to attach a zipped .cu8 of this signal?

Is this the correct product page for this doorbell or do you have a different model?

[Swann SWHOM-DC810B Wireless Door Chime](https://support.swann.com/hc/en-us/articles/4645281542041-HOM-DC810B-User-Manual)

tom-hooper, 2024-01-09:
It is this one. 

[Swann SWHOM-DC820P Wireless Doorbell Chime](https://support.swann.com/hc/en-us/articles/4645286448153-HOM-DC820P-User-Manual)

[Uploading swann_doorbell_sample.tar.gz…](Swann recording)

Theres a few signals in there but most of them are recognised

gdt, 2024-06-03:
@tom-hooper We are waiting a PR so we can close this issue.

klohner, 2024-06-04:
FWIW:

- Only the g002, g004, and g005 samples in [swann_doorbell_sample.tar.gz](https://github.com/merbanan/rtl_433/files/13880716/swann_doorbell_sample.tar.gz) seem to contain the doorbell signal (g001 is identified as "Cotech-367959" and g003 is "Ambientweather-F007TH").
- Only g002 seems to be a complete capture of the doorbell signal (g004 is truncated and g005 is missing the the beginning).
- The complete signal seems to repeat the full data row 65 times which seems so long that rtl_433 wants to chop the signal into two signals.  Thus, the decoder needs something like "repeats >= 18" (at least) to prevent double-reporting the signal.

With no CRC or Checksum, it may be worth being more discriminating in the decoder, such as:

```
decoder {
    name        = Swann-Doorbell,
    modulation  = OOK_PWM,
    short       = 190,
    long        = 570,
    gap         = 660,
    reset       = 6000,
    bits        = 25,
    repeats    >= 25,
    unique,
    get         = @0:{24}:id,
}

```

Or perhaps even:

```
decoder {
    name        = Swann-Doorbell_v2,
    modulation  = OOK_PCM,
    short       = 190,
    long        = 190,
    gap         = 2538,
    reset       = 6000,
    repeats    >= 18,
    bits        = 110,
    unique      ,
    symbol_one  = {4}8,
    symbol_zero = {4}e,
    get         = @0:{24}:id,
}

```

[image]

gdt, 2024-06-05:
Is it then time to submit a PR to add a flex decoder?   if not, what's the plan to make progress towards being able to close this?

klohner, 2024-06-05:
I defer the PR to @tom-hooper if wants to continue testing and add it.

gdt, 2025-07-04:
@klohner @tom-hooper We are still waiting a PR.  I don't want to have issues stay open forever without progress as we are over 100 of them.