merbanan/rtl_433 #2785: rtl_433 suddenly stopped working for all but 1 sensor

bkenobi, 2024-01-09:
Background:
I have quite a few sensors being received by rtl_433 which passes the readings on to Node Red via MQTT for processing.  The system has been in place in its current configuration for over a year without many issues at all.  When I lose power, it's not uncommon for me to have to restart my HA software and sometimes rtl_433 will also fail to work correctly.  The RPi I have these on is behind a UPS but things don't always run smoothly for one reason or another.

Issue:
My rtl_433 is run as a service so I don't see the interface unless I need to check on something.  I noticed that I was only getting data from 1 of my Acurite sensors as of last Wednesday (4Jan2024).  This is certainly an issue with the receiving side because they all stopped at exactly the same time.

[image]

As you can see, only 1 device was receiving over the last week (one is an ESP based device, so only 1 rtl_433 device).  I reset everything this morning and I am now getting intermittent response from a couple other devices.  The device that is working consistently is within 5' of the antenna but the others that are intermittently are not exactly close.  One is in my attic (the SDR is in the basement, so 3 floors/ceilings away).  If the antenna suddenly died I would think everything except very close devices would disappear.  I have closer devices that are not being picked up so this is a bit strange.  I can try moving the SDR to a different location to see if it changes anything but I'm really not sure what would cause such a sudden drop off.

zuckschwerdt, 2024-01-09:
If you didn't change hardware or software then it might be interference. Can you check the band with some SDR UI? Or as poor man's solution grab a short time (`rtl_433 YOUR-USUAL-ARGS -w filename_433.92M_250k.cu8 -T 30` -- records 30 seconds, change the filename to match actuall freq and rate) and drop on https://triq.org/pdv/

bkenobi, 2024-01-09:
I took a recording earlier and just got a chance to look at the data.  I don't see anything odd when analyzing it.  I just took another recording and noticed that many of my sensors were being displayed this time.  I restarted the service as normal and the sensors seem to be being detected (whitnessed through mosquitto_sub initially, confirmed through grafana).
[image]

As far as the data sets go, I don't actually see any difference between what was collected so it seems like it must be some strange issue with rtl_433 not detecting them for some reason (maybe the SDR was unhappy and needed time to work something out)?  I'm not sure since it had no issue with one and intermittently picked up 2 others.  But the other devices were not there.

[20240109_1001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/13878313/20240109_1001_433.92M_250k.cu8.zip)
[20240109_1143_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/13878314/20240109_1143_433.92M_250k.cu8.zip)

[image]

This can be marked resolved if you prefer.  I don't know what caused this and so I don't know how to recreate to help debug.  I will post again if I see it repeat.

zuckschwerdt, 2024-01-09:
Reopen if you again encounter this oddity.

closed 2024-01-09 by zuckschwerdt