merbanan/rtl_433 #2787: Issue: Vauno EN8822C - Temp and Hum values - Wide variations

deckwurst, 2024-01-12:
Hello!

I have a couple of [Vauno EN8822C](https://github.com/merbanan/rtl_433/pull/2231) and I'm trying to use them on Home Assistant.
I'm using [this receiver](https://www.amazon.ca/gp/product/B009U7WZCA/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1).

I'm able to receive a signal on Home Assistant, but the sensor (I'm testing two of them, side by side) looks very unstable **AND** completely incoherent:

[image]

Here are some debug info:
https://ibb.co/HpcTxBp

I have two hypothesis:

1. This is more likely to come from the sensors themselves, but I hope not...
2. The config file of rtl_433 and/or config files of the sensor are not good.
3. Running rtl_433 inside Home Assistant is not the way to do it...

If you look at the debug info file, it's mentionned that the sensor model is "Conrad-S3318P", but it should be "vauno_en8822c".
And here is the log when I start the addon rtl_433:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] Publishing MQTT data to core-mosquitto port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] Registered 176 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-58 63 67-71 73-81 83-87 91-94 96-100 102-105 108-109 111-116 119 121 124-128 130-139 141-149 151-155 157-161 163-167 170-175 177-179 181-185 187-197 199 202 204-215 217-223 ]
[rtl_433] Found Rafael Micro R820T/2 tuner
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] Sample rate set to 250000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 433.920MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection established.
```

So I clearly don't have the device 230 (Vauno EN8822C) active.
But I'm absolutely unable to find where is the default config file...

But, when I tried to change the Home Assistant rtl_433 addon default config file for [this one](https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf) (I have an option to setup a custom config file path in the addon options), the addon refuse to start with this error:
```
    [220]  Maverick XR-30 BBQ Sensor
    [221]  Fine Offset Electronics WN34 temperature sensor
    [222]  Rubicson Pool Thermometer 48942
    [223]  Badger ORION water meter, 100kbps (-f 916450000 -s 1200000)

* Disabled by default, use -R n or a conf file to enable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

I don't know if my issue is more related to rtl_433, or to Home Assistant...

@loopy321, any ideas? Is it still working for you?

Thanks in advance!!

zuckschwerdt, 2024-01-12:
You need a newer version of rtl_433 to get protocol 230.

Both protocol, [Vauno-EN8822C](https://github.com/merbanan/rtl_433/blob/master/src/devices/vauno_en8822c.c) and [Conrad-S3318P](https://github.com/merbanan/rtl_433/blob/master/src/devices/s3318p.c) may look the similar to the decoders but values are coded differently, thus the "broken" readings.

deckwurst, 2024-01-12:
Ok, so I was using this addon:https://github.com/pbkhrv/rtl_433-hass-addons
And now I switched to this one: https://github.com/catduckgnaf/rtl_433_haos_addon

Now I can see a good config file with the latest update.

But, still not working well...
My MQTT broker is working.
I created an user in Home Assistant, and I'm able to use it to connect to MQTT Explorer.

So I used the same credentials in rtl_433.conf, but when I start the addon, I get this log for the first few seconds:
```
Registering protocol [241] "EezTire E618, Carchet TPMS"
Registering protocol [246] "TFA 30.3151 Weather Station"
Registering protocol [247] "Bresser water leakage"
Registering protocol [212] "Renault 0435R TPMS"
Registering protocol [208] "AVE TPMS"
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
```

And then:
```
[MQTT] MQTT Connected...
[MQTT] MQTT Connection error: 5
[MQTT] MQTT Connection lost, reconnecting...
[MQTT] MQTT Connected...
[MQTT] MQTT Connection error: 5
[MQTT] MQTT Connection lost, reconnecting...
[MQTT] MQTT Connected...
[MQTT] MQTT Connection error: 5
[MQTT] MQTT Connection lost, reconnecting...
[MQTT] MQTT Connected...
```

And here is my MQTT log:
```
2024-01-12 18:11:20: New connection from 172.30.33.2:47768 on port 1883.
2024-01-12 18:11:20: Client rtl_433-5744ffff9253 disconnected, not authorised.
2024-01-12 18:11:20: New connection from 172.30.33.2:47774 on port 1883.
2024-01-12 18:11:20: Client rtl_433-5744ffff9253 disconnected, not authorised.
2024-01-12 18:11:20: New connection from 172.30.33.2:47776 on port 1883.
2024-01-12 18:11:20: Client rtl_433-5744ffff9253 disconnected, not authorised.
```

I don't understand what I am doing wrong here...

Maybe I shouldn't use rtl_433 as an addon but run it in a LXC container on Proxmox?

Thanks!

deckwurst, 2024-02-12:
Problem solved...
The " are not recommended in the config file... lol
That's where my mistake came from...

closed 2024-02-12 by deckwurst