merbanan/rtl_433 #2788: No output running device directly, get output running device through rtl_tcp

Jsherbinow, 2024-01-12:
After updating my laptop to debian 12, i no longer get any output from my device, when use the device through rtl_tcp, i do get output.

`jeremy@jeremy-katana:~$ rtl_433 -d 0 -t biastee=1
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/jeremy/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Enabled direct sampling mode, input 2
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
[R82XX] PLL not locked!
Disabled direct sampling mode
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us`

And using rtl_tcp:

`jeremy@jeremy-katana:~$ rtl_tcp -d 0 -T -a 127.0.0.1 -p 1234 -f 433920000
Found 3 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
  1:  Realtek, RTL2838UHIDIR, SN: 00000002
  2:  Realtek, RTL2838UHIDIR, SN: 00000003

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Enabled direct sampling mode, input 2
[R82XX] PLL not locked!
Disabled direct sampling mode
Tuned to 433920000 Hz.
activated bias-T on GPIO PIN 0
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
`
`jeremy@jeremy-katana:~$ rtl_433 -d rtl_tcp:127.0.0.1:1234
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/jeremy/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
rtl_tcp input from 127.0.0.1 port 1234
rtl_tcp connected to 127.0.0.1:1234 (Tuner: R820T)
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:17:45
model     : Acurite-606TX id        : 20
Battery   : 0            Temperature: -17.2 C      Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:17:51
model     : Acurite-5n1  message_type: 56          id        : 2629
channel   : A            sequence_num: 0           Battery   : 1             wind_speed: 11.8 km/h     temperature: 0.5 F
humidity  : 65 %         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:17:51
model     : Acurite-5n1  message_type: 56          id        : 2629
channel   : A            sequence_num: 1           Battery   : 1             wind_speed: 11.8 km/h     temperature: 0.5 F
humidity  : 65 %         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:17:51
model     : Acurite-5n1  message_type: 56          id        : 2629
channel   : A            sequence_num: 2           Battery   : 1             wind_speed: 11.8 km/h     temperature: 0.5 F
humidity  : 65 %         Integrity : CHECKSUM
^CSignal caught, exiting!
WARNING: sync read failed. -1
rtl_tcp: Interrupted system call
`

I am sure it is something i am doing wrong, but cant seem to figure it out? Anybody know what i am doing wrong?

zuckschwerdt, 2024-01-12:
Strange. Try also with `-Y autolevel -M level -M noise` to report signal strength -- there should at least be that in the output then.

Jsherbinow, 2024-01-12:
jeremy@jeremy-katana:~$ rtl_433 -d 0 -t biastee=1 -Y autolevel -M level -M noise
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/jeremy/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Enabled direct sampling mode, input 2
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
[R82XX] PLL not locked!
Disabled direct sampling mode
Tuned to 433.920MHz.
Estimated noise level is -18.5 dB, adjusting minimum detection level to -15.5 dB
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Estimated noise level is -21.5 dB, adjusting minimum detection level to -18.5 dB
Estimated noise level is -24.1 dB, adjusting minimum detection level to -21.1 dB
Estimated noise level is -26.3 dB, adjusting minimum detection level to -23.3 dB
Estimated noise level is -28.3 dB, adjusting minimum detection level to -25.3 dB
Estimated noise level is -30.0 dB, adjusting minimum detection level to -27.0 dB
Estimated noise level is -31.5 dB, adjusting minimum detection level to -28.5 dB
Estimated noise level is -32.9 dB, adjusting minimum detection level to -29.9 dB
Current noise level -42.1 dB, estimated noise -32.9 dB
Estimated noise level is -34.0 dB, adjusting minimum detection level to -31.0 dB
Estimated noise level is -35.0 dB, adjusting minimum detection level to -32.0 dB
Estimated noise level is -36.7 dB, adjusting minimum detection level to -33.7 dB
Estimated noise level is -38.0 dB, adjusting minimum detection level to -35.0 dB
Estimated noise level is -39.4 dB, adjusting minimum detection level to -36.4 dB
Estimated noise level is -40.5 dB, adjusting minimum detection level to -37.5 dB
Current noise level -42.1 dB, estimated noise -41.4 dB
Estimated noise level is -41.6 dB, adjusting minimum detection level to -38.6 dB
Current noise level -42.1 dB, estimated noise -42.1 dB
Current noise level -42.1 dB, estimated noise -42.1 dB
Current noise level -42.1 dB, estimated noise -42.1 dB
^CSignal caught, exiting!

Jsherbinow, 2024-01-12:
And using rtl_tcp input

jeremy@jeremy-katana:~$ rtl_433 -d rtl_tcp:127.0.0.1:1234 -Y autolevel -M level -M noise
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/jeremy/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
rtl_tcp input from 127.0.0.1 port 1234
rtl_tcp connected to 127.0.0.1:1234 (Tuner: R820T)
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Estimated noise level is -17.2 dB, adjusting minimum detection level to -14.2 dB
Estimated noise level is -18.4 dB, adjusting minimum detection level to -15.4 dB
Estimated noise level is -20.4 dB, adjusting minimum detection level to -17.4 dB
Current noise level -27.6 dB, estimated noise -20.4 dB
Estimated noise level is -22.1 dB, adjusting minimum detection level to -19.1 dB
Estimated noise level is -23.9 dB, adjusting minimum detection level to -20.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:47:12
model     : Acurite-606TX id        : 20
Battery   : 0            Temperature: -16.6 C      Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -7.0 dB      SNR       : 24.8 dB       Noise     : -31.7 dB
Estimated noise level is -25.2 dB, adjusting minimum detection level to -22.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:47:15
model     : Acurite-5n1  message_type: 49          id        : 2629
channel   : A            sequence_num: 0           Battery   : 1             wind_speed: 8.5 km/h      wind_dir_deg: 270.0
Rainfall Accumulation: 8.05 in                     Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 30.6 dB       Noise     : -30.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:47:15
model     : Acurite-5n1  message_type: 49          id        : 2629
channel   : A            sequence_num: 1           Battery   : 1             wind_speed: 8.5 km/h      wind_dir_deg: 270.0
Rainfall Accumulation: 8.05 in                     Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 30.6 dB       Noise     : -30.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:47:15
model     : Acurite-5n1  message_type: 49          id        : 2629
channel   : A            sequence_num: 2           Battery   : 1             wind_speed: 8.5 km/h      wind_dir_deg: 270.0
Rainfall Accumulation: 8.05 in                     Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 30.6 dB       Noise     : -30.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-01-12 12:47:16
model     : Acurite-606TX id        : 233
Battery   : 1            Temperature: -18.1 C      Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -18.3 dB     SNR       : 13.4 dB       Noise     : -31.7 dB
Estimated noise level is -26.6 dB, adjusting minimum detection level to -23.6 dB
Estimated noise level is -27.9 dB, adjusting minimum detection level to -24.9 dB
Current noise level -30.3 dB, estimated noise -27.9 dB
^CSignal caught, exiting!
WARNING: sync read failed. -1
rtl_tcp: Interrupted system call

zuckschwerdt, 2024-01-12:
>  Current noise level -42.1 dB, estimated noise -42.1 dB

That is dead silence for 8-bit receivers (RTL-SDR). There is radio data incoming but it's all 0's.
You can record a file with `rtl_433 -d 0 -t biastee=1 -w file_433.92M_250k.cu8 -T 5` then drop on https://triq.org/pdv/ -- it should be "blank" (e.g. not a spectrogram/waterfall like you expect).

No idea if the incoming radio frames are 0's or the data is corrupted somehow. Perhaps build the current version, the SDR input was rewritten. (No need to install if you don't like to, just run `./src/rtl_433` from the `build` dir.)

Jsherbinow, 2024-01-29:
Building the latest rtl_433 did not have any different results, i did find a solution though. I had originally built the the rtl-sdr driver from rtl-sdr-blog, After uninstalling that driver, and installing the driver from the debian repository, everything worked as expected. Thank you for your time and effort. :)

zuckschwerdt, 2024-01-29:
This could be a general problem then? Do you know which version, date, or git tag the problematic rtl-sdr-blog driver was?

Jsherbinow, 2024-01-29:
[https://github.com/rtlsdrblog/rtl-sdr-blog](url)

Jsherbinow, 2024-01-29:
I installed the lates version about 2 months ago

gdt, 2024-06-03:
calling this concluded.  Please comment with up-to-date status if you want it reopened, or start fresh.

closed 2024-06-03 by gdt