merbanan/rtl_433 #2793: Acurite TX592 battery threshold

bkenobi, 2024-01-17:
I have a number of Acurite TX592 temperature/humidity sensors deployed.  I have recently improved my battery notification algorithm in Node Red so it actually works as desired.  After having 2 sensors alert me to the battery_ok transitioning from 1 to 0 (dead battery) I checked the voltage on both sets.  There is no issue here, just a suggestion to add to the documentation for the Acurite sensors that the threshold voltage for battery_ok notification is 2.5v.  I did not see it mentioned so if appropriate, perhaps adding it would help someone.

zuckschwerdt, 2024-01-17:
Thanks! Do you want to PR a note to https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L993 or should I just sneak it in?

gdt, 2024-01-17:
I guess we need a locking protcocol!    I got the mail and just pushed a commit.

I don't think we can conclude about the rest.    They seem to vary.    6045 in particular shows low on NiMH always.  So I noted this only for this model.

closed 2024-01-17 by gdt

zuckschwerdt, 2024-01-17:
@gdt needs push to upstream ;) And the model name "Acurite-Tower" would be prefered to the file name "acurite.c" in the commit message, s.a. https://github.com/merbanan/rtl_433/blob/master/docs/CONTRIBUTING.md#commit-messages
I.e. this would look good: "minor: Add battery note to Acurite-Tower (closes #2793)" and optionally in the commit body
`Co-authored-by: bkenobi <bkenobi@users.noreply.github.com>` for attribution.

bkenobi, 2024-01-17:
Next time the 5in1 needs batteries I will provide the voltage so it can be included too.  Thanks!

bkenobi, 2024-02-21:
My Acurite 5-in-1 (same protocol) stopped transmitting the other day.  I investigated and the battery_ok dropping from 1 to 0 wasn't picked by my node red code so I don't know the threshold for the transition.  I measured the batteries which are now at the no-longer-transmit threshold and they were all at 1.20v.  I will try to capture the voltage for the transition of battery_ok next time since I should get a notification.

rct, 2024-02-21:
Initial "indoor" testing that I had done, most Acurite devices started signalling low battery when the _average voltage per cell_ was between 1.2-1.4Vdc.  They were no longer able to transmit below 1.2Vdc. For the 5n1, when I changed batteries because it appeared to be dead the per cell average was usually < 1.1Vdc or around 1.05Vdc. 

However, aside from questions of calibration/accuracy, remember that battery voltage will vary with temperature (and different battery chemistries have different properties).  I can also confirm that NiMh enerloop type batteries usually do fall into low threshold, so the battery level flag is useless for them.

At low temperatures and low voltages you might see drop outs overnight before it is completely dead during the day.  This is also where you will tend to get invalid readings like -40 temperature and 0-1% humidity.  Also Note, the filtering code applied a year or two back should filter out those messages and make it look like the unit is dead when it is still actually transmitting.   (The raw values in those messages tend to be all zeros or all ones but the checksum would still pass.) 

Most of the time the Acurite devices set the low battery flag for weeks before the batteries are actually to low to function.  I've also seen multiple transitions in the weeks before the batteries die.

Bottom line, it's nice to collect what we know about all of these devices from testing and actual use. However, it is often not very actionable other than hints/advice.

bkenobi, 2024-02-21:
I agree with everything you said and have seen this in my devices as well.  I was posting in case it was helpful to document since more knowledge may help someone down the line.  While it doesn't necessarily help me to know what the voltage threshold is, I think it's appropriate to add it to documentation wherever applicable (which may not be here).

FWIW, I looked back at my data and it shows that the sensors started indicating battery low months ago but my Node Red code didn't properly inform me.  The data finally stopped recently but there were a few gaps in the data (the week long gap was not related, the smaller gaps to the right are the drops at night I believe).  If I zoom in on the period where it's showing 0.5, the 5in1 is just oscillating non-stop between 0 and 1.

[image]
[image]

commit 3051265a (2024-01-17): minor: acurite.c: Note that 592TXR low battery means < 2.5V

From bkenobi.  Fixes #2793.