merbanan/rtl_433 #2795: unknown temp/humidity sensor from Rossmann - unable to find proper protocol/frequency?

tomeq82, 2024-01-19:
Hi, sorry for a "long shot post" but... I knew I have read somewhere that this particular sensor has been somewhat reverse engineered, but it was way before my interest in rtl_433. Now, i'm successfully using the tool but only for well known sensors like Bresser etc. 

I have one at home that i'm basically unable to "cacth". Eg. I cant even catch it using SDR++ tuned to 433.92Mhz. I don't see an exact pattern when it sends the data, which is odd. As usually it is very clear eg. sending Nice remote signal or any other weather sensors. 

This one seems very popular, bought at Rossmann Poland.  I have pictures of this stuff. Maybe someone can help me debug/figure out. I'm open to suggestions and quite savvy to do so. Here's pictures. I deliberately omitted any external cases but if it is needed then I will upload. External unit has LCD displaying temp/humidity/channel and it uses "1,2,3" channel switch near the battery bay. 

[image]
[image]

zuckschwerdt, 2024-01-19:
Let's call it the `XS1043` (REV02).
The antenna is connected to that 6 pin SOIC. What are the markings on that?

Chances indeed are that a cheap device use 433.92 MHz. Can you grab a sample or maybe a short recording if that doesn't work?

tomeq82, 2024-01-20:
Here are the SOIC markings - I will try to cactch the sample reading, but as I said, despite it looks like standard freq I was unable to corelate moment with observation in SDR++ when it seem to send something (eg. when you push "channel" button for longer on the base station it will start to re-establish the link with the external unit). But I will try again.

[image]

zuckschwerdt, 2024-01-20:
I had no luck finding anything on "4455JK20". You can maybe remove the SDR antenna, place the sender nearby and record 2 minutes at 1024k (`rtl_433 -s 1024k -w file_433.92M_1024k.cu8 -T 120`) then scan through the recording.

Pushing a channel button on the receiver usually only allows a new sender ID for that channel to be locked in.

tomeq82, 2024-01-20:
Hmm, thanks for the good tip :) removing antenna and placing this external unit close to SDR, revealed that this unit indeed sends something decodable:
`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-20 19:22:49
model     : Nexus-TH     House Code: 74
Channel   : 3            Battery   : 1             Temperature: 25.70 C      Humidity  : 43 %
`
so I wonder, why I was unable to find it before and station communicated well with the unit.... hm. This is it! :) 

Anyway, I have this 120 seconds capture if you want. 122 megs compressed with xz. Want me to upload somewhere?

zuckschwerdt, 2024-01-20:
You can run with `rtl_433 -vv -R 19` to get some codes (e.g. `{36}b510bef47` 12 times), the full recording is not needed. But if the values shown are what you expect then all is well.

Improve reception, check frequency, levels, and noise with `-Y autolevel -M level -M noise`

We'll add `Unmarked sensor form Rossmann Poland, board markings XS1043 REV02.` to the decoder.

tomeq82, 2024-01-20:
Thanks, strange thing is that with fresh batteries, but with antenna attached and sensor 20 cm from the antenna - I get basically nothing at all (house code 74) while I receive some other aroud..... without antenna, back to back - I received quite what I wanted to get. Will investigate it further, but thanks for all the help here :)

zuckschwerdt, 2024-01-20:
You might be overloading the antenna?

tomeq82, 2024-01-20:
moving it further makes no difference. Strange....

tomeq82, 2024-01-21:
Still, cant figure this thing out - it communicates with the "base sttion" easily when in other room (more that 10 meters apart, walls etc.), but being apart from SDR with antenna like 2-3 meters I receive zero. While testing with SDR without antenna back to back as you suggested - bingo, I see it speaks, changing channels etc. works, data is ok. But not when in usual operation.

zuckschwerdt, 2024-01-21:
It might be that the frequency is not really in the selected band around 433.92M and without antenna you receive some harmonic? I'd use a UI app and try to pin down the exact frequency and then experiment with the antenna.

tomeq82, 2024-01-21:
yeah, this is it... after messing again and tuning into proper freq I got it. Thanks!

`time      : 2024-01-21 15:46:27                    Protocol  : 19
model     : Nexus-TH     House Code: 112
Channel   : 2            Battery   : 1             Temperature: 2.80 C       Humidity  : 70 %
Modulation: ASK          Freq      : 434.3 MHz
RSSI      : -0.1 dB      SNR       : 27.1 dB       Noise     : -27.2 dB`

closed 2024-01-21 by tomeq82