merbanan/rtl_433 #2798: Elsner Solexa 230V Sunblind

rauar, 2024-01-20:
Manufacturer: Elsner Elektronik

[Product Website](https://www.elsner-elektronik.de/de/solexa-230-v.html
)

This device shows the following measures:

- temperature outdoor (degree celcius)
- temperature indoor (degree celcius)
- light intensity (lux)
- wind speed (m/s)

Furthermore there are controls for:

- open sunblind
- stop sunblind
- close sunblind

So far so good. My wind sensor always shows 70m/s which causes the sunblind to never open. The windsensor is a separate device (not shown on the product website). Hardware replacement seems complicated and it's physically difficult to get hands on the wind sensor.

The plan: reverse engineer the handset. If successful then i can control everything from HomeAssistant and potentially use a different 3rd party wind sensor (or an existing one in my neighbourhood).

I collected quite a few samples for open/stop/close and sensor measures.

Using rtl_433 -R 0  -X 'n=elsner,m=FSK_MC_ZEROBIT,short=11,long=21,gap=25,reset=300' -f 868200000 I can see decoded bytes. But as I'm new with modulations and decoders I'm not 100% if this is an appropriate setting.

[image]

rauar, 2024-01-20:
Some sample cu8's.
[elsner.zip](https://github.com/merbanan/rtl_433/files/13998843/elsner.zip)

zuckschwerdt, 2024-01-20:
You should use `-X 'n=elsner,m=FSK_MC_ZEROBIT,short=11,long=11,reset=25'` and `-f 868.2M`
Short is long for MC (NRZ), s.a. https://triq.org/rtl_433/PULSE_FORMATS.html
There are no packets separated by a gap, reset needs to be the first invalid gap, i.e. somewhat longer than 22 µs.

rauar, 2024-01-28:
Thanks for the hints. I did look into the payloads for a couple of hours and tried to find the temperature value but the whole payload seems to be at least for the most part (except a static preamble or network id) obfuscated or even encrypted.

Not sure if I will put more effort into it with questionable outcome. 

One question which I could not figure out yet is: when rtl_433 detects and decodes a page it shows "code: {<some number>}<payload>. I guess <some number> is the payload length but it does not match up with the payload string length it outputs. What exactly is that number?

For future reference - the handset seems to send payload with "lnegth" 314. The outdoor station sends messages with 315 "length".

Some messages:
```
02b3ab64eeae603a45dbf7f5aad1f3a70468e070988330904f602cecaca32fb576b93e489d79c6c
02b3ab64eea7105c8a240815eb123468876eed4848226f0fc9d55741d737d9489d46993e48ee150
02b3ab64ee9cbaf2b63c3835eb1274e9ca34775ef53c6337d9f55741d7789acbe39332306ccbca0
02b3ab64eeb933a49a040815eb12342806ad69027c0a7f31c9d55741d73819891e489d0658c5b1c
02b3ab64ee88d2a459c3c7d5aad1b366c3e81fed11b5149edf402cecace3b0f7bb0250ad6681028
02b3ab64ee9efc3c09e3c7d5aad1b3260264d860f944f4675e7fecac6c222db16ea91e081c0bbd8
02b3ab64eeb6aa9a461c3835eb123428472eab8705dda5d8640a821701cd441e081c44953651a88
02b3ab64eead17747a040815eb123468c8306f0e543a5f2c89955741d737d9081c44953678f81ca
02b3ab64ee9a312862142815eb1274e9c9f3f65cb0f391da546a8217018cc31c44953678fdb4d10
02b3ab64eea80ca052342815eb123468c7f02e8d52365736e9955741d737d9085cc5973a40c56fc
02b3ab64ee87cb2011f3e7d5aad1b3668366dc2807a12c90ff402cecace3b0b77a814ea95eb47dc
02b3ab64ee8f9a3e4e2c1835eb1274a90870f01017c1adddc48a8217018d039d46997e891e4cd74
02b3ab64ee9f38fc4a240815eb1274e9ca34b7dfb6ffa9d2b4aa8217018cc31c04143474f5952c0
02b3ab64eebe36b842142815eb123428066ce840b8c3b1d614aa821701cd445e895ec99f0a5c0f4
02b3ab64eeb8ebe419c3c7d5aad1f3e7c5ec26bd31b51489df002cecaca2ef3474f5b77a816b9ac
02b3ab64eeb19cc65e0c1835eb12342846ee2ac5c3185b1c79b55741d73819c99f0a200c64c9aec
02b3ab64ee86865a461c3835eb1274a94932b39725dda5f4a48a8217018d03ddc7db8352f19cc88
02b3ab64ee8d53f63dcbd7f5aad1b366c4285f2e547ae06e0f1fecac6c21ed302c649536382535c
02b3ab64ee94201072342815eb1274e9897334996a267708c9955741d7789a8b62d1af2a60c23ba
02b3ab64eea807e052342815eb123468c7f02e8d52365709e9955761d737d9085cc5973a40d12ec
02b3ab64eea0780215fbf7f5aad1f3a744ea22b56256d85d2f1fecac6c626e3270eda71a000cf84
02b3ab64eeae527a05dbf7f5ead1f3a70468e070988330b0cfe02cecaca32fb576b93e489d50afc
02b3ab64eeb35b8e2debd7f5aad1f3e785aba73baeaf08b6ef602cecaca2ef74f5f7fb83528a9f4
02b3ab64eea6819a461c3835eb123468876eed88c92473115a355741d737db489d46d9ff8b2dd14
02b3ab64ee959e567e0c1835eb1274e989733619eb287b353a355741d7789a8b62d1efeba33b00a
02b3ab64eeab092c6a240815eb123468c7efef4c50324f1d0a155741d737d9085c85d6f9bf10934
02b3ab64eea3368c6a240815eb12346887afaf0b8eef89f434aa821701cd849f0a200c64d58da8c
02b3ab64ee99e3a821d3e7d5aad1b32602a55a623b893c98bf402cecace37076f9bf4aa14e93f54
02b3ab64eeb3968e6e2c1835eb12342846edeb443fd18dfb048a821701cd441e489d46993e55db6
02b3ab64eea77e1e0debd7f5aad1f3a7452aa537a69f28b0af602cecaca32ff5f7fbc393729757c
02b3ab64ee9d69b63dcbd7f5aad1b32602651a21ba87388c6f602cecace37076b93e081c8559198
02b3ab64eea8bde2563c3835eb123468c7f02fcd11f595cf248a821701cd84df8b2250ad2647c70
02b3ab64ee910e845a140815eb1274e989b3765a6c2a7f21c9d55741d7789a8b2250ad2658f0c78
02b3ab64ee83310c6a240815eb1274a948f1f35460124f00c9d55741d7785a0a608566d9bf3c998
02b3ab64ee8636d842142815eb1274a94932b4d7661e672ba9d55741d7785a0a200c64d5b72d9b8
02b3ab64ee8c437235fbf7f5aad1b366c42860ae9cbb20a08f602cecace3b0f7fb831270ad630d0
02b3ab64eeb18d065e0c1835eb12342846ee2c15c3185b203a355741d73819c99f0a200c64ca230
02b3ab64ee8d03747a040815eb1274a9087132119ac7b9d634aa8217018d039d46d9ff8b2256bc8
02b3ab64ee9a5dea25dbf7f5aad1b32602a59ae37e4eca78af5fecac6c222db12e285cc5d7c31a8
02b3ab64eea97b661dcbd7f5aad1f3a704a962725c4ac06a8f5fecac6c626e72f1af6ae1cfc8354
02b3ab64ee8a3be862142815eb1274a908b1b2d35e0e47382a155741d7785a4ae1cfab2250f1a94
02b3ab64eea8b862563c3835eb123468c7f02fcd11f595d564ca821701cd84df8b2250ad2642c20
02b3ab64ee9149c61dcbd7f5aad1b32642e61c258297189aefa02cecace3703678fdc79b42b7eac
02b3ab64eeae42ba05dbf7f5aad1f3a70468e1b09083308acfa02cecaca32fb576b93e489d685ec
02b3ab64eea4ee9439c3c7d5aad1f3a7452a64b6e55ce4729f7fecac6c626e32306ce5d7fbf504c
02b3ab64ee8d3df47a040815eb1274a9087132119ac7b9e3f4ea8217018d039d46d9ff8b227f952
02b3ab64ee93ca9031f3e7d5aad1b32642e5db643f910c8d7f802cecace3703678bd46997eac69c
02b3ab64ee8f413e0debd7f5aad1b366c428a12f96bf28a12fa02cecace3b0f7fbc3d3b3728b1c8
02b3ab64ee9edffc09e3c7d5aad1b3260264d9a0f944f4479f7fecac6c222db16ea91e081c13128
02b3ab64ee8ade2469e3c7d5be91b366c3d122448952a54ad5ab56ed9af5ab166c9931e34691b90
```

and here some info on change frequency and number of different hexcodes used (calculated with 2 byte width):
```
Change Frequencies:
Pos 0+2:  0.00%
Pos 1+2:  0.00%
Pos 2+2:  0.00%
Pos 3+2:  0.00%
Pos 4+2:  0.00%
[... 128 lines cut ...]
Pos 54+2: ## 5
Pos 55+2: # 4
Pos 56+2: # 4
Pos 57+2: ### 8
Pos 58+2: ##### 12
Pos 59+2: ###### 15
Pos 60+2: ####### 16
Pos 61+2: ############ 28
Pos 62+2: ############ 27
Pos 63+2: ################ 36
Pos 64+2: ################ 36
Pos 65+2: ################# 40
Pos 66+2: ################ 38
Pos 67+2: ################ 38
Pos 68+2: ################# 39
Pos 69+2: ################# 39
Pos 70+2: ################# 39
Pos 71+2: ################# 39
Pos 72+2: ################# 39
Pos 73+2: ################### 43
Pos 74+2: #################### 46
Pos 75+2: #################### 47
Pos 76+2: ################### 44
Pos 77+2: ################# 39
```

zuckschwerdt, 2024-01-28:
> it shows "code: {_some number_}. I guess _some number_ is the payload length

Yes, the code length in bits. The hex string is nibbles and thus rounded up to 4.

The codes have little structure and are too long for such little data to be easy to figure out, I agree :/

rauar, 2024-05-20:
> hello @rauar <https://github.com/rauar>, do you have progress on this? I
> also have a Solexa. I am searching the web if somebody was successful in
> connecting it to other systems. Thanks

Hi, no - unfortunately not. I did not have spare time to investigate
further and I doubt it's possible without a protocol spec from the vendor
as it seems it's using some kind of encryption...

ProfBoc75, 2024-05-23:
Hi, 

I did a little analysis from the data shared here, I don't have these devices, just want to share my findings.
First, with sync/preamble = 0x0a, we have 304 bit = 38 bytes, tried with 
`-X 'n=elsner,m=FSK_MC_ZEROBIT,short=11,long=11,reset=25,preamble=0a'`

The 2 last byte are a CRC-16, poly 0x1021, init 0x68b3, from the previous 36 bytes.

Then there is a structure, but difficult to say what is what ?
Few nibbles are static or with very few changes like only 2 values 0x0 and 0xF for example , other nibbles are more dynamic and change each message, and others have a small change, could be Temp values with slow variations.

My understanding is that the signal is bi-directional, the console is able to send and receive signal, and in such situation we may find into the message, some commands, the sender and the receiver ids into the frame, acknowledge and so on.

I noticed that sometimes data is present from one signal to the next, but shifted to the left. Could be 1 bit left shift, or 1 byte left shift. Here we can recognize the bytes shifted to the left line after line on the right part of the message just before the CRC-16.

    {304}cead93bae7a88c874f9f18cb47cf9f55b088c7bfaf8f4fcfcfcfcfcfcfcfceccc8c0b090 4f71 [ open ]
    {304}cead93bacf772cc7cf9f18cb47cf9e54aec7bfaf8f2b31ffcfcfcfcfcfceccc8c0b09050 2010 [ open ]
    {304}cead93ba9f166c47cf9f18cb47ce9d52aa7aaf8f4fcfcfcfcfcfcfcfceccc8c0b09050d0 48cd [ open ]

[bitbench](https://triq.net/bitbench#c=%7B304%7Dcead93ba66768578306099cc49d3a765d00b36bead8a45baa57a257b277f2e8c48c0b193%20e878%20%5B%20automode%20%5D&c=%7B304%7Dcead93bacd4424b7af5f98cb47cf9e54aec7beac881c11c050d0d0d0d0d1d3d7dfee0d4a%20d953%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba9adc551870e099cc48d1a25bbbe1d6165dea053aa57a257a247922751b67fe2c%20ee71%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba340fbde8102019cc49d2a45fc4f3165deae19bd57a257a247922751b67fe2c89%20db36%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba697994976fdf98cb46cc984896964ceb073faf8f4fcfcfcecdcac4b8a173175f%208dcc%20%5B%20automode%20%5D&c=%7B304%7Dcead93bad24935d8f0e099cc48d0a15ab9ddeb063d86e66aa57a257a257b277e2c8841b3%2079ed%20%5B%20automode%20%5D&c=%7B304%7Dcead93baa4ec7d68102019cc48d1a35dc0ebe93daa853aa57a257a257b277e2c8841b396%203ce5%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba48330d8850a019cc49d3a664ce063daa851606aa257a257b277e2c8841b3965c%201b0e%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba91c534d7efdf98cb47ce9d52a9bd998438a0701050d0d0d0d1d3d7deec0942b5%20fbcf%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba23e87c670f1f18cb46cd9b4da0aa8438a04c7180d0d0d0d1d3d7deec0942b59a%208d18%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba472f0c874f9f18cb46cc9949989b4aff2f8f4fcfcfcfcfcecdcbc6bca8803090%208e71%20%5B%20automode%20%5D&c=%7B304%7Dcead93ba8ebb25b8b06099cc48d1a25cbee7ff2e8d2626eaa57a257a247923771f6f0e4c%202a2d%20%5B%20automode%20%5D&c=&c=%7B304%7Dcead93ba81caf457efdf98cb47ce9d51a886a57b277f2f8f4fcfcfcfceccc9c3b69c6800%20a482%20%5B%20close%20%5D&c=%7B304%7Dcead93ba03cdfc670f1f18cb46cd9a4c9da57b277f0bf17fcfcfcfceccc9c3b69c680030%209fed%20%5B%20close%20%5D&c=%7B304%7Dcead93ba06d10578306099cc49d2a561c8b8277e2d8a45baa57a257b267c2983379f6e0c%20703a%20%5B%20close%20%5D&c=%7B304%7Dcead93ba0cdc1da8902019cc49d2a561c7f821721536472a257a257b267c2982359b66fc%203f11%20%5B%20close%20%5D&c=%7B304%7Dcead93ba19f654176fdf98cb46cd9a4b9b6d721458e0f01050d0d0d1d3d6ddea0439a275%20e5e2%20%5B%20close%20%5D&c=%7B304%7Dcead93ba3226b5d8f0e099cc49d2a45fc4af1459e2f51a65fa257a247922751b66fd2b86%2022bc%20%5B%20close%20%5D&c=%7B304%7Dcead93ba658c84772f5f98cb46cc9848959459e3f7fbd13f4fcfcfcecdcac4b9a2741860%20f65b%20%5B%20close%20%5D&c=%7B304%7Dcead93baca54159870e099cc48d0a159b7a5e3f61d6a053aa57a257a257b267d2b873eac%20363b%20%5B%20close%20%5D&c=%7B304%7Dcead93ba94e73de8102019cc48d1a25bbce3f61d6ae19cd57a257a257b267d2b873eac89%207507%20%5B%20close%20%5D&c=%7B304%7Dcead93ba291194976fdf98cb46cd9b4d9f6682349860f01050d0d0d1d3d7deed0b46bdaa%201913%20%5B%20close%20%5D&c=%7B304%7Dcead93ba526135d8f0e099cc49d3a663cc02349962d17c95fa257a247821721459e2f419%204721%20%5B%20close%20%5D&c=%7B304%7Dcead93baa50784772f5f98cb47ce9c50a5809963f71f6f0f4fcfcfcfcecdcbc6bdab873f%203341%20%5B%20close%20%5D&c=%7B304%7Dcead93ba4a4f159870e099cc49d3a664ce063ca98211fc95fa257a24782173175eed0b46%206224%20%5B%20close%20%5D&c=%7B304%7Dcead93ba95e544f72f5f98cb47ce9d52a979a983379f6f0f4fcfcfcfceccc8c1b29458e0%20b34d%20%5B%20close%20%5D&c=%7B304%7Dcead93ba2a0d959870e099cc49d2a460c6f61c690211fc95fa257a247922741861f21459%20c2a0%20%5B%20close%20%5D&c=%7B304%7Dcead93ba54603de8102019cc49d3a663cccf369d6a053aa57a257a247821721458e1f316%20f3a0%20%5B%20close%20%5D&c=%7B304%7Dcead93baa8078d8850a019cc48d1a35dbfe902359a415c557a257a257b277e2d8b46bca9%20e827%20%5B%20close%20%5D&c=%7B304%7Dcead93ba515934d7efdf98cb46cc994a995a6a0438a0701050d0d0d1d2d4d8e1f3165dea%2032d5%20%5B%20close%20%5D&c=%7B304%7Dcead93baa2f97558f0e099cc48d1a35dc0ea0439a2517c95fa257a257b277e2c8942b499%209063%20%5B%20close%20%5D&c=&c=%7B304%7Dcead93ba0ef925b8b06099cc49d2a561c7b320711255dae5fa257a247923761d6a053ba6%203c02%20%5B%20open%20%5D&c=%7B304%7Dcead93ba1c145d28902019cc49d2a562cabc2e8d4ac5baa57a257a247923771f6f0e4cc9%2011bd%20%5B%20open%20%5D&c=%7B304%7Dcead93ba394ed4176fdf98cb46cd9b4ea2ae8d4bc79b11bf4fcfcfceccc8c0b09153d7df%20340d%20%5B%20open%20%5D&c=%7B304%7Dcead93ba73c1bce70f1f18cb46cc9847944d54d8e0f01050d0d0d0d1d2d5dae4f922751a%208a2e%20%5B%20open%20%5D&c=%7B304%7Dcead93bae7a88c874f9f18cb47cf9f55b088c7bfaf8f4fcfcfcfcfcfcfcfceccc8c0b090%204f71%20%5B%20open [... cut ...] 837af5fd6ab46cd9b10a284be5afca284be80b3b2b38ec3dfef0f4ecdca%202c72&c=cead93ba7b7ff0278f1f56ab46cc9809936683e513d11e7dffb2b1b088b6c5baa4782070%204c4a&c=cead93ba2b7891a78f1f56fa46cd9b0f448912254a952b56ad5bb66bd6ac59b264c78d1a%2046e4&f=ID32h%20LEFT_SHIFT_BIT8b%208h%204b%208h%20SEVEN%2FEIGHT4b%204h%20ZERO%2FF4b%204h%20ZERO%2FF4b%208b%208h%20IV%204h%204h%20C%2FD4h%204h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%20CRC16h2x%0A&z=1&cw=4)

m1cha77, 2024-05-24:
Hello, I have some information about the communication protocol of the WiFi interface [SOL ](https://www.elsner-elektronik.de/de/sol.html)here. I hope this helps. Unfortunately, I can't do much with it myself, but I have an elsner system and would like to integrate it into Home Assistant. Greetings - Micha
[WLAN_Kommunikation.pdf](https://github.com/merbanan/rtl_433/files/15432608/WLAN_Kommunikation.pdf)

ProfBoc75, 2024-05-24:
> Hello, I have some information about the communication protocol of the WiFi interface [SOL ](https://www.elsner-elektronik.de/de/sol.html)here. I hope this helps. Unfortunately, I can't do much with [...]

Thanks, unfortunately, the SOL is a gateway between Solexa II RF 868.2 Mhz and Wireless 2.4 Ghz LAN/TCPIP, and nothing is detailed at RF level, only the TCPIP part with the TCP/UPD messages/queries. We can't use it as the RF data layout.

I found also some information from Elsner KNX and Elsner MODBUS/RS485 modules, but again, not related to RF signal, not sure if we can replicate these findings at the RF data layout. (mostly at FHEM and EM in github), the closest is at MODBUS/RS485, with 34 bytes data layout but it doesn't match.

gdt, 2025-07-04:
status?  has gone dormant.

elemakil, 2025-07-05:
I would be very interested in controlling my Solexa through rtl_433 as well. I bought an RF stick for use with SDR software, so I could provide additional data if this helps. That would require some guidance what and how to capture helpful data though as I am a novice to SDR and RF communication protocols.

dbuezas, 2026-11-02:
I have an **Elsner Arexa 230V** (the window version of the Solexa: battery RC + weather station that drives the window motor) and decoded the protocol completely, including the sensor values and sending commands. It is the same radio protocol as the Solexa; my model also decodes the samples in `rtl_433_tests/tests/Elsner-Solexa`.

The short version:
- Chips are exactly **10.000 µs** (100 kbaud), measured from I/Q; `short=11` only works because the flex decoder is tolerant.
- The scrambling: with `d[k] = c[k] - 2*c[k-1]` (mod 256, raw bytes), every payload byte is `d[k] = value[k] + (parity(seed & mask[k]) ^ inv[k])`, with `seed = c[4]` (random per packet) and a fixed mask per byte position. This rebuilds all 353 frames I recorded bit for bit, and it is where the `spread_k(p0)` relations come from.
- Bytes 5..7 are a packet counter (the station only accepts newer ones), 8..16 the system ID and the sender/receiver addresses set by the teach-in, 17..19 the RC command (open `25 ba 30`, stop `27 b6 30`, down `36 98 30`, auto `6a 1f 52`, manual `6a 13 6a`; the same codes on the Solexa samples).

The full tables (masks, byte map, commands, the Arexa station's measurements) are here:
https://github.com/dbuezas/esphome-elsner-arexa#protocol

I also put it into an ESPHome component for a CC1101 (it receives everything and sends the RC buttons; it learns the system ID from the RC's own packets, so no re-pairing is needed).

If it helps, I can share my captures (several hundred CRC-valid frames with RC display readings) or help add the scrambling model and the Arexa fields to the `elsner_solexa` decoder.