merbanan/rtl_433 #2800: 4Channel rc switch outlet plug 50074 unitec/cotech

plutphil, 2024-01-22:
Hi,
I'd like to control these 433mhz rc outlets with a 433mhz module or cc1101. Here are the signals from the 8 buttons (A/B/C/D ON/OFF). The remote sends four different signals for one button.
<details>
<summary>signals from the 8 buttons (A/B/C/D ON/OFF)</summary>
<blockquote>
*** signal_start = 1202585439, signal_end = 1203317370, signal_len = 731931, pulses_found = 207
Iteration 1. t: 3535    min: 1025 (203)    max: 6045 (4)    delta 1046754
Iteration 2. t: 3535    min: 1025 (203)    max: 6045 (4)    delta 0
Pulse coding: Short pulse length 1025 - Long pulse length 6045

Short distance: 2047, long distance: 4993, packet distance: 15075

p_limit: 3535
bitbuffer:: Number of rows: 5
[00] {105} 00 00 00 00 00 00 00 00 00 00 00 00 00 80
[01] {28} 00 00 00 10                               : 00000000 00000000 00000000 0001
[02] {25} 00 00 00 80                               : 00000000 00000000 00000000 1
[03] {25} 00 00 00 80                               : 00000000 00000000 00000000 1
[04] {24} 00 00 00                                  : 00000000 00000000 00000000
*** signal_start = 1207100104, signal_end = 1207832129, signal_len = 732025, pulses_found = 206
Iteration 1. t: 3598    min: 1135 (202)    max: 6061 (4)    delta 1276964
Iteration 2. t: 3598    min: 1135 (202)    max: 6061 (4)    delta 0
Pulse coding: Short pulse length 1135 - Long pulse length 6061

Short distance: 1939, long distance: 4986, packet distance: 15082

p_limit: 3598
bitbuffer:: Number of rows: 5
[00] {104} 00 00 00 00 00 00 00 00 00 00 00 00 01
[01] {27} 00 00 00 20                            : 00000000 00000000 00000000 001
[02] {25} 00 00 00 80                            : 00000000 00000000 00000000 1
[03] {26} 00 00 00 40                            : 00000000 00000000 00000000 01
[04] {24} 00 00 00                               : 00000000 00000000 00000000

*** signal_start = 1211225409, signal_end = 1211957398, signal_len = 731989, pulses_found = 211
Iteration 1. t: 3553    min: 1055 (207)    max: 6052 (4)    delta 1109098
Iteration 2. t: 3553    min: 1055 (207)    max: 6052 (4)    delta 0
Pulse coding: Short pulse length 1055 - Long pulse length 6052

Short distance: 1964, long distance: 5007, packet distance: 15073

p_limit: 3553
bitbuffer:: Number of rows: 5
[00] {111} 00 00 00 00 00 00 00 00 00 00 00 00 00 02
[01] {25} 00 00 00 80                               : 00000000 00000000 00000000 1
[02] {26} 00 00 00 40                               : 00000000 00000000 00000000 01
[03] {25} 00 00 00 80                               : 00000000 00000000 00000000 1
[04] {24} 00 00 00                                  : 00000000 00000000 00000000

*** signal_start = 1219357130, signal_end = 1220089268, signal_len = 732138, pulses_found = 566
Iteration 1. t: 2210    min: 465 (564)    max: 3956 (2)    delta 1511690
Iteration 2. t: 2210    min: 465 (564)    max: 3956 (2)    delta 0
Pulse coding: Short pulse length 465 - Long pulse length 3956

Short distance: 753, long distance: 5007, packet distance: 15102

p_limit: 2210
bitbuffer:: Number of rows: 5
[00] {387} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[01] {42} 00 00 00 00 00 80                                                                                                                                  : 00000000 00000000 00000000 00000000 00000000 10
[02] {42} 00 00 00 00 00 00                                                                                                                                  : 00000000 00000000 00000000 00000000 00000000 00
[03] {45} 00 00 00 00 00 10                                                                                                                                  : 00000000 00000000 00000000 00000000 00000000 00010
[04] {50} 00 00 00 00 00 00 00                                                                                                                               : 00000000 00000000 00000000 00000000 00000000 00000000 00
*** signal_start = 1224148076, signal_end = 1224880028, signal_len = 731952, pulses_found = 845
Iteration 1. t: 2528    min: 288 (844)    max: 4769 (1)    delta 81796
Iteration 2. t: 2528    min: 288 (844)    max: 4769 (1)    delta 0
Pulse coding: Short pulse length 288 - Long pulse length 4769

Short distance: 544, long distance: 5014, packet distance: 15097

[... 285 lines cut ...]

p_limit: 3558
bitbuffer:: Number of rows: 5
[00] {107} 00 00 00 00 00 00 00 00 00 00 00 00 00 20
[01] {25} 00 00 00 80                               : 00000000 00000000 00000000 1
[02] {27} 00 00 00 20                               : 00000000 00000000 00000000 001
[03] {25} 00 00 00 80                               : 00000000 00000000 00000000 1
[04] {25} 00 00 00 00                               : 00000000 00000000 00000000 0

*** signal_start = 41106542, signal_end = 41838306, signal_len = 731764, pulses_found = 204
Iteration 1. t: 3569    min: 1071 (200)    max: 6068 (4)    delta 1143225
Iteration 2. t: 3569    min: 1071 (200)    max: 6068 (4)    delta 0
Pulse coding: Short pulse length 1071 - Long pulse length 6068

Short distance: 2016, long distance: 4983, packet distance: 15046

p_limit: 3569
bitbuffer:: Number of rows: 5
[00] {104} 00 00 00 00 00 00 00 00 00 00 00 00 01
[01] {25} 00 00 00 80                            : 00000000 00000000 00000000 1
[02] {25} 00 00 00 80                            : 00000000 00000000 00000000 1
[03] {26} 00 00 00 40                            : 00000000 00000000 00000000 01
[04] {24} 00 00 00                               : 00000000 00000000 00000000
</blockquote>
</details>

ProfBoc75, 2024-01-22:
@plutphil : This is a classic remote control @ 433 MHz for 1 to 4 plugs kits. I have the same remote, Chacon 50074 to command outside plugs. And the same , it decoded as a "Smoke-GS558" model.

But this type of remote control is very well managed with RFXCOM, so you can receive and send signal to the plugs. The protocol is "Lighting 5", and sub type is "Kangtai / Cotech" under Domoticz with RFXCOM.

The remote control is identified with an unique ID.
For each plug you have the 2 signals ON / OFF.
The plug 1 is identified as Unit 1, plug 2 , Unit 2. ... And to command all plugs (all ON or all OFF)  = Unit 0.

I'm able to look at that and try to understand why it's not well handle under rtl_433, at least the decoding part.

To be continued ....

zuckschwerdt, 2024-01-22:
You need to analyze with `-A`. Do not use `-a`, it is not supported.

The Smoke-GS558 is an EV1527 protocol and this collides with other EV1527 protocols (i.e. most cheap OOK doorbell / remote button things).

closed 2024-01-22 by zuckschwerdt

plutphil, 2024-01-22:
<details>
<summary>
using "-A" i get this when pressing the A ON button. 
</summary>
<blockquote>
Detected OOK package    2024-01-22 19:33:06
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 19:33:06
model     : Smoke-GS558  id        : 29729
unit      : 24           learn     : 0             Raw Code  : 0e8438
Analyzing pulses...
Total count:  200,  width: 347.43 ms            (86857 S)
Pulse width distribution:
 [ 0] count:   68,  width:  320 us [292;340]    (  80 S)
 [ 1] count:   64,  width: 1024 us [948;1104]   ( 256 S)
 [ 2] count:    4,  width: 3028 us [3024;3036]  ( 757 S)
 [ 3] count:   64,  width:  448 us [412;464]    ( 112 S)
Gap width distribution:
 [ 0] count:    4,  width: 2364 us [2364;2368]  ( 591 S)
 [ 1] count:  127,  width: 1156 us [1080;1244]  ( 289 S)
 [ 2] count:   32,  width:  452 us [444;484]    ( 113 S)
 [ 3] count:    4,  width: 7284 us [7276;7296]  (1821 S)
 [ 4] count:   32,  width:  580 us [568;600]    ( 145 S)
Pulse period distribution:
 [ 0] count:    4,  width: 2688 us [2676;2704]  ( 672 S)
 [ 1] count:  191,  width: 1544 us [1496;1580]  ( 386 S)
 [ 2] count:    4,  width: 10312 us [10308;10320]       (2578 S)
Pulse timing distribution:
 [ 0] count:   68,  width:  320 us [292;340]    (  80 S)
 [ 1] count:  191,  width: 1112 us [948;1244]   ( 278 S)
 [ 2] count:    4,  width: 3028 us [3024;3036]  ( 757 S)
 [ 3] count:   96,  width:  452 us [412;484]    ( 113 S)
 [ 4] count:    4,  width: 2364 us [2364;2368]  ( 591 S)
 [ 5] count:    4,  width: 7284 us [7276;7296]  (1821 S)
 [ 6] count:   32,  width:  580 us [568;600]    ( 145 S)
 [ 7] count:    1,  width: 30364 us [30364;30364]       (7591 S)
Level estimates [high, low]:  15896,    972
RSSI: -0.1 dB SNR: 12.1 dB Noise: -12.3 dB
Frequency offsets [F1, F2]:   -5555,      0     (-21.2 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB0130801014004580BD401C4093C1C740244769C8455+AAB02B0803014004580BD401C4093C1C740244769C8181819393938181818193818181819381939393818181818455+AAB02B0801014004580BD401C4093C1C740244769C818181939393818181819381818181938193939381818181A555+AAB02B0803014004580BD401C4093C1C740244769CB1B1B1969696B1B1B1B196B1B1B1B196B1969696B1B1B1B1A555+AAB02A0801014004580BD401C4093C1C740244769CB1B1B1969696B1B1B1B196B1B1B1B196B1969696B1B1B1B755

Detected OOK package    2024-01-22 19:33:08
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 19:33:08
model     : Smoke-GS558  id        : 27218
unit      : 8            learn     : 0             Raw Code  : 0d4a48
Analyzing pulses...
Total count:  200,  width: 347.36 ms            (86841 S)
Pulse width distribution:
 [ 0] count:   68,  width:  320 us [296;336]    (  80 S)
 [ 1] count:   64,  width: 1028 us [952;1100]   ( 257 S)
 [ 2] count:    4,  width: 3024 us [3024;3032]  ( 756 S)
 [ 3] count:   64,  width:  452 us [424;464]    ( 113 S)
Gap width distribution:
 [ 0] count:    4,  width: 2360 us [2352;2368]  ( 590 S)
 [ 1] count:  127,  width: 1156 us [1080;1244]  ( 289 S)
 [ 2] count:   32,  width:  452 us [440;464]    ( 113 S)
 [ 3] count:    4,  width: 7280 us [7276;7292]  (1820 S)
 [ 4] count:   32,  width:  580 us [568;604]    ( 145 S)
Pulse period distribution:
 [ 0] count:    4,  width: 2688 us [2676;2700]  ( 672 S)
 [ 1] count:  191,  width: 1544 us [1512;1580]  ( 386 S)
 [ 2] count:    4,  width: 10304 us [10300;10316]       (2576 S)
Pulse timing distribution:
 [ 0] count:   68,  width:  320 us [296;336]    (  80 S)
 [ 1] count:  191,  width: 1112 us [952;1244]   ( 278 S)
 [ 2] count:    4,  width: 3024 us [3024;3032]  ( 756 S)
 [ 3] count:   96,  width:  452 us [424;464]    ( 113 S)
 [ 4] count:    4,  width: 2360 us [2352;2368]  ( 590 S)
[... 58 lines cut ...]
Gap width distribution:
 [ 0] count:    4,  width: 2352 us [2352;2360]  ( 588 S)
 [ 1] count:  127,  width: 1148 us [1068;1232]  ( 287 S)
 [ 2] count:   34,  width:  448 us [424;556]    ( 112 S)
 [ 3] count:    4,  width: 7276 us [7264;7288]  (1819 S)
 [ 4] count:   30,  width:  568 us [560;592]    ( 142 S)
Pulse period distribution:
 [ 0] count:    4,  width: 2692 us [2684;2704]  ( 673 S)
 [ 1] count:  191,  width: 1544 us [1516;1572]  ( 386 S)
 [ 2] count:    4,  width: 10308 us [10296;10324]       (2577 S)
Pulse timing distribution:
 [ 0] count:   68,  width:  328 us [308;352]    (  82 S)
 [ 1] count:  191,  width: 1108 us [960;1232]   ( 277 S)
 [ 2] count:    4,  width: 3032 us [3028;3044]  ( 758 S)
 [ 3] count:  128,  width:  484 us [424;592]    ( 121 S)
 [ 4] count:    4,  width: 2352 us [2352;2360]  ( 588 S)
 [ 5] count:    4,  width: 7276 us [7264;7288]  (1819 S)
 [ 6] count:    1,  width: 30444 us [30444;30444]       (7611 S)
Level estimates [high, low]:  15918,   1368
RSSI: -0.1 dB SNR: 10.7 dB Noise: -10.8 dB
Frequency offsets [F1, F2]:   -5767,      0     (-22.0 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB0110701014804540BD801E409301C6C76EC8455+AAB0290703014804540BD801E409301C6C76EC8181819393938181818193818181819381939393818181818455+AAB0290701014804540BD801E409301C6C76EC818181939393818181819381818181938193939381818181A555+AAB0290703014804540BD801E409301C6C76ECB1B1B1939393B1B1B1B193B1B1B1B193B1939393B1B1B1B1A555+AAB0280701014804540BD801E409301C6C76ECB1B1B1939393B1B1B1B193B1B1B1B193B1939393B1B1B1B655
</blockquote>
</details>

zuckschwerdt, 2024-01-22:
Looks like PWM but with two different timings (likely for compat with receivers).
1st is 320 / 1100 µs, sync is 2300 µs gap,
2nd is 450 / 1000 µs, sync is 3000 µs pulse,
This should work for both:
`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=320,l=1100,y=3000g=1500,r=3500,bits>=24'`

Data is e.g. `EE 66 CF` or ` EE 66 CB` (ignores the last bit which is sync).

plutphil, 2024-01-22:
a on:
<details>
```_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:11
model     : name         count     : 1             num_rows  : 3             rows      :
len       : 3            data      : 6,
len       : 99           data      : e5e2bff2f15ff978affcbc57e,
len       : 0            data      :
codes     : {3}6, {99}e5e2bff2f15ff978affcbc57e, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:11
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : e5e2bf,
len       : 0            data      :
codes     : {24}e5e2bf, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:11
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : e5e2bf,
len       : 0            data      :
codes     : {24}e5e2bf, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:11
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : e5e2bf,
len       : 0            data      :
codes     : {24}e5e2bf, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:11
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 24           data      : e5e2bf
codes     : {24}e5e2bf
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:13
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 100          data      : f1ef47f8f7a3fc7bd1fe3de8f,
len       : 0            data      :
codes     : {100}f1ef47f8f7a3fc7bd1fe3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:13
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : e3de8f,
len       : 0            data      :
codes     : {24}e3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:13
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : e3de8f,
len       : 0            data      :
codes     : {24}e3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:13
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : e3de8f,
len       : 0            data      :
codes     : {24}e3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:13
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 24           data      : e3de8f
codes     : {24}e3de8f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:15
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 100          data      : f6d6a7fb6b53fdb5a9fedad4f,
len       : 0            data      :
codes     : {100}f6d6a7fb6b53fdb5a9fedad4f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:25:15
model     : name         count     : 1             num_rows  : 2             rows      :
[... 259 lines cut ...]
codes     : {100}f73366fb99b37dccd9bee66cd, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:28:18
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : ee66cd,
len       : 0            data      :
codes     : {24}ee66cd, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:28:18
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : ee66cd,
len       : 0            data      :
codes     : {24}ee66cd, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:28:18
model     : name         count     : 1             num_rows  : 2             rows      :
len       : 24           data      : ee66cd,
len       : 0            data      :
codes     : {24}ee66cd, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:28:18
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 24           data      : ee66cd
codes     : {24}ee66cd
</details>

zuckschwerdt, 2024-01-22:
Oops, there was a comma missing, use this and you shouldn't see and `{100}` messages anymore. The shorter codes look good.
`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=320,l=1100,y=3000,g=1500,r=3500,bits>=24'`

plutphil, 2024-01-22:
<details>
````
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:14
model     : name         count     : 4             num_rows  : 6
rows      :
len       : 1            data      : 8,
len       : 25           data      : e3de8f8,
len       : 25           data      : e3de8f8,
len       : 25           data      : e3de8f8,
len       : 24           data      : e3de8f,
len       : 0            data      :
codes     : {1}8, {25}e3de8f8, {25}e3de8f8, {25}e3de8f8, {24}e3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:14
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : e3de8f,
len       : 0            data      :
codes     : {24}e3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:14
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : e3de8f,
len       : 0            data      :
codes     : {24}e3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:14
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : e3de8f,
len       : 0            data      :
codes     : {24}e3de8f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:14
model     : name         count     : 1             num_rows  : 1
rows      :
len       : 24           data      : e3de8f
codes     : {24}e3de8f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:15
model     : name         count     : 4             num_rows  : 6
rows      :
len       : 1            data      : 8,
len       : 25           data      : edad4f8,
len       : 25           data      : edad4f8,
len       : 25           data      : edad4f8,
len       : 24           data      : edad4f,
len       : 0            data      :
codes     : {1}8, {25}edad4f8, {25}edad4f8, {25}edad4f8, {24}edad4f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:15
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : edad4f,
len       : 0            data      :
codes     : {24}edad4f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:15
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : edad4f,
len       : 0            data      :
codes     : {24}edad4f, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:15
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : edad4f,
[... 479 lines cut ...]
len       : 24           data      : ee66cf,
len       : 0            data      :
codes     : {24}ee66cf, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:36
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : ee66cf,
len       : 0            data      :
codes     : {24}ee66cf, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:36
model     : name         count     : 1             num_rows  : 2
rows      :
len       : 24           data      : ee66cf,
len       : 0            data      :
codes     : {24}ee66cf, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-22 20:52:36
model     : name         count     : 1             num_rows  : 1
rows      :
len       : 24           data      : ee66cf
codes     : {24}ee66cf
````
</details>

zuckschwerdt, 2024-01-22:
Perfect. Now the data codes and the PWM timings are known. You can rebuild a clean signal to send with that.

plutphil, 2024-01-24:
Finally got it working. the Arduino RC-switch library did not work. so i hard-coded the timings. Fun-Fact: it worked even without the second slower transmission. I couldn't get RC-switch to make the sync and transmit the 4 packages at once. it made gaps between. I tested on an Arduino nano atmega328p and a XD-FST 433mhz RF module.
<pre><code>#define TX_PIN 10
void setup() {
  pinMode(TX_PIN,OUTPUT);
}
void sendSync(){
  digitalWrite(TX_PIN,1);
  delayMicroseconds(320);
  digitalWrite(TX_PIN,0);
  delayMicroseconds(2300 );
}
void sendOne(){
  digitalWrite(TX_PIN,1);
  delayMicroseconds(320);
  digitalWrite(TX_PIN,0);
  delayMicroseconds(1100);
}
void sendZero(){
  digitalWrite(TX_PIN,1);
  delayMicroseconds(1100);
  digitalWrite(TX_PIN,0);
  delayMicroseconds(320);
}

void sendBits(uint32_t x,uint8_t bits){
  for(int i=bits-1;i>=0;i--){
    if((x>>i)&1){
      sendOne();
    }else{
      sendZero();
    }
  }  
}
void loop() {
  sendSync();
  sendBits(0xe3de8f,24);
  sendBits(0xe3de8f,24);
  sendBits(0xe3de8f,24);
  delay(5000);
}

zuckschwerdt, 2024-01-24:
Great to hear, and thanks for the code example. This will help others get started with similar projects!

ProfBoc75, 2024-01-26:
Hi @zuckschwerdt : I know this issue is closed, but I would like to share few findings here.

* smoke_gs558.c : The "Raw Code" is not "raw" but already altered, ie, inverted and the order of bytes is reversed (2 1 0 instead of 0 1 2). Very simple to check, with the -R 86:v you don't have the same value between this "Raw Code" and "codes".
This could impact external decoder and add some useless complexity. It's very simple to correct.

* I rewrote the device decoder kangtai.c (Kangtai/Cotech RC protocol, improvement of an [already committed old decoder](https://github.com/merbanan/rtl_433/blob/a8f438a5fb52f24f854368b581a08aa413bb7654/src/devices/kangtai.c) but never merged).

It's in conflict with smoke_gs558 so it can't be enabled by default for the moment, I didn't find yet a way to identify smoke device and rc device.

Anaway, I'm able to decode the Kangtai/Cotech protocol and extract :
* ID
* Learn situation
* Command : ON or OFF 
* Which button ( = Unit ID)
* Loop number : The data is simply encrypted/decrypted using a map table and for a same command you have 4 codes from where you can extract the position into this Loop from 0 to 3. Each time you press a button you have the next code into the loop (a rolling code).

So remaining the DIM command and DIM value, I don't have such device to test, but I can try to add it based on what I found on the internet, like [here](https://github.com/telldus/tellstick-server/blob/master/rf433/src/rf433/ProtocolKangtai.py)

From @plutphil above details share in this issue, I'm able to decode raw code like this :

Data | Cmd | Button / Unit | ID
-- | -- | -- | --
{24}e5e2bf | on | 1 | 815f
{24}e3de8f | on | 1 | 815f
{24}edad4f | on | 1 | 815f
{24}ee66cf | on | 1 | 815f
{24}e1b86f | off | 1 | 815f
{24}e3de8b | on | 2 | 815f
{24}e5e2bb | on | 2 | 815f
{24}e1b86b | off | 2 | 815f
{24}e5e2b3 | on | 3 | 815f
{24}e1b863 | off | 3 | 815f
{24}e939dd | on | 4 | 815f
{24}edad4d | off | 4 | 815f

zuckschwerdt, 2024-02-01:
I have no idea about the Kangtai decoder, I have never seen it. The collision with Smoke GS558 is expected. The protocol is simply EV1527 and we should default-disable GS558 and this decoder -- a user has to know and choose one of these decoders, otherwise it's just random decodings.