merbanan/rtl_433 #2803: Problem with RTL_433

rowanbradley, 2024-01-27:
I'm trying to run RTL_433 with the command line:
    rtl_433_64bit_static.exe -R 43 -F json:testlog.jsn
I'm getting:
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Program Files\rtl_433-win-x64-21.12\rtl_433.conf"...
Trying conf file at "C:\Users\Rowan\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 1 out of 207 device decoding protocols [ 43 ]
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules

I found a file called rtl-sdr.rules. Do I have to modify this file? Where do I put it? What else do I have to do to get it to run?

Thank you - Rowan

zuckschwerdt, 2024-01-27:
That advice does not apply to Windows. You need to ensure proper SDR setup, something about installing Zadig.

rowanbradley, 2024-01-29:
I am making a bit of progress. Here is the JSON logged:

{"time" : "2024-01-29 12:30:01.431416", "enabled" : 188, "since" : "2024-01-29T12:20:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 12:40:01.201413", "enabled" : 188, "since" : "2024-01-29T12:30:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 12:50:01.496458", "enabled" : 188, "since" : "2024-01-29T12:40:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 13:00:01.267007", "enabled" : 188, "since" : "2024-01-29T12:50:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 13:10:01.037050", "enabled" : 188, "since" : "2024-01-29T13:00:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 13:20:01.331347", "enabled" : 188, "since" : "2024-01-29T13:10:01", "frames" : {"count" : 17, "fsk" : 5, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 13:30:01.102036", "enabled" : 188, "since" : "2024-01-29T13:20:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 13:40:01.396446", "enabled" : 188, "since" : "2024-01-29T13:30:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 13:49:37.678563", "protocol" : 81, "model" : "Oil-SonicStd", "id" : 40283, "flags" : 112, "alarm" : 0, "binding_countdown" : 0, "depth_cm" : 15, "mod" : "FSK", "freq1" : 433.948, "freq2" : 433.877, "rssi" : -0.257, "snr" : 15.018, "noise" : -15.274}
{"time" : "2024-01-29 13:50:01.167269", "enabled" : 188, "since" : "2024-01-29T13:40:01", "frames" : {"count" : 0, "fsk" : 1, "events" : 1}, "stats" : [{"device" : 81, "name" : "Oil Ultrasonic STANDARD FSK", "events" : 1, "ok" : 1, "messages" : 1}]}
{"time" : "2024-01-29 14:00:01.461870", "enabled" : 188, "since" : "2024-01-29T13:50:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 14:10:01.232948", "enabled" : 188, "since" : "2024-01-29T14:00:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 14:42:57.820674", "enabled" : 188, "since" : "2024-01-29T14:42:57", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 14:42:57.860715", "enabled" : 188, "since" : "2024-01-29T14:42:57", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 14:42:57.901610", "enabled" : 188, "since" : "2024-01-29T14:42:57", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 14:50:01.362325", "enabled" : 188, "since" : "2024-01-29T14:42:57", "frames" : {"count" : 5, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 15:00:01.133087", "enabled" : 188, "since" : "2024-01-29T14:50:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 15:10:01.427482", "enabled" : 188, "since" : "2024-01-29T15:00:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 15:20:01.198199", "enabled" : 188, "since" : "2024-01-29T15:10:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 15:24:29.247739", "protocol" : 81, "model" : "Oil-SonicStd", "id" : 4055, "flags" : 96, "alarm" : 0, "binding_countdown" : 0, "depth_cm" : 36, "mod" : "FSK", "freq1" : 433.952, "freq2" : 433.887, "rssi" : -0.173, "snr" : 15.331, "noise" : -15.505}
{"time" : "2024-01-29 15:30:01.493594", "enabled" : 188, "since" : "2024-01-29T15:20:01", "frames" : {"count" : 0, "fsk" : 1, "events" : 1}, "stats" : [{"device" : 81, "name" : "Oil Ultrasonic STANDARD FSK", "events" : 1, "ok" : 1, "messages" : 1}]}
{"time" : "2024-01-29 15:33:01.851967", "protocol" : 81, "model" : "Oil-SonicStd", "id" : 5140, "flags" : 48, "alarm" : 1, "binding_countdown" : 0, "depth_cm" : 97, "mod" : "FSK", "freq1" : 433.953, "freq2" : 433.886, "rssi" : -0.191, "snr" : 15.422, "noise" : -15.613}
{"time" : "2024-01-29 15:40:01.263317", "enabled" : 188, "since" : "2024-01-29T15:30:01", "frames" : {"count" : 0, "fsk" : 1, "events" : 1}, "stats" : [{"device" : 81, "name" : "Oil Ultrasonic STANDARD FSK", "events" : 1, "ok" : 1, "messages" : 1}]}
{"time" : "2024-01-29 15:49:54.166092", "protocol" : 81, "model" : "Oil-SonicStd", "id" : 40283, "flags" : 112, "alarm" : 0, "binding_countdown" : 0, "depth_cm" : 15, "mod" : "FSK", "freq1" : 433.946, "freq2" : 433.875, "rssi" : -0.271, "snr" : 14.229, "noise" : -14.500}
{"time" : "2024-01-29 15:50:01.033844", "enabled" : 188, "since" : "2024-01-29T15:40:01", "frames" : {"count" : 0, "fsk" : 1, "events" : 1}, "stats" : [{"device" : 81, "name" : "Oil Ultrasonic STANDARD FSK", "events" : 1, "ok" : 1, "messages" : 1}]}
{"time" : "2024-01-29 16:00:01.329455", "enabled" : 188, "since" : "2024-01-29T15:50:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 16:10:01.100186", "enabled" : 188, "since" : "2024-01-29T16:00:01", "frames" : {"count" : 0, "fsk" : 0, "events" : 0}, "stats" : []}
{"time" : "2024-01-29 16:13:08.072450", "protocol" : 81, "model" : "Oil-SonicStd", "id" : 63175, "flags" : 112, "alarm" : 1, "binding_countdown" : 0, "depth_cm" : 23, "mod" : "FSK", "freq1" : 433.952, "freq2" : 433.883, "rssi" : -0.260, "snr" : 16.055, "noise" : -16.316}
[... cut ...]
2. Only the records showing model = Oil-SonicStd are of interest to me, as far as I know. The other records don't seem to contain any information telling me where they are coming from. How do I find out? How do I tell rtl_433 the return only the Oil-SonicStd records?

3. I think that of the Oil-SonicStd records, only the ones with id = 40283 are mine. Presumably the others are oil tanks of neighbours. How do I tell rtl_433 to only return records from id 40283?

4. It seemed to miss the reading that should have been taken at 14:49. Why might it have done this?

Thank you - Rowan

zuckschwerdt, 2024-01-29:
- The format is [JSON Lines](https://jsonlines.org/) or JSONL, not a JSON Document. (S.a. [Why jsonl is better than a regular JSON](https://hackernoon.com/json-lines-format-76353b4e588d))
-  the reports are activated by the `-M stats` option or the `report_meta stats` line in the conf file. "Oil Ultrasonic STANDARD FSK" is protocol 81 (see `rtl_433 -R`) and can be selected by `-R 81`
- We have no further options to filter. You need other additional processing to filter by e.g. `id`.
- Interference might cause dropouts, it shouldn't be too often though.

rowanbradley, 2024-01-31:
>the reports are activated by the -M stats option or the report_meta stats line in the conf file. "Oil Ultrasonic STANDARD FSK" is protocol 81 (see rtl_433 -R) and can be selected by -R 81

Why is it using protocol 81 if I've specified -R 43, which claims to be the correct protocol for my oil level monitor (Apollo)? What is the difference between protocols 43 and 81?

Thank you - Rowan

zuckschwerdt, 2024-01-31:
Is 43 (older Apollo Sonic Smart, OOK) the right protocol though? This seems to be 81 (newer Apollo Ultrasonic, FSK).

closed 2024-02-29 by zuckschwerdt