merbanan/rtl_433 #2807: Weird problem with rtl_433

carefulcomputer, 2024-01-29:
I am using a Fitipower FC0012 tuner and trying to capture traffic from honeywell door sensors (protocol 70)
Here is my conf file
```
output kv:/rtlData/rtl_433.log
frequency     395M
convert     si
protocol 70
report_meta time:iso:usec:tz
report_meta level
report_meta noise
report_meta stats
report_meta protocol
pulse_detect autolevel
```

for weird reasons i see any data only when I add verbose mode ( -vvv). Otherwise there is no output at all. What could I be doing wrong ?

carefulcomputer, 2024-01-29:
closing this. found the issue. I had to set the sampling rate to 1024k. after that it worked fine.

closed 2024-01-29 by carefulcomputer

zuckschwerdt, 2024-01-29:
Interesting. We'd expect 250k to work (this protocol is OOK). Maybe try offsetting the center frequency a little, e.g. `-f 345.1M` or `-f 344.9M`

carefulcomputer, 2024-01-29:
Here is full config which finally worked for me. will try offsetting the center frequency and report back (with default sample rate)

```
convert     si
sample_rate 1024k
frequency 345M
decoder n=honeywell_door_sensor2gig,m=OOK_MC_ZEROBIT,s=156,r=292
report_meta time:iso:usec:tz
report_meta protocol
pulse_detect autolevel
pulse_detect squelch
```

carefulcomputer, 2024-01-30:
here is where the things are.  i don't see any decoded values on command line, however when i add -vvvv it seems to print all the packets but I don't see any decoded values.
here is my command line

`rtl_433  -F kv -Y auto -C si -vvvv`

and here is output (i know it is not 345mhz, this is just for testing). it doesn't show any nicely printed decoded output. when i remove -vvvv then it just sits with no values after initial startup messages. do i need some specific additional parameters to decode all the data it is seeing ? My assumption was if i don't specify any -R parameter, it will decode all values. 

```
pulse_slicer_pwm] EMOS E6016 rain gauge
codes     : {1}8
[pulse_slicer_pwm] Rubicson Pool Thermometer 48942
codes     : {1}8
[tpms_tyreguard400_callback] Bad message in row 0 need 88 bits got 1
codes     : {1}0
[tpms_tyreguard400_callback] Bad transmission
[pulse_slicer_manchester_zerobit] TyreGuard 400 TPMS
codes     : {1}0
[pulse_slicer_pcm] Neptune R900 flow meters
codes     : {12}800
[pulse_slicer_pwm] Govee Water Leak Detector H5054
codes     : {1}8
[pulse_slicer_pwm] Revolt NC-5642 Energy Meter
codes     : {1}0
[tpms_eezrv_decode] Preamble not found
[pulse_slicer_manchester_zerobit] EezTire E618, Carchet TPMS
codes     : {1}8

and so on..
```

zuckschwerdt, 2024-01-30:
Yes, `rtl_433` without any options will run all decoders. It seems you have interference or problems with the RTL-SDR stick.
Capture the band an visually check it. https://triq.org/rtl_433/ANALYZE.html#verify-a-transmission