merbanan/rtl_433 #2810: BRESSER-11-Tage-4CAST-CV-WLAN-Wetterstation-7-in-1

tembhub, 2024-01-30:
i saw this issue: https://github.com/merbanan/rtl_433/issues/1492, and i am having the same problem. i already did some captures with "rtl_433 -f 868.3M -S unknown", and the data looks promising to me:

[image]

is there a chance that this device can be added to the known decodings? what kind of capture data can i provide to help out?

link to device i used here: https://www.bresser.de/Wetter-Zeit/Wettercenter/BRESSER-11-Tage-4CAST-CV-WLAN-Wetterstation-7-in-1.html

zuckschwerdt, 2024-01-30:
The Bresser-7in1 was added in late 2020, it should be available in any release after that date.

Maybe tune up a bit, e.g. `-f 868.2M` or perhaps `-f 868.3M` (but that is aweful close to the signal shown).

Try `rtl_433 -f 868.2M -R 173,vv` to get debug info from the decoder.

Worst case try `rtl_433 -f 868.2M -s 1000k -X 'n=Bresser-7in1,m=FSK_PCM,s=125,l=125,r=9000,preamble=2dd4'` to try to get codes.

tembhub, 2024-01-30:
thanks for your reply. i tried both suggestions, but i got no results. how long should i wait for results to show up? i waited a few minutes after each command.

1)
rtl_433 -f 868.2M -R 173 -vv
rtl_433 version 23.11-37-g0ae39cc5 branch master at 202401281113 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

[Protocols] Registered 1 out of 251 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Nooelec, NESDR SMArt v5, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 1000000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] Tuned to 868.200MHz.
Allocating 15 zero-copy buffers
[Baseband] low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
^CSignal caught, exiting!
[Input] sdr_handler exit
[rtl_433] stopping...
Reattached kernel driver

2)
rtl_433 -f 868.2M -s 1000k -X 'n=Bresser-7in1,m=FSK_PCM,s=125,l=125,r=9000,preamble=2dd4'
rtl_433 version 23.11-37-g0ae39cc5 branch master at 202401281113 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
Reattached kernel driver

zuckschwerdt, 2024-01-31:
Try also `-Y autolevel -M level -M noise` to at least get the noise stats. Something seems to prevent reception.

zuckschwerdt, 2024-01-31:
That's too much noise. You'd normally see somewhere around -20 dB

tembhub, 2024-01-31:
that was a really great tip, i moved the antenna somewhere else and get ~-20dB now. success:

```
[Auto Level] Estimated noise level is -19.2 dB, adjusting minimum detection level to -16.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-31 15:36:51
model     : Bresser-7in1 id        : 43954
Temperature: 8.8 C       Humidity  : 71            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s
Direction : 252          Rain      : 353.8 mm      Light     : 8.112 klx     Light     : 8112.000 lux
UV Index  : 0.6          Battery   : 1             Integrity : CRC
Modulation: FSK          Freq1     : 868.2 MHz     Freq2     : 868.3 MHz
RSSI      : -0.1 dB      SNR       : 20.1 dB       Noise     : -20.2 dB
[Auto Level] Current noise level -19.6 dB, estimated noise -19.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-01-31 15:37:04
model     : Bresser-7in1 id        : 43954
Temperature: 8.8 C       Humidity  : 71            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s
Direction : 252          Rain      : 353.8 mm      Light     : 8.094 klx     Light     : 8094.000 lux
UV Index  : 0.6          Battery   : 1             Integrity : CRC
Modulation: FSK          Freq1     : 868.2 MHz     Freq2     : 868.3 MHz
RSSI      : -0.1 dB      SNR       : 21.0 dB       Noise     : -21.1 dB
^CSignal caught, exiting!

thank you VERY mouch for your help, much appreciated. :-)
```
[rtl_433] stopping...

closed 2024-01-31 by zuckschwerdt